dusty-nv/jetson-containers #123: ROS2 builds are not for release

jplapp, 2022-02-16:
We discovered that the ROS2 builds, e.g. for Galactic, are not release builds and thus run around 2x-3x slower than when compiled with `--cmake-args -DCMAKE_BUILD_TYPE=Release` 

The reason is that release is not specified by default in colcon: https://github.com/colcon/colcon-cmake/issues/96 

Is that intentional, or can it be changed to a release build by default?

dusty-nv, 2022-02-16:
Thanks @jplapp, I was not aware of this.  I am testing the `--cmake-args -DCMAKE_BUILD_TYPE=Release` flag to `colcon build` now, and will include it in future builds if all goes well.

dusty-nv, 2022-02-16:
Just patched this in https://github.com/dusty-nv/jetson-containers/commit/5d0f8b47b9b9ef986d88b14693805feec8cdf41a

zjeffer, 2022-03-01:
@dusty-nv @jplapp Is this working yet? Publishing with `ros:galactic-pytorch-l4t-r32.6.1` as a base image seems to be extremely slow.

dusty-nv, 2022-03-01:
@zjeffer yes, but IIRC only `dustynv/ros:foxy-ros-base-l4t-r32.6.1` was retroactively rebuilt and pushed to dockerhub.  

I will kick off builds of the other ones too.

zjeffer, 2022-03-02:
@dusty-nv Is there an ETA for when these builds will complete? I tried building it on my own nano but I just gave up, because after waiting 5+ hours it looked as if it just restarted the build for some reason. No idea what went wrong.

dusty-nv, 2022-03-02:
@zjeffer they are pushing to dockerhub now - they should be up within the hour

closed 2025-02-21 by johnnynunez