dusty-nv/jetson-containers #124: foonathan_memory_vendor package fails to build

canberkgurel, 2022-02-16:
I'm trying to build ros:foxy-ros-base-l4t-r32.5.1 container on my Xavier NX (Ubuntu 18.04.5 and Jetpack 4.5.1)

This is how I used the build script `./scripts/docker_build_ros.sh --distro foxy`

But I'm running into an error. `foonathan_memory_vendor` fails to build. @dusty-nv could you please check the terminal output provided below.

[docker-ros-build-foxy-output.txt](https://github.com/dusty-nv/jetson-containers/files/8080837/docker-ros-build-foxy-output.txt)

canberkgurel, 2022-02-16:
In my understanding, it's related to the CMake version.  

> CMake 3.11 or higher is required.  You are running version 3.10.2

This is related to the CMake in the container (not in the actual environment), right? Is CMake version 3.11 part of the container?

dusty-nv, 2022-02-16:
Hi @canberkgurel, I just encountered this too - the `foonathan_memory_vendor` must have been updated an now requires a newer version of CMake.   I updated the Foxy/Galactic dockerfiles to install a newer version of CMake in commit https://github.com/dusty-nv/jetson-containers/commit/5d0f8b47b9b9ef986d88b14693805feec8cdf41a

closed 2022-02-17 by canberkgurel