dusty-nv/jetson-containers #125: bash: /usr/bin/nvidia-smi: No such file or directory

an99990, 2022-02-18:
docker run --gpus all -it --ipc=host --ulimit memlock=-1 --ulimit stack=67108864 --rm -v $(pwd):/opt/ -w /opt/ nvidiajetson/l4t-ros2-foxy-pytorch:r32.5

the docker doesnt seem to have torch either, is it normal ?
[image]

I was looking for a container with torch and ros2 foxy or galactic

dusty-nv, 2022-02-18:
PyTorch would be installed in python3, not python2, so try running python3 instead.

You can also try [dustynv/ros:foxy-pytorch-l4t-r32.5.0](https://hub.docker.com/layers/dustynv/ros/foxy-pytorch-l4t-r32.5.0/images/sha256-099d55e77c3fbcd8ebefcb7354c6f467953d28e05e2d619b28c7fa1e703909b9?context=repo) container

an99990, 2022-02-18:
hi thank you
i just tried `docker run --gpus all -it --ipc=host --ulimit memlock=-1 --ulimit stack=67108864 --rm -v $(pwd):/opt/ -w /opt/ dustynv/ros:foxy-pytorch-l4t-r32.5.0`

and nvidia-smi still doesnt work

```
WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested
sourcing   /opt/ros/foxy/install/setup.bash
ROS_ROOT   /opt/ros/foxy
ROS_DISTRO foxy
root@8bc8ca60c015:/opt/halodi# nvidia-smiu
bash: nvidia-smiu: command not found
root@8bc8ca60c015:/opt/halodi# nvidia-smi
bash: /usr/bin/nvidia-smi: No such file or directory
root@8bc8ca60c015:/opt/halodi# python --version
Python 3.6.9
root@8bc8ca60c015:/opt/halodi# python
Python 3.6.9 (default, Oct  8 2020, 12:12:24)
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 188, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 141, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
```

an99990, 2022-02-18:
thank you, I am currently testing on my Windows machine. How can i test if the gpu is accesed in the docker ?
@dusty-nv

dusty-nv, 2022-02-18:
These containers are for Jetson and they will only work on ARM/aarch64

an99990, 2022-02-21:
hi so im in my jetson xavier and i got this error 
```
docker run dustynv/ros:foxy-pytorch-l4t-r32.5.0 
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.24/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'
```

an99990, 2022-02-22:
so im able to import torch with the following command, 

`docker run --gpus all -it --ipc=host --ulimit memlock=-1 --ulimit stack=67108864 --rm -v $(pwd):/opt/ -w /opt/ dustynv/ros:foxy-pytorch-l4t-r32.5.0`

how do i make sure it is run on cuda and not cpu ?

dusty-nv, 2022-02-22:
The normal way that you run stuff on GPU with cuda - call `.cuda()` on the tensors, model(s), loss function, ect

You can also run this test script which will enumerate the GPU support in PyTorch:  https://github.com/dusty-nv/jetson-containers/blob/master/test/test_pytorch.py

an99990, 2022-02-22:
hi @dusty-nv it seems that the docker contains an old version of pytorch, where can i find a PyTorch >= 1.8

dusty-nv, 2022-02-22:
If you upgrade to JetPack 4.6 (L4T R32.6.1), then `dustynv/ros:foxy-pytorch-l4t-r32.6.1` should have PyTorch 1.9 in it

That, or you can install newer PyTorch wheel [from here](https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-10-now-available/72048) in container yourself

an99990, 2022-02-22:
i believe i amusing latest jetpack

```
cat /etc/nv_tegra_release 
# R32 (release), REVISION: 6.1, GCID: 27863751, BOARD: t186ref, EABI: aarch64, DATE: Mon Jul 26 19:36:31 UTC 2021
```
so i was able to pull the mentionned docker.
thank you

closed 2022-02-22 by an99990