dusty-nv/jetson-containers #126: dustynv/ros:foxy-pytorch-l4t-r32.5.0 is missing libcurand.so.10 etc

jamestkpoon, 2022-02-22:
Hi!

I'm running the following command on a Jetson AGX Xavier, using an image pulled via docker:
```
docker run --gpus all --runtime nvidia -it --ipc=host --ulimit memlock=-1 --ulimit stack=67108864 --rm dustynv/ros:foxy-pytorch-l4t-r32.5.0
```
It loads successfully, but when I open python3 and try ```import torch```  I get the following trace:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 188, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 141, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
```
Output of ```ls /usr/local/cuda-10.2/targets/aarch64-linux/lib/``` is
```
libcudadevrt.a  libcudart_static.a  stubs
```
so there is some CUDA stuff, but not CuDNN etc. Are there any prerequisite steps I could have missed?
Thanks in advance!

Additional info:
From other message boards, my output of ```cat /etc/nv_tegra_release``` outside docker is 
```
# R32 (release), REVISION: 3.1, GCID: 18284527, BOARD: t186ref, EABI: aarch64, DATE: Mon Dec 16 21:38:34 UTC 2019
```
so may I assume there actually is an existing (outdated?) Jetpack installation?

dusty-nv, 2022-02-22:
> R32 (release), REVISION: 3.1

Yes, you are on L4T R32.3.1 (JetPack 4.3), but the container you are running is for L4T R32.5.0 (JetPack 4.5).   I recommend upgrading to a more recent version of JetPack.

an99990, 2022-02-22:
i was able to import torch with this jetpack image https://developer.nvidia.com/embedded/jetpack
```
cat /etc/nv_tegra_release 
# R32 (release), REVISION: 6.1, GCID: 27863751, BOARD: t186ref, EABI: aarch64, DATE: Mon Jul 26 19:36:31 UTC 2021
```

jamestkpoon, 2022-02-23:
Thank you both for your quick replies. my output of ```cat /etc/nv_tegra_release``` is now
```
# R32 (release), REVISION: 6.1, GCID: 27863751, BOARD: t186ref, EABI: aarch64, DATE: Mon Jul 26 19:36:31 UTC 2021
```

but the problem persists.

jfultr, 2022-02-24:
Hello! I have same traceback with running `./scripts/docker_build_ml.sh pytorch`. 
```
Traceback (most recent call last):
  File "setup.py", line 12, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 196, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 149, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in init
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
```
I using `JetPack 4.6`, some log from script:
```
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.6.1
l4t-base image:  nvcr.io/nvidia/l4t-base:r32.6.1
building PyTorch torch-1.9.0-cp36-cp36m-linux_aarch64.whl, torchvision v0.10.0 (pillow), torchaudio v0.9.0
Building l4t-pytorch:r32.6.1-pth1.9-py3 container...
```
Error raised after this step:
```
Step 15/28 : RUN git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision
```
**I noticed that the error can be repeated with other repositories that are installed via the `python3 setup.py install` command, and there is always a problem with the `libcurand.so.10` library.**
Initially, I wanted to build Dockerfile with cloning and `python3 setup.py install` `https://github.com/JunnYu/mish-cuda` repository, but I got this error.
Also, before that, I upgraded the system more than once and downgraded docker version

jamestkpoon, 2022-02-24:
Closing this issue. I had to manually reinstall the SDK components via the sdk manager program and then it all works. Sorry for the erroneous issue, but it was a bit frustrating when it didn’t work out of the box ;)

closed 2022-02-24 by jamestkpoon