dusty-nv/jetson-containers #127: python 3.6.9

an99990, 2022-02-23:
Hi,
So i have successfully run the dustynv/ros:galactic-pytorch-l4t-r32.6.1 docker and notice that the python version is 3.6.9.
Is there a docker with new python or should i just upgrade the python version ? 

thank you

dusty-nv, 2022-02-23:
Hi @an99990, simply upgrading the Python version would lose the Python packages for PyTorch, ROS, ect since those are installed for Python 3.6.   You would need to rebuild the PyTorch wheel for Python 3.9, and then rebuild the l4t-pytorch, jetson-inference, and ROS containers for Python 3.9.  So all in all, a significant effort.

Python 3.6 is used in these containers because they are based on Ubuntu 18.04 (the version of Ubuntu for JetPack 4.x) which has a default Python version of 3.6.   When JetPack 5.x comes out, the default Python version will change to Python 3.8.

closed 2022-02-23 by an99990

JonathanNash21, 2022-10-22:
If I absolutely need a Python version that is 3.7 or higher, is there somewhere I can look to so that I properly rebuild all of the items needed? Or would simply editing the galactic pytorch dockerfile to install python 3.7 and have that set as default before all of the other packages are install work?

dusty-nv, 2022-10-24:
> Or would simply editing the galactic pytorch dockerfile to install python 3.7 and have that set as default before all of the other packages are install work?

Hi @JonathanNash21, assuming you are on JetPack 4.x, then yes I would install the libpython3.7-dev packages in the container first and then set python3.7 as the default, before galactic gets built.

Note that this won't work for the pytorch stuff, because the pytorch wheels are all built for Python 3.6 on JetPack 4.x.  You would need to independently rebuild the pytorch wheel for Python 3.7, and then rebuild l4t-pytorch with that wheel and Python 3.7 instead.

If you're on JetPack 5.x, the default is already Python 3.8.