dusty-nv/jetson-containers #131: Latest ROS2 Foxy container has RMW error

Hermanoid, 2022-03-03:
We were having success using the dustynv/ros:foxy-ros-base-l4t-r32.6.1 image off GitHub.  Today, we pulled "044612a74613" - pushed eight hours ago - and saw this error appear:

`hermanoid@lucas-is-shnazzy:~/mate$ docker run -it dustynv/ros:foxy-ros-base-l4t-r32.6.1 bash
WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested
sourcing   /opt/ros/foxy/install/setup.bash
ROS_ROOT   /opt/ros/foxy
ROS_DISTRO foxy
root@145147775ffa:/# ros2 run demo_nodes_py talker
Unsupported setsockopt level=0 optname=32
1646269244.734822 [0]     talker: ddsi_udp_create_conn: set IP_MULTICAST_IF failed: Bad Parameter
[ERROR] [1646269244.739727198] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'error not set, at /opt/ros/foxy/src/rcl/rcl/src/rcl/node.c:277'

with this new error message:

  'rcl node's rmw handle is invalid, at /opt/ros/foxy/src/rcl/rcl/src/rcl/node.c:429'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1646269244.744252497] [rcl]: Failed to fini publisher for node: 1
Traceback (most recent call last):
  File "/opt/ros/foxy/install/lib/demo_nodes_py/talker", line 33, in <module>
    sys.exit(load_entry_point('demo-nodes-py==0.9.3', 'console_scripts', 'talker')())
  File "/opt/ros/foxy/install/lib/python3.6/site-packages/demo_nodes_py/topics/talker.py", line 41, in main
    node = Talker()
  File "/opt/ros/foxy/install/lib/python3.6/site-packages/demo_nodes_py/topics/talker.py", line 24, in __init__
    super().__init__('talker')
  File "/opt/ros/foxy/install/lib/python3.6/site-packages/rclpy/node.py", line 170, in __init__
    enable_rosout
_rclpy.RCLError: Unknown error creating node: rcl node's rmw handle is invalid, at /opt/ros/foxy/src/rcl/rcl/src/rcl/node.c:429`

This article implies that a compile flag is missing:
https://github.com/ros2/sros2/issues/166

dusty-nv, 2022-03-03:
Not sure, but it may be related to https://github.com/ros2/rmw_cyclonedds/issues/273

ros2/sros2#166<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fros2%2Fsros2%2Fissues%2F166&data=04%7C01%7Cdustinf%40nvidia.com%7C94d6c6e8d41f4c6442dd08d9fcb27bd7%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C637818665777907021%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=ms97a%2FNXFukBaBjWU1asW4O6dnL4beTZIS9PQrjrDP8%3D&reserved=0>

dusty-nv, 2022-03-03:
Hi @Hermanoid, I'm unable to replicate this.   As per this warning from your log, it appears you are trying to run the container on x86?   These containers are built for arm64.

``` bash
hermanoid@lucas-is-shnazzy:~/mate$ docker run -it dustynv/ros:foxy-ros-base-l4t-r32.6.1 bash
WARNING: The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested
```

Hermanoid, 2022-03-03:
Yeah!  We're cross building on our Linux laptops and running them locally to test them.  After some testing this morning, I think there are some network emulation troubles with the QEMU emulator that docker uses. I got a more direct error with the fastrtps middleware - "Unsupported setsockopt level=0 optname=32"

I can't run the container on the actual Jetson just yet, but I'll post an update once I do.

dusty-nv, 2022-03-03:
OK yea - I run these on actual Jetson's.  You will run into more problems with the GPU stuff and loading any libraries/modules that depend on CUDA/ect if you try to emulate it.

Hermanoid, 2022-03-04:
Yep - just confirmed that this is 100% a problem with QEMU emulation. Everything works swimmingly on an actual amd64 device.

closed 2022-03-04 by Hermanoid

mohanrobotics, 2024-03-20:
@Hermanoid Did you find a solution to emulate this on x86 ?

Hermanoid, 2024-03-26:
No, unfortunately. There might be one out there if you care to dig into QEMU and open that can of worms. We decided it wasn't worth the effort and only build on the actual hardware. Actually, that's not quite true - as a workaround, we maintain both an x86 and an amd64 image so we can use Github Actions to run test builds.
If you want to check that setup out, it's open source: [https://github.com/Colorado-School-of-Mines-Robotics-Club/MATE_ROV_Programming](https://github.com/Colorado-School-of-Mines-Robotics-Club/MATE_ROV_Programming)

haolei523, 2024-04-13:
I 'm running `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp` , solved this problem, but topics can not received. 
[image]

DZZLab, 2025-02-16:
> I 'm running `export RMW_IMPLEMENTATION=rmw_fastrtps_cpp` , solved this problem, but topics can not received. [image]

Did you ever find a workaround for this?