dusty-nv/jetson-containers #132: Foxy build  for Jetson TX2 fails

jarekkt, 2022-03-11:
I have tried to build from source - as I hope to get latest Foxy release as I could not find info what version was in your docker images .
We are facing serious problems due to broke multithreaded executors ( this causes huge empty load killing our  system), I was hoping that latest release might have fixed the issue.

Full log attached, but briefly :

Finished <<< rosidl_typesupport_introspection_c [11.0s]
Starting >>> rosidl_typesupport_introspection_cpp
--- stderr: libyaml_vendor
Cloning into 'libyaml-0.2.5'...
HEAD is now at 2c891fc Changes for v0.2.5 release
error: can't open patch '/opt/ros/foxy/src/libyaml_vendor/0001-Install-yaml.h-to-INSTALL_INCLUDE_DIR.patch': No such file or directory
make[2]: *** [libyaml-0.2.5-prefix/src/libyaml-0.2.5-stamp/libyaml-0.2.5-patch] Error 128
make[1]: *** [CMakeFiles/libyaml-0.2.5.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< libyaml_vendor [16.3s, exited with code 2]
Aborted  <<< rosidl_typesupport_introspection_cpp [7.69s]
Aborted  <<< rosidl_generator_c [32.2s]
Aborted  <<< rcl_logging_spdlog [1min 28s]

Aborted  <<< rmw [2min 58s]
^AAborted  <<< fastrtps [17min 49s]

Summary: 93 packages finished [21min 9s]
  1 package failed: libyaml_vendor
  5 packages aborted: fastrtps rcl_logging_spdlog rmw rosidl_generator_c rosidl_typesupport_introspection_cpp
  27 packages had stderr output: ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pep257 ament_uncrustify ament_xmllint cyclonedds domain_coordinator fastcdr foonathan_memory_vendor launch launch_testing launch_xml launch_yaml libyaml_vendor mimick_vendor orocos_kdl osrf_pycommon rcutils rpyutils urdfdom_

headers
  118 packages not processed

[buildlog.txt](https://github.com/dusty-nv/jetson-containers/files/8230750/buildlog.txt)

dusty-nv, 2022-03-11:
`error: can't open patch '/opt/ros/foxy/src/libyaml_vendor/0001-Install-yaml.h-to-INSTALL_INCLUDE_DIR.patch': No such file or directory`

Hi @jarekkt, TBH I'm not sure why this error just started, but it appears that this `0001-Install-yaml.h-to-INSTALL_INCLUDE_DIR.patch` file was added/modified just a few days ago:

https://github.com/ros2/libyaml_vendor

Would you please file this issue against the libyaml_vendor repo so they know about it?

jarekkt, 2022-03-13:
Done.
https://github.com/ros2/libyaml_vendor/issues/47
Lets see...

jarekkt, 2022-03-14:
Ok, fresh morning and coffe showed this:

    # patch libyaml - https://github.com/dusty-nv/jetson-containers/issues/41#issuecomment-774767272
    rm ${ROS_ROOT}/src/libyaml_vendor/CMakeLists.txt && \
    wget --no-check-certificate https://raw.githubusercontent.com/ros2/libyaml_vendor/master/CMakeLists.txt -P ${ROS_ROOT}/src/libyaml_vendor/ && \
    
You patch just  foxy branch from main. But the latest file contains new reference ( the one which fails). So you would have to add the one . Or check if the patch is still needed. I will tru and let you know

jarekkt, 2022-03-14:
I was able to succesfully build the ROS 2 - when removing the libyaml patch completely.

jarekkt, 2022-03-14:
Nice, thanks.