dusty-nv/jetson-containers #133: Deepstream support on noetic-ros-base

rishabhkjain, 2022-03-16:
Is the deepstream SDK supported on the `noetic-ros-base` package? 

Generally we want to run YOLOv3 on a Jetson AGX Xavier and from the nvidia forums it seems the best way to do so will be using the Deepstream SDK. I saw that there seems to be some [docker containers](https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_docker_containers.html) for running Deepstream on a Jetson but I have not found anything about a setup with the following components

- AGX Xavier 
- Jetpack 4.6 (or 4.6.1) 
- ROS Noetic 
- Deepstream SDK 

Alternatively, using YOLO with Darknet seems to be a bit easier to setup with a prexisting ROS package [darknet_ros](https://github.com/leggedrobotics/darknet_ros). Do you have any insights on how much better the performance will be with the Deepstream SDK on our platform?

dusty-nv, 2022-03-16:
Hi @rishabhkjain, deepstream isn't installed into my Noetic container, no.  What I would try in your situation is changing the base container of it to the deepstream container, and rebuilding.

rishabhkjain, 2022-03-17:
@dusty-nv Thanks, will do. In order to speed up the build, can we follow this process to build on a x86 platform and then run the resulting container on the Jetson? https://github.com/NVIDIA/nvidia-docker/wiki/NVIDIA-Container-Runtime-on-Jetson#building-jetson-containers-on-an-x86-workstation-using-qemu

closed 2025-02-21 by johnnynunez