dusty-nv/jetson-containers #134: Intel Relasense not working in the foxy-slam-l4t-r32.6.1

Brac24, 2022-03-17:
Hello,

I am using a Jetson Nano to experiment with ROS 2 and SLAM using an Intel Realsense camera. I have a D435i and D455 that I am using but when I run the docker container I see an error that says: 
not found: "/opt/ros/foxy/install/share/librealsense2/local_setup.bash"

When I try to run the realsense2_camera_node I get an error saying failed to set power state.

[image]

Not sure if this is because the SDK is not installed in the container. Usually I am able to run realsense example programs when the SDK is installed like rs-depth. I have the SDK installed on the Jetson Nano itself and it works well with no issues when I run it there.

I pulled the dustynv/ros:foxy-slam-l4t-r32.6.1 image from docker hub. I was looking at the Dockerfile.ros.slam file which I assumed is what the image was based on and it doesn't seem to install the realsense SDK. I am not sure how to go about doing that in the dockerfile. 

Any help is greatly appreciated thank you.

dusty-nv, 2022-03-18:
Hi @Brac24, I don't really maintain the slam dockerfiles anymore due to the complexity of the dependencies and continuous breakages in those packages - I should probably mark them for removal.  However if you are interested, you should be able to install the realsense SDK in the dockerfile pretty much the same way that you would otherwise (just in a docker `RUN` statement in the dockerfile).  I also don't have a realsense device to test it with - sorry to not be of more help.

Brac24, 2022-03-18:
No worries. Thank you @dusty-nv .
I think I was able to get it working.

One more question though. Do you know if there's a way to build the docker images on a different platform like a more powerful PC to then run on the Jetson nano?

It's just that building the docker images on the nano takes about 6 hours to complete and one little change to the docker file causes it to build from scratch again just to add the SDK.

closed 2025-02-21 by johnnynunez