dusty-nv/jetson-containers #138: Are there desktop images for ROS2 Jetson?

sisaha9, 2022-04-03:
I see there are images for the ros_base and such but none for the Desktop versions of ROS in Noetic, Foxy and Galactic. Are there plans to release prebuilt containers for these distros?

dusty-nv, 2022-04-04:
Hi @sisaha9, IIRC there was some trouble building the full desktop packages so I don't have the pre-built images for them.

sisaha9, 2022-04-11:
Hi @dusty-nv , now that Jetpack 5.0 is out is it possible to create Desktop images for the above distros? I haven't seen a Jetpack 5.0 l4t-base image so I am unable to attempt it

dusty-nv, 2022-04-18:
In theory it should be possible, although I haven't migrated my ros2 dockerfiles over to installing OSRF's focal packages and am still building from source.   l4t-base is up here:  `docker pull nvcr.io/nvidia/l4t-base:r34.1`

brianlmerritt, 2022-06-01:
@sisaha9 - start with the container that has everything you want apart from ROS and then install that ROS distro requirements and then do a full desktop install.  Will probably have to be build from scratch unless your container is Ubuntu 20.04 (e.g. newer JetPack on Orin or Xavier)

closed 2025-02-21 by johnnynunez