dusty-nv/jetson-containers #140: ValueError: Unknown CUDA arch () or GPU not supported

jonpol01, 2022-04-12:
> Traceback (most recent call last):
  File "setup.py", line 488, in <module>
    'clean': clean,
  File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py", line 74, in run
    self.do_egg_install()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install.py", line 116, in do_egg_install
    self.run_command('bdist_egg')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/bdist_egg.py", line 164, in run
    cmd = self.call_command('install_lib', warn_dir=0)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/bdist_egg.py", line 150, in call_command
    self.run_command(cmdname)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/install_lib.py", line 11, in run
    self.build()
  File "/usr/lib/python3.6/distutils/command/install_lib.py", line 109, in build
    self.run_command('build_ext')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/usr/local/lib/python3.6/dist-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 709, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.6/dist-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/usr/lib/python3.6/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 457, in unix_wrap_single_compile
    cflags = unix_cuda_flags(cflags)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 424, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 1575, in _get_cuda_arch_flags
    raise ValueError(f"Unknown CUDA arch ({arch}) or GPU not supported")
ValueError: Unknown CUDA arch () or GPU not supported
The command '/bin/sh -c git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision' returned a non-zero code: 1

Need help on this error.. 
tried the command `./scripts/docker_build_ml.sh pytorch `this on a fresh Xavier NX Jetpack 4.6 [L4T 32.6.1].

Things did already
1. extra. [bootfromexternalstorage](https://github.com/jetsonhacks/bootFromExternalStorage) 
2. clone this repo, [default-runtime-nvidia](https://github.com/jetsonhacks/bootFromExternalStorage). done

kevin00730, 2022-05-05:
Same error on NVIDIA Tegra X2
Below are entire build logs

> root@tegra-ubuntu:/usr/sdcard/jetson-containers# ./scripts/docker_build_ml.sh all
> reading L4T version from /etc/nv_tegra_release
> L4T BSP Version:  L4T R32.6.1
> l4t-base image:  nvcr.io/nvidia/l4t-base:r32.6.1
> selecting OpenCV for L4T R32.6.1...
> OPENCV_URL=https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz
> OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
> Python3 version:  3.6
> building PyTorch torch-1.9.0-cp36-cp36m-linux_aarch64.whl, torchvision v0.10.0, torchaudio v0.9.0, cuda arch 5.3;6.2;7.2;
> Building l4t-pytorch:r32.6.1-pth1.9-py3 container...
> Sending build context to Docker daemon  178.2kB
> Step 1/26 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.4.4
> Step 2/26 : FROM ${BASE_IMAGE}
>  ---> 1735192d3d51
> Step 3/26 : ENV DEBIAN_FRONTEND=noninteractive
>  ---> Using cache
>  ---> ca9a5a9c6c8f
> Step 4/26 : RUN apt-get update &&     apt-get install -y --no-install-recommends             python3-pip 		  python3-dev 		  libopenblas-dev 		  libopenmpi-dev             openmpi-bin             openmpi-common 		  gfortran 		  libomp-dev     && rm -rf /var/lib/apt/lists/*     && apt-get clean
>  ---> Using cache
>  ---> b496d92f9383
> Step 5/26 : RUN pip3 install --no-cache-dir setuptools Cython wheel
>  ---> Using cache
>  ---> 47674f2963d0
> Step 6/26 : RUN pip3 install --no-cache-dir --verbose numpy
>  ---> Using cache
>  ---> d651682d906f
> Step 7/26 : ARG PYTORCH_URL=https://nvidia.box.com/shared/static/lufbgr3xu2uha40cs9ryq1zn4kxsnogl.whl
>  ---> Using cache
>  ---> 781a2295e37c
> Step 8/26 : ARG PYTORCH_WHL=torch-1.2.0-cp36-cp36m-linux_aarch64.whl
>  ---> Using cache
>  ---> 9bf0ca72e1ef
> Step 9/26 : RUN wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${PYTORCH_URL} -O ${PYTORCH_WHL} &&     pip3 install --no-cache-dir --verbose ${PYTORCH_WHL} &&     rm ${PYTORCH_WHL}
>  ---> Using cache
>  ---> 5187b58092dc
> Step 10/26 : ARG TORCHVISION_VERSION=v0.4.0
>  ---> Using cache
>  ---> a280157bebca
> Step 11/26 : ARG TORCH_CUDA_ARCH_LIST="5.3;6.2;7.2;8.7"
>  ---> Using cache
>  ---> 10064ce27a0d
> Step 12/26 : RUN printenv && echo "torchvision version = $TORCHVISION_VERSION" && echo "TORCH_CUDA_ARCH_LIST = $TORCH_CUDA_ARCH_LIST"
>  ---> Using cache
>  ---> f5c07a5d26f5
> Step 13/26 : RUN apt-get update &&     apt-get install -y --no-install-recommends 		  git 		  build-essential             libjpeg-dev 		  zlib1g-dev     && rm -rf /var/lib/apt/lists/*     && apt-get clean
>  ---> Using cache
>  ---> 73685779ae15
> Step 14/26 : RUN git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision
>  ---> Running in abd91d638457
> Cloning into 'torchvision'...
> Note: checking out '300a8a40132c5d3ee7dc936034c6046fd30ca4c0'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by performing another checkout.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -b with the checkout command again. Example:
> 
>   git checkout -b <new-branch-name>
> 
> Building wheel torchvision-0.10.0a0+300a8a4
> PNG found: False
> Running build on conda-build: False
> Running build on conda: False
> JPEG found: True
> Building torchvision with JPEG image support
[... 359 lines cut ...]
>   File "/usr/lib/python3.6/distutils/command/build_ext.py", line 339, in run
>     self.build_extensions()
>   File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 709, in build_extensions
>     build_ext.build_extensions(self)
>   File "/usr/local/lib/python3.6/dist-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
>     _build_ext.build_ext.build_extensions(self)
>   File "/usr/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
>     self._build_extensions_serial()
>   File "/usr/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
>     self.build_extension(ext)
>   File "/usr/local/lib/python3.6/dist-packages/setuptools/command/build_ext.py", line 202, in build_extension
>     _build_ext.build_extension(self, ext)
>   File "/usr/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
>     depends=ext.depends)
>   File "/usr/lib/python3.6/distutils/ccompiler.py", line 574, in compile
>     self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
>   File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 457, in unix_wrap_single_compile
>     cflags = unix_cuda_flags(cflags)
>   File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 424, in unix_cuda_flags
>     cflags + _get_cuda_arch_flags(cflags))
>   File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 1575, in _get_cuda_arch_flags
>     raise ValueError(f"Unknown CUDA arch ({arch}) or GPU not supported")
> ValueError: Unknown CUDA arch () or GPU not supported
> The command '/bin/sh -c git clone -b ${TORCHVISION_VERSION} https://github.com/pytorch/vision torchvision &&     cd torchvision &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision' returned a non-zero code: 1
> root@tegra-ubuntu:/usr/sdcard/jetson-containers#

jonpol01, 2022-05-05:
> 
@dusty-nv 
oh right sorry for the delay, but I think I have the same build logs with @kevin00730 
and also I already moveon onto just using the `Docker.slam.ros foxy --with-slam` instead since I need Pytorch + Ros2 than Tensorflow atm.

jonpol01, 2022-05-05:
@dusty-nv 
Hi, thanks that was an easy fix.. I coulve have got that if I just looked to it more.

anyway tried `docker_build_ml.sh pytorch` and got this at the end..

>  it is not compatible with this Python
    Found link https://files.pythonhosted.org/packages/85/e5/74fcd803fe3a32e8bed53d51889313089124c3ab8fa2a6eeab5cbfba29ab/cmake-3.22.4.tar.gz#sha256=d408349597bf3b9f3ad0364aaed8c55f50dfa011b396d299a7cabf3c799510af (from https://pypi.org/simple/cmake/), version: 3.22.4
  Using version 3.22.3 (newest of versions: 3.22.3)
  Starting new HTTPS connection (1): files.pythonhosted.org
  https://files.pythonhosted.org:443 "GET /packages/8b/76/6ac2819c82af2d37ec123e56ab67396aaf76652a60858d9e83e8db787fe1/cmake-3.22.3.tar.gz HTTP/1.1" 200 31557
  Downloading https://files.pythonhosted.org/packages/8b/76/6ac2819c82af2d37ec123e56ab67396aaf76652a60858d9e83e8db787fe1/cmake-3.22.3.tar.gz
  Downloading from URL https://files.pythonhosted.org/packages/8b/76/6ac2819c82af2d37ec123e56ab67396aaf76652a60858d9e83e8db787fe1/cmake-3.22.3.tar.gz#sha256=6e1d1991775915dac047af851f2feec7eb35ab4b4610d3e7fd94d93bce8faf58 (from https://pypi.org/simple/cmake/)
  Running setup.py (path:/tmp/pip-build-16c13zqc/cmake/setup.py) egg_info for package cmake
    Running command python setup.py egg_info
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-16c13zqc/cmake/setup.py", line 8, in <module>
        from skbuild import setup
    ModuleNotFoundError: No module named 'skbuild'
Cleaning up...
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-16c13zqc/cmake/
Exception information:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 342, in run
    requirement_set.prepare_files(finder)
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 634, in _prepare_file
    abstract_dist.prep_for_dist()
  File "/usr/lib/python3/dist-packages/pip/req/req_set.py", line 129, in prep_for_dist
    self.req_to_install.run_egg_info()
  File "/usr/lib/python3/dist-packages/pip/req/req_install.py", line 439, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/usr/lib/python3/dist-packages/pip/utils/__init__.py", line 725, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-16c13zqc/cmake/
The command '/bin/sh -c pip3 install --upgrade --no-cache-dir --verbose cmake==3.22.3' returned a non-zero code: 1

Prior to that plenty of "is not compatible with this Python" came up and skipped most of the command.
I believe this doesnt even started building stuffs and ended on Step 16.

> reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.6.1
l4t-base image:  nvcr.io/nvidia/l4t-base:r32.6.1
selecting OpenCV for L4T R32.6.1...
OPENCV_URL=https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
Python3 version:  3.6
building PyTorch torch-1.9.0-cp36-cp36m-linux_aarch64.whl, torchvision v0.10.0, torchaudio v0.9.0, cuda arch 5.3;6.2;7.2
Building l4t-pytorch:r32.6.1-pth1.9-py3 container...
Sending build context to Docker daemon  184.3kB
Step 1/26 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.4.4
Step 2/26 : FROM ${BASE_IMAGE}
 ---> 1735192d3d51
Step 3/26 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> 85597bf48657
Step 4/26 : RUN apt-get update &&     apt-get install -y --no-install-recommends             python3-pip                  python3-dev             libopenblas-dev                 libopenmpi-dev             openmpi-bin             openmpi-common               gfortran          libomp-dev     && rm -rf /var/lib/apt/lists/*     && apt-get clean
 ---> Using cache
 ---> 90cc60fa245c
Step 5/26 : RUN pip3 install --no-cache-dir setuptools Cython wheel
 ---> Using cache
 ---> b2935c419e7f
Step 6/26 : RUN pip3 install --no-cache-dir --verbose numpy
 ---> Using cache
 ---> 4c1a708422d6
Step 7/26 : ARG PYTORCH_URL=https://nvidia.box.com/shared/static/lufbgr3xu2uha40cs9ryq1zn4kxsnogl.whl
 ---> Using cache
[... 4 lines cut ...]
Step 9/26 : RUN wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${PYTORCH_URL} -O ${PYTORCH_WHL} &&     pip3 install --no-cache-dir --verbose ${PYTORCH_WHL} &&     rm ${PYTORCH_WHL}
 ---> Using cache
 ---> 9c9bfc177980
Step 10/26 : ARG TORCHVISION_VERSION=v0.4.0
 ---> Using cache
 ---> 729c813ccf7c
Step 11/26 : ARG TORCH_CUDA_ARCH_LIST="5.3;6.2;7.2;8.7"
 ---> Using cache
 ---> 0b1cf6b073bd
Step 12/26 : RUN printenv && echo "torchvision version = $TORCHVISION_VERSION" && echo "TORCH_CUDA_ARCH_LIST = $TORCH_CUDA_ARCH_LIST"
 ---> Using cache
 ---> c041fcfd3bcb
Step 13/26 : RUN apt-get update &&     apt-get install -y --no-install-recommends                 git             build-essential             libjpeg-dev                 zlib1g-dev     && rm -rf /var/lib/apt/lists/*     && apt-get clean
 ---> Using cache
 ---> 47bdf95b75f1
[... cut ...]
Step 15/26 : RUN apt-get update &&     apt-get install -y --no-install-recommends                 software-properties-common              apt-transport-https             ca-certificates                 gnupg                   lsb-release     && rm -rf /var/lib/apt/lists/*     && apt-get clean
 ---> Using cache
 ---> 737df89eb3d7
Step 16/26 : RUN pip3 install --upgrade --no-cache-dir --verbose cmake==3.22.3
 ---> Running in 47dba072d800

I have only been doin ROS2 and jupyterlab added on the ROS2 docker, so I think my Host (Xavier NX) is still quite default.

arpaterson, 2022-06-01:
> 

I just experienced this 'no module named skbuild' error too. on nano, JP r32.5.1.
I updated jetson-containers and am trying again.
Should this be its own issue?

gonzarub, 2022-06-01:
> 

I've been struggling trying to build the full ML container (PyTorch1.x + TF2.7), and one of the first issues I fixed was the "no module named skbuild". I did the following `(1)`  to fix it from Dockerfile.pytorch :

`
# (2) https://stackoverflow.com/a/66057286
ENV OPENBLAS_CORETYPE=ARMV8

RUN apt-get update && \
    apt-get install -y --no-install-recommends \
            python3-pip \
		  python3-dev \
		  libopenblas-dev \
		  libopenmpi-dev \
            openmpi-bin \
            openmpi-common \
		  gfortran \
		  libomp-dev \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean

# (1) Insert this:
RUN pip3 install --upgrade pip
`
I figured that out after reading this: https://github.com/opencv/opencv-python#frequently-asked-questions

Notice fix `(2)` is not related to this, but helped to fix an error, also related to the stackoverflow link in the comment.

Hope that helps

- Ruben

gonzarub, 2022-06-02:
@arpaterson 
I also found that one. If you look carefully to the single whl file found in the following URL, it is not the same referenced in the build script:
https://developer.download.nvidia.com/compute/redist/jp/v461/pytorch/

Of course, I replaced it, and I got to do some more fixes to do [NOTE 1] until I gave up with the master branch. I'm not on PyTorch, and I only wanted a working ML container, so ended up using the last stable branch for L4T 32.7.1 (which is my current version).

I have to say, even that branch required a small fix in the docker_base.sh script.

[NOTE 1] This is what I did before giving up with the master branch:

- After changing that path to the torch whl file, I found [this link](https://qengineering.eu/install-pytorch-on-jetson-nano.html) saying that PyTorch 1.11 requires Python 3.7, so I stopped trying this version.

- Back to PyTorch 1.10, I detected an invalid apt package (libopenmpi3 had to be replaced with libopenmpi-dev).

- And finally, when I thought it was going to work, I got a bunch of Cython-related warnings and an error while building cupy.

          aarch64-linux-gnu-g++ -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE\=2 build/temp.linux-aarch64-3.6/cupy/cudnn.o -L/usr/local/cuda/lib64 -lcudnn -o build/lib.linux-aarch64-3.6/cupy/cudnn.cpython-36m-aarch64-linux-gnu.so -Wl,--disable-new-dtags,-rpath,/usr/local/cuda/lib64
          ^[[91mwarning: cupy/cuda/memory.pyx:119:4: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:495:9: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:539:9: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:697:9: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:713:9: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:728:9: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:739:9: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:754:10: Unreachable code
          warning: cupy_backends/cuda/api/runtime.pyx:773:4: Unreachable code
          warning: cupy_backends/cuda/libs/nvrtc.pyx:87:9: Unreachable code
          warning: cupy_backends/cuda/libs/nvrtc.pyx:183:9: Unreachable code
          ^[[0m^[[91mTraceback (most recent call last):
            File "setup.py", line 149, in <module>
              cmdclass={'build_ext': cupy_builder._command.custom_build_ext},
            File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 153, in setup
              return distutils.core.setup(**attrs)
         .....
        File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
            raise CalledProcessError(retcode, cmd)
        subprocess.CalledProcessError: Command '['/usr/local/cuda/bin/nvcc', '-D_FORCE_INLINES=1', '-DCUPY_CUB_VERSION_CODE=100800', '-DCUPY_JITIFY_VERSION_CODE=4a37de0', '-I/cupy/cupy/_core/include/cupy/cub', '-I/cupy/cupy/_core/include', '-I/us\r/local/cuda/include', '-c', 'cupy/cuda/cupy_cub.cu', '-o', 'build/temp.device_objects/cupy/cuda/cupy_cub.cu.o', '--generate-code=arch=compute_53,code=sm_53', '--generate-code=arch=compute_62,code=sm_62', '--generate-code=arch=compute_72,\code=sm_72', '--generate-code=arch=compute_87,code=sm_87', '-O2', '--compiler-options="-fPIC"', '--std=c++11', '-Xcompiler=-fno-gnu-unique']' returned non-zero exit status 1.

YuJuChiu, 2022-11-02:
> 

Hello, did you manage the issue?
I face the exact same problem as yours

YuJuChiu, 2022-11-02:
@gonzarub 
Hello, did you manage the issue? I face the exact same problem as yours

closed 2025-02-21 by johnnynunez