dusty-nv/jetson-containers #141: JP 5

felrock, 2022-04-12:
I saw that the developer preview is out for JetPack 5.0, will we see a container for this any time soon? Thanks

dusty-nv, 2022-04-18:
Hi @felrock, the l4t-pytorch container has been pushed to NGC awhile:  https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-pytorch

The l4t-ml and l4t-tensorflow containers should be pushed shortly.
The jetson-inference container is already pushed to dockerhub.
Working on the ROS containers now.

asimonov, 2022-04-19:
Dustin, 

Thank you for updating this repo for JP5!

I have managed to build my images based on L4T 34.1 base image from the hub and using your scripts for the rest of stuff as an outline. All builds fine on jetpack 5.
But starting the container throws errors.

I added details into this thread: 
https://forums.developer.nvidia.com/t/error-running-privileged-l4t-containers-with-the-nvidia-runtime-on-jetpack-5-0-developer-preview/211108/8

can you run your containers ok on jetpack JP 5 dev preview?

asimonov, 2022-04-19:
I just checked. If I start the basic L4T image r34.1 with --priviledged OR --volume="/dev:/dev" it fails with above errors.
If I remove both --priviledged AND --volume="/dev:/dev" it starts ok.

fortminors, 2022-04-21:
Hello! How did you manage to build it on Jetpack 5.0?

I am trying to run:
$ `./scripts/docker_build_ml.sh pytorch
`

What I get is:
`reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R34.1.0
l4t-base image:  nvcr.io/nvidia/l4t-base:r34.1
selecting OpenCV for L4T R34.1.0...
OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
Python3 version:  3.8
building PyTorch torch-1.10.0-cp38-cp38-linux_aarch64.whl, torchvision v0.11.1, torchaudio v0.10.0, cuda arch 7.2;8.7
Building l4t-pytorch:r34.1.0-pth1.10-py3 container...
Sending build context to Docker daemon    169kB
Step 1/26 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.4.4
Step 2/26 : FROM ${BASE_IMAGE}
pull access denied for jetpack, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
`

Default runtime is set to nvidia. This is on Jetson Xavier AGX

dusty-nv, 2022-04-21:
BTW, prebuilt [l4t-pytorch](https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-pytorch) image is now published on NGC if you would just prefer using that:

```
sudo docker pull nvcr.io/nvidia/l4t-pytorch:r34.1.0-pth1.12-py3
```

fortminors, 2022-04-29:
Yes, that works, thanks a lot! Is there any approximate date that Jetpack 5.0 will be released for production?

asimonov, 2022-04-29:
> I just checked. If I start the basic L4T image r34.1 with --priviledged OR --volume="/dev:/dev" it fails with above errors.
> If I remove both --priviledged AND --volume="/dev:/dev" it starts ok.

FYI there is workaround on the forum link quoted above

fortminors, 2022-04-29:
@dusty-nv I double checked, did what you have suggested and after running the docker_build_jetpack.sh, it terminates with a message: `E: Unable to locate package python3-vpi2` on step 14/19

The problem I have with l4t-pytorch image on Jetpack 5.0 is that running `torch.cuda.is_available()` inside the container returns `False` for some reason.

I've the same thing with Jeptack 4.6.1 and it works perfectly, but I can't use it since I need python >= 3.7 and Jetpack 4.6.1 with the suggested l4t-pytorch image runs on python 3.6

Am I doing something wrong?

fortminors, 2022-04-29:
I even tried deviceQuery cuda sample from the inside the container:

I am pulling the container as suggested:

`sudo docker pull nvcr.io/nvidia/l4t-pytorch:r34.1.0-pth1.12-py3`

I am launching the container as follows: 

`sudo docker run --rm -it --entrypoint bash nvcr.io/nvidia/l4t-pytorch:r34.1.0-pth1.12-py3`

```
cd /usr/local/cuda/samples/1_Utilities/deviceQuery
sudo make
./deviceQuery
```

```
./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

cudaGetDeviceCount returned 35
-> CUDA driver version is insufficient for CUDA runtime version
Result = FAIL

```

How do I make it work? This is on Jetson Xavier AGX

dusty-nv, 2022-04-29:
> How do I make it work? This is on Jetson Xavier AGX

Is this on JetPack 5.0 DP?   `l4t-pytorch:r34.1.0-pth1.12-py3` is for JetPack 5.0

fortminors, 2022-04-29:
@dusty-nv Yes, correct. Jetpack 5.0 DP.

I've had everything perfectly working thanks to your images on Jetpack 4.6.1, but now I need python of version at least 3.7 so I switched to Jetpack 5.0, where these issues occur

fortminors, 2022-05-11:
@dusty-nv Hello! Are there any updates on this? Jetpack 5.0 doesn't seem to be able to detect cuda for some reason

javadan, 2022-05-11:
Might help:  I fixed the missing cuda issue on 4.6.1 by adding host cuda directories as volumes to the docker, and adding them to LD_LIBRARY_PATH, and running 'sudo ldconfig'.

fortminors, 2022-05-11:
@javadan Thanks, however I have everything working on Jetpack 4.6.1. 

Jetpack 5.0 DP has got this weird problem that I am unable to fix

fortminors, 2022-05-11:
@javadan Maybe I have misunderstood you. Are you working with python 3.8 on jetpack 4.6.1?
I just need this exact setup as well as torch and torchvision of appropriate versions (so >= 1.11 for python 3.8)

javadan, 2022-05-11:
@fortminors 
No, sorry, still python 3.6.9.  I meant I've had missing CUDA lib issues on 4.6.1, and that's how I eventually fixed it, adding docker volumes, and refreshing C++ linker paths.

I'm just following the issues here, so I know when to upgrade to 5.0, haha.

fortminors, 2022-05-11:
@javadan Got it, thanks :) yes, looking forward to the full release of 5.0. There haven't been any announcements on the approximate dates, right?

dusty-nv, 2022-05-11:
> sudo docker run --rm -it --entrypoint bash nvcr.io/nvidia/l4t-pytorch:r34.1.0-pth1.12-py3

It looks like you are missing `--runtime nvidia`, which is what enables GPU in the container.  Can you try starting the container with that?

fortminors, 2022-05-11:
@dusty-nv You are absolutely right. I completely forgot about this flag, sorry for spamming, but maybe it helps someone else :)
Now torch sees cuda

closed 2023-04-11 by felrock