dusty-nv/jetson-containers #144: python version in docker image for ros-noetic with pytorch

LongfeiDE, 2022-04-27:
I have jetpack 4.6.1 and would like to work with a docker image with pytorch(python ==3.8) and ros noetic.

I followed the description to build an image using:  ./scripts/docker_build_ros.sh --distro noetic --with-pytorch
 
I noticed that the python version is set to be 3.8 as an argument in 
ARG PYTHON3_VERSION=3.8 (line 12 in https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.noetic)

However in PYTHON3_VERSION=`python3 -c 'import sys; version=sys.version_info[:3]; print("{0}.{1}".format(*version))'`(line 16 of https://github.com/dusty-nv/jetson-containers/blob/master/scripts/python_version.sh) it is adapting the parameter to 3.6 since I have jetpack 4.6.1 and default python has 3.6.9

I manually changed line 16 in https://github.com/dusty-nv/jetson-containers/blob/master/scripts/python_version.sh to PYTHON3_VERSION=3.8.

Still, the python in the docker image is 3.6.9.

Is there a way to have pytorch with python=3.8 and ros noetic?

Thanks a lot!

LongfeiDE, 2022-04-27:
Update: building an image with the commit 09074b16bcdaf53013ecd914b584a6b61fe421d4 for ros-noetic without the --with-pytorch tag also ends up with python version = 3.6.9 in jetpack 4.6.1, where no changes to the python in the system were conducted.

dusty-nv, 2022-04-27:
@LongfeiDE, the `PYTHON3_VERSION` arg is intended to say what's the OS-default Python3 version inside the container, not which Python3 version you personally want installed.  I think you'll want to add steps to install your desired Python3 version in the Dockerfile, and make sure ROS is built against it.  

BTW if you want to use stuff like PyTorch in the container, those would also need rebuilt against your desired Python 3 version as well...so this can turn into a bigger deal.

LongfeiDE, 2022-04-28:
> @LongfeiDE, the `PYTHON3_VERSION` arg is intended to say what's the OS-default Python3 version inside the container, not which Python3 version you personally want installed. I think you'll want to [...]

Thanks for the quick reply. 

Now I understand the argument. I am turning to use l4t-ml:r34 for my purpose. So this can be closed. 

Thanks again. 👍

closed 2022-04-28 by LongfeiDE