dusty-nv/jetson-containers #146: libcudart.so.10.2 is not available in tensorflow image

xya22er, 2022-05-15:
Hi @dusty-nv 

I am trying to build tensorflow Docker image on Xavier NX which has JetPack 4.5.

I built the tensorflow docker image using the command `./scripts/docker_build_ml.sh tensorflow`. Note that I needed tensorflow 2.5, so I commented out [these lines](https://github.com/dusty-nv/jetson-containers/blob/159acdc69c6f195e901c1c8aa9c0c8dcee3740d8/scripts/docker_build_ml.sh#L193-L196), to skip installing tensoflow 1.15, and keep tensorflow 2.5

The image was successfully built.
```bash
REPOSITORY                TAG                 IMAGE ID       CREATED          SIZE
l4t-tensorflow            r32.5.2-tf2.5-py3   72c4dc0ab51f   38 minutes ago   2.38GB
nvcr.io/nvidia/l4t-base   r32.5.0             d9141f00a2c3   17 months ago    635MB
```
There is something weird, though. The pulled base image has a tag of `r.32.5.0`. However, the tensorflow image has a tag of `r32.5.2-tf2.5-py3`

When I open a python3 interpreter and import tensorflow, I get the following error.

```bash
import tensorflow
2022-05-15 06:23:59.616179: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudart.so.10.2'; dlerror: libcudart.so.10.2: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /usr/local/cuda/lib64:/usr/local/cuda-10.2/targets/aarch64-linux/lib:
2022-05-15 06:23:59.616271: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
```

Tensorflow version
```bash
tensorflow.__version__
'2.5.0'
```

When I checked  `/usr/local/cuda/lib64`and `/usr/local/cuda-10.2/targets/aarch64-linux/lib` , they don't not have the required library `libcudart.so.10.2` 
```bash
libcudadevrt.a  libcudart_static.a  stubs
```

However, on the host system, Xavier NX OS, the required lib is available
```bash
libcublas.so                  libcuinj64.so             libcusparse.so            libnppicc.so.10          libnppig.so             libnppisu_static.a      libnvrtc-builtins.so
libcudadevrt.a                libcuinj64.so.10.2        libcusparse.so.10         libnppicc.so.10.2.1.89   libnppig.so.10          libnppitc.so            libnvrtc-builtins.so.10.2
libcudart.so                  libcuinj64.so.10.2.89     libcusparse.so.10.3.1.89  libnppicc_static.a       libnppig.so.10.2.1.89   libnppitc.so.10         libnvrtc-builtins.so.10.2.89
libcudart.so.10.2             libculibos.a              libcusparse_static.a      libnppicom.so            libnppig_static.a       libnppitc.so.10.2.1.89  libnvrtc.so
libcudart.so.10.2.89          libcupti.so               liblapack_static.a        libnppicom.so.10         libnppim.so             libnppitc_static.a      libnvrtc.so.10.2
libcudart_static.a            libcupti.so.10.2          libmetis_static.a         libnppicom.so.10.2.1.89  libnppim.so.10          libnpps.so              libnvrtc.so.10.2.89
libcufft.so                   libcupti.so.10.2.75       libnppc.so                libnppicom_static.a      libnppim.so.10.2.1.89   libnpps.so.10           libnvToolsExt.so
libcufft.so.10                libcurand.so              libnppc.so.10             libnppidei.so            libnppim_static.a       libnpps.so.10.2.1.89    libnvToolsExt.so.1
libcufft.so.10.1.2.89         libcurand.so.10           libnppc.so.10.2.1.89      libnppidei.so.10         libnppist.so            libnpps_static.a        libnvToolsExt.so.1.0.0
libcufft_static.a             libcurand.so.10.1.2.89    libnppc_static.a          libnppidei.so.10.2.1.89  libnppist.so.10         libnvgraph.so           stubs
libcufft_static_nocallback.a  libcurand_static.a        libnppial.so              libnppidei_static.a      libnppist.so.10.2.1.89  libnvgraph.so.10
libcufftw.so                  libcusolver.so            libnppial.so.10           libnppif.so              libnppist_static.a      libnvgraph.so.10.2.89
libcufftw.so.10               libcusolver.so.10         libnppial.so.10.2.1.89    libnppif.so.10           libnppisu.so            libnvgraph_static.a
libcufftw.so.10.1.2.89        libcusolver.so.10.3.0.89  libnppial_static.a        libnppif.so.10.2.1.89    libnppisu.so.10         libnvperf_host.so
libcufftw_static.a            libcusolver_static.a      libnppicc.so              libnppif_static.a        libnppisu.so.10.2.1.89  libnvperf_target.so

```

Any idea how to solve this error?

In addition, when I try to import `tensorrt` , it seem that it's not built into the Docker image
```bash
 python3
Python 3.6.9 (default, Mar 15 2022, 13:55:28) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tensorrt
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'tensorrt'

```

**NOTE** I run the container with `--runtime nvidia` 

Thank you in advance.

dusty-nv, 2022-05-16:
Hi @xya22er, it seems the mechanism which mounts the jetpack/cuda files into the container when `--runtime nvidia` is used isn't working properly.  Do you have these files?

``` bash
 ls /etc/nvidia-container-runtime/host-files-for-container.d/
cuda.csv  cudnn.csv  l4t.csv  tensorrt.csv  visionworks.csv
```

xya22er, 2022-05-17:
Thanks for your response 
 Yes, the jetpack has these files

hugopaliot, 2022-05-20:
Hello @xya22er, 

Have you found out the solution of your problem ?

i'm currently trying to run CUDA-enabled docker container on my Jetson Xavier NX (L4T R32.5.0) based on the `foxy-ros-base-l4t-r32.5.0` image, and i'm also dealing with missing runtime `libcudart.so` library and the other ones..
I tried to build it it from source (with default-runtime as nvidia as demanded) but it didn't help.

hugopaliot, 2022-05-20:
Finally got the libraries installed in `/usr/local/cuda/lib64` in my containers.

What I finally tried is downgrading some packages that are dependencies of `nvidia-container` (which installs other packages I had already installed by hand I suppose)

```The following packages have unmet dependencies:
 nvidia-container : Depends: libnvidia-container0 (= 0.9.0~beta.1) but 0.11.0+jetpack is to be installed
                    Depends: libnvidia-container-tools (= 0.9.0~beta.1) but 1.9.0-1 is to be installed
                    Depends: nvidia-container-runtime (= 3.1.0-1) but 3.9.0-1 is to be installed
                    Depends: nvidia-container-toolkit (= 1.0.1-1) but 1.9.0-1 is to be installed
                    Depends: nvidia-docker2 (= 2.2.0-1) but 2.10.0-1 is to be installed
```

After that I got the cuda libraries directly brought in my docker containers as expected.

It's been almost a week since I started to try running a CUDA-enabled container on my Jetson, no need to say I have installed/modified/uninstalled a few packages in the process so that may not exact be the reason you're not having the libraries in your container, but you can give it a try.

dusty-nv, 2022-05-23:
> failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown

Hi @hugopaliot, if you haven't yet seen this, here is the fix for this:  https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-995090398

Tal-Bass, 2022-05-24:
I had the same exact issue as xya22er and was able to solve the issue. My solution was to remove all of the https://nvidia.github.io sources from apt. Then uninstall and reinstall libnvidia-container0 (which also reinstalled the other packages for the nvidia docker runtime). I then ran into the "failed to create shim: OCI runtime create failed" which I had seen before, so to fix that I went through fix that you linked.

However I noticed the fix that you linked actually adds the nvidia.github.io repos to the source list. Then any time I sudo apt-get upgrade I get the original issue where libcudart.so.10.2 was not available and the files were not showing up in the docker container when using the nvidia runtime.

At this point I can reliably reproduce and fix the issue. To prevent this issue from happening again I removed the nvidia.github.io repos from the source list after installing nvidia-docker2. I have a feeling that adding the various nvidia.github.io repos to the source list eventually breaks the nvidia docker runtime because the related libraries get updated to incompatible versions. @dusty-nv  can you look into this and confirm if that is the case?

SkalskiP, 2022-06-21:
@dusty-nv any updates on that? @Tal-Bass / @hugopaliot did you manage to solve this issue? I think I have a pretty similar problem but with pytorch image: https://github.com/dusty-nv/jetson-containers/issues/164

Tal-Bass, 2022-06-21:
@SkalskiP yes I am able to solve the issue, though it is kind of a workaround that should have a better fix. Here are my steps:

1.  List apt sources:
grep ^ /etc/apt/sources.list /etc/apt/sources.list.d/*

2. Comment out any nvidia github repos (just the ones that start with https://nvidia.github.io)

3. sudo apt remove nvidia-container0

4. sudo apt --reinstall install nvidia-container

5. At this point you've removed the incorrect libraries and replaced them with the correct libraries.  However, you will most likely run into a different nvidia runtime docker issue. Try running the docker container. If you get the failed to create SHIM issue, run:

distribution=$(. /etc/os-release;echo $ID$VERSION_ID) \
   && curl -s -L https://nvidia.github.io/nvidia-docker/gpgkey | sudo apt-key add - \
   && curl -s -L https://nvidia.github.io/nvidia-docker/$distribution/nvidia-docker.list | sudo tee /etc/apt/sources.list.d/nvidia-docker.list

sudo apt-get update
sudo apt-get install nvidia-docker2=2.8.0-1

Note that this is a fix that @dusty-nv posted on another thread. Also note that this fix adds https://nvidia.github.io back to the package list. 

7. Because the fix adds the https://nvidia.github.io repos to the package list, you must once again remove those by repeating steps 1 and 2. If you don't, then doing an apt-get update and upgrade will trigger the issue again. This is the step that very much feels like a workaround.

8. Not sure if this is necessary, but step to add cuda paths. I think I needed to do this on my machine for an unrelated issue so you may not have to do this:

sudo nano /home/jetson/.bashrc

And past this to the end of the file:

# Add this to your .bashrc file
export CUDA_HOME=/usr/local/cuda
# Adds the CUDA compiler to the PATH
export PATH=$CUDA_HOME/bin:$PATH
# Adds the libraries
export LD_LIBRARY_PATH=$CUDA_HOME/lib64:$LD_LIBRARY_PATH

closed 2025-02-21 by johnnynunez