dusty-nv/jetson-containers #149: Can I run r34.1 container with CUDA on r32.6 (Jetpack 4)?

evshary, 2022-05-26:
I tried to run r34.1 container on Jetson NX with version r32.6.
The container can run successfully, but I found that I can't use CUDA.
Is it available to run r34.1 with CUDA on r23.6?
Thank you!

My steps:
1. Run the docker
```bash
docker run --rm -it --gpus all nvcr.io/nvidia/l4t-ml:r34.1.0-py3 bash
``` 
2. Build example
```bash
cd /usr/local/cuda/samples/1_Utilities/deviceQuery
make
```
3. Run
```bash
./deviceQuery
```
4. Error log
```
./deviceQuery Starting...

 CUDA Device Query (Runtime API) version (CUDART static linking)

cudaGetDeviceCount returned 3
-> initialization error
Result = FAIL
```

More information: The same steps are successful while using docker container r32.6 version.

dusty-nv, 2022-05-26:
Hi @evshary, the r34.1.0 containers only run (with CUDA) on r34.1.0 and newer

evshary, 2022-05-26:
@dusty-nv Thank you for the answers.
I have another question:
If I run Ubuntu 20.04 container on r32.6 and pass the native host's CUDA library (that is, r32.6 one) to the container,
do you think it's possible to run CUDA with the library on Ubuntu 20.04 container?

dusty-nv, 2022-05-26:
I'm not sure, you could try like this person did:  https://carlosedp.medium.com/upgrading-your-nvidia-jetson-xavier-nx-to-latest-ubuntu-focal-fossa-20-04-5e92ccc5a66

evshary, 2022-05-27:
Thanks a lot! I'll close the issue.

closed 2022-05-27 by evshary