dusty-nv/jetson-containers #150: Image appears to be missing

brianlmerritt, 2022-05-28:
On Jetson Orin I run the command below (or any other build - even just one ros distro without pytorch or slam) and get an error on the docker pull seemingly

`./scripts/docker_build_ros.sh --distro all --with-pytorch --with-slam`

The output is

```
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R34.1.1
l4t-base image:  nvcr.io/nvidian/nvidia-l4t-base:r34.1.1
selecting OpenCV for L4T R34.1.1...
OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
ROS_DISTRO:   all
ROS_PACKAGE:  ros_base
WITH_PYTORCH: on
WITH_SLAM:    on
[sudo] password for blm:

Building container ros:noetic-ros-base-l4t-r34.1.1
BASE_IMAGE=jetpack:r34.1.1

Building ros:noetic-ros-base-l4t-r34.1.1 container...
Sending build context to Docker daemon  179.2kB
Step 1/18 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.5.0
Step 2/18 : FROM ${BASE_IMAGE}
pull access denied for jetpack, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
```

And yet if I do a `docker pull nvcr.io/nvidia/l4t-base:r32.5.0` that pulls fine

Any thoughts?  Are the Orin / Jetpack 5 containers for ROS there yet?

dusty-nv, 2022-05-31:
Hi @brianlmerritt, for JetPack 5, you need to run `scripts/docker_build_jetpack.sh` first

Also, the `--with-slam` option is untested on JetPack 5 and may have issues

brianlmerritt, 2022-06-09:
So if I log into nvcr.io and run the docker build command manually from jetson-containers directory I get

```
docker build --network="host" -t "jetpack:r34.1.1" -f "Dockerfile.jetpack" --build-arg "BASE_IMAGE=nvcr.io/nvidian/nvidia-l4t-base:r34.1.1" --build-arg "OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz" .
Sending build context to Docker daemon  186.4kB
Step 1/19 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r34.1
Step 2/19 : FROM ${BASE_IMAGE}
pull access denied for nvcr.io/nvidian/nvidia-l4t-base, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
```

The error occurs even after logging into nvcr.io with docker and my access token.