dusty-nv/jetson-containers #154: Still struggling with container build on Jetson Xavier NX and Orin Development Kits

brianlmerritt, 2022-06-09:
## Log into nvcr.io and run script/docker_build_jetpack.sh

```
docker login nvcr.io
Username: $oauthtoken
Password:
Login Succeeded
xxx@jetson-xavier-nx:~$  cd jetson/jetson-containers/
xxx@jetson-xavier-nx:~/jetson/jetson-containers$ scripts/docker_build_jetpack.sh
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R34.1.1
l4t-base image:  nvcr.io/nvidian/nvidia-l4t-base:r34.1.1
selecting OpenCV for L4T R34.1.1...
OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
cuda
cuda-11-4
cuda-cccl-11-4
cuda-command-line-tools-11-4
cuda-compiler-11-4
cuda-cudart-11-4
cuda-cudart-dev-11-4
cuda-cuobjdump-11-4
cuda-cupti-11-4
cuda-cupti-dev-11-4
cuda-cuxxfilt-11-4
cuda-documentation-11-4
cuda-driver-dev-11-4
cuda-gdb-11-4
cuda-gdb-src-11-4
cuda-libraries-11-4
cuda-libraries-dev-11-4
cuda-minimal-build-11-4
cuda-nvcc-11-4
cuda-nvdisasm-11-4
cuda-nvml-dev-11-4
cuda-nvprof-11-4
cuda-nvprune-11-4
cuda-nvrtc-11-4
cuda-nvrtc-dev-11-4
cuda-nvtx-11-4
cuda-runtime-11-4
cuda-samples-11-4
cuda-sanitizer-11-4
cuda-toolkit-11-4
cuda-toolkit-11-4-config-common
cuda-toolkit-11-config-common
cuda-toolkit-config-common
cuda-tools-11-4
cuda-visual-tools-11-4
deepstream-6.1
graphsurgeon-tf
libcublas-11-4
libcublas-dev-11-4
libcudla-11-4
libcudla-dev-11-4
libcudnn8
libcudnn8-dev
libcudnn8-samples
libcufft-11-4
libcufft-dev-11-4
libcurand-11-4
libcurand-dev-11-4
libcusolver-11-4
libcusolver-dev-11-4
libcusparse-11-4
libcusparse-dev-11-4
libnpp-11-4
libnpp-dev-11-4
libnvidia-container0
libnvidia-container1
libnvidia-container-tools
libnvinfer8
libnvinfer-bin
libnvinfer-dev
libnvinfer-doc
libnvinfer-plugin8
libnvinfer-plugin-dev
libnvinfer-samples
libnvonnxparsers8
libnvonnxparsers-dev
libnvparsers8
libnvparsers-dev
libnvvpi2
libopencv
libopencv-dev
libopencv-python
libopencv-samples
nsight-graphics-for-embeddedlinux-2022.2.0.0
nsight-systems-2022.2.3
nsight-systems-linux-tegra-public-2021.5.4.20-6e27cb6
nvidia-container-runtime
nvidia-container-toolkit
nvidia-docker2
opencv-licenses
python3.8-vpi2
python3.9-vpi2
python3-libnvinfer
python3-libnvinfer-dev
tensorrt
uff-converter-tf
vpi2-demos
vpi2-dev
vpi2-samples

Building container jetpack:r34.1.1
BASE_IMAGE=nvcr.io/nvidian/nvidia-l4t-base:r34.1.1
L4T_APT_SOURCE=
L4T_APT_SOURCE_CLEAN=

Building jetpack:r34.1.1 container...
[sudo] password for xxx:
Sending build context to Docker daemon  186.4kB
Step 1/19 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r34.1
Step 2/19 : FROM ${BASE_IMAGE}
unauthorized: authentication required
```
## nv_tegra_release

```
cat /etc/nv_tegra_release
# R34 (release), REVISION: 1.1, GCID: 30414990, BOARD: t186ref, EABI: aarch64, DATE: Tue May 17 04:20:55 UTC 2022
```

## nvcc --version

```
nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Thu_Nov_11_23:44:05_PST_2021
Cuda compilation tools, release 11.4, V11.4.166
Build cuda_11.4.r11.4/compiler.30645359_0
```

## Login

```
Welcome to Ubuntu 20.04.4 LTS (GNU/Linux 5.10.65-tegra aarch64)
```

brianlmerritt, 2022-06-09:
So if I run the docker build command manually from jetson-containers after already logging into nvcr.io using my auto token I get

```
xxx@jetson-xavier-nx:~/jetson/jetson-containers$ sudo docker build --network="host" -t "jetpack:r34.1.1" -f "Dockerfile.jetpack" --build-arg "BASE_IMAGE=nvcr.io/nvidian/nvidia-l4t-base:r34.1.1" --build-arg "OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz" .
[sudo] password for xxx:
Sending build context to Docker daemon  186.4kB
Step 1/19 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r34.1
Step 2/19 : FROM ${BASE_IMAGE}
unauthorized: authentication required
```

And if I run the above without sudo I get a bit more detail

```
pull access denied for nvcr.io/nvidian/nvidia-l4t-base, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
```

brianlmerritt, 2022-06-09:
Looks like the base image being requested is wrong - the following pulls in a valid image

`docker build --network="host" -t "jetpack:r34.1.1" -f "Dockerfile.jetpack" --build-arg "BASE_IMAGE=nvcr.io/nvidia/l4t-base:r34.1.1" --build-arg "OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz" .`

brianlmerritt, 2022-06-09:
Either the tooling to setup an image at `nvcr.io/nvidian/nvidia-l4t-base:r34.1.1` is missing or `docker_base.sh` needs to have the BASE_DEVEL in line 30 set to `BASE_DEVEL="nvcr.io/nvidia/l4t-base:r34.1.1"`

dusty-nv, 2022-06-09:
Hi @brianlmerritt, sorry about that, this should be fixed now in commit https://github.com/dusty-nv/jetson-containers/commit/be0dca3d19b30e64d129c92e93425c9ede40d65f

brianlmerritt, 2022-06-09:
Sorry @dusty-nv - the BASE_DEVEL still refers to a non-existant image (or to an image which even with my auth token I am not allowed to access)

```
Login Succeeded
xxx@jetson-xavier-nx:~/jetson/jetson-containers$ docker pull nvcr.io/nvidian/nvidia-l4t-base:r34.1
Error response from daemon: pull access denied for nvcr.io/nvidian/nvidia-l4t-base, repository does not exist or may require 'docker login': denied: requested access to the resource is denied
```

dusty-nv, 2022-06-09:
> the BASE_DEVEL still refers to a non-existant image (or to an image which even with my auth token I am not allowed to access)

The BASE_DEVEL images are for my internal use while I am building containers before a version of JetPack is officially released.  You should be able to `docker pull nvcr.io/nvidia/l4t-base:r34.1` though

brianlmerritt, 2022-06-09:
I can, and that worked fine.  I actually did a `docker pull nvcr.io/nvidia/l4t-base:r34.1.1` and that worked really well.

It's just confusing if the `./scripts/docker_build_jetpack.sh` insists upon building from a container with BASE_DEVEL set to a one which most developers don't have access :D  Or should I set a flag to say I don't have access to BASE_DEVEL?

dusty-nv, 2022-06-10:
> It's just confusing if the `./scripts/docker_build_jetpack.sh` insists upon building from a container with BASE_DEVEL set to a one which most developers don't have access :D

It shouldn't currently be trying to use a BASE_DEVEL image - when a new JetPack is released, I update the script so that it doesn't use it anymore (but I had forgotten to do that for JetPack 5.0.1 - oops!)

brianlmerritt, 2022-06-11:
No worries - it makes sense now

closed 2022-06-11 by brianlmerritt