dusty-nv/jetson-containers #155: Base image not found

CourchesneA, 2022-06-10:
It seems [a recent commit ](https://github.com/dusty-nv/jetson-containers/commit/807641387dff89f5c30b4513fad740877521a29c) modified the $BASE_IMAGE variable in `docker_base.sh` to allow x86 support, but it breaks ros builds for me.

Specifically on r34.1.1 when I try to build using `test`, the $BASE_IMAGE gets set to `jetpack:r34.1.1` as per (this line)[		BASE_IMAGE="jetpack:r$L4T_VERSION"]
and then throws an error since the image does not exist.

I am not sure why this change was made in the first place so I won't offer a PR, but as a work around just commenting out the line solves the problem for me. I guess the variable $BASE_IMAGE is being migrated to $BASE_IMAGE_L4T ? In that case we might want to do the change in the build scripts too.

dusty-nv, 2022-06-10:
Hi @CourchesneA, if you are on JetPack 5.0 / R34.1, then you should run `scripts/docker_build_jetpack.sh` first

CourchesneA, 2022-06-10:
That is what I was missing, thanks !

closed 2022-06-10 by CourchesneA