dusty-nv/jetson-containers #158: Galactic image opencv issues

enricosutera, 2022-06-14:
Hello there,
I'm trying to build a ros2 project using this image `dustynv/ros:foxy-ros-base-l4t-r34.1.1` on a Nvidia Jetson NX with JP 4.6.1.
I did not build the image, I used the already built one.
Unfortunately I have issues with dependencies concerning opencv.
I can't manage to install a lot of ros2 packages because the root dependency are libopencv and libopencv-dev.
Of course they are not being installed because actually openCV is already installed (4.5.0) but not recognized. What am I missing?
This is the output of `fix --broken`
```
$apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
  librhash0 libuv1
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  libopencv libopencv-dev
The following NEW packages will be installed:
  libopencv libopencv-dev
0 upgraded, 2 newly installed, 0 to remove and 20 not upgraded.
261 not fully installed or removed.
Need to get 12.5 MB of archives.
After this operation, 70.0 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 https://repo.download.nvidia.com/jetson/common r34.1/main arm64 libopencv arm64 4.5.4-8-g3e4c170df4 [11.3 MB]
Get:2 https://repo.download.nvidia.com/jetson/common r34.1/main arm64 libopencv-dev arm64 4.5.4-8-g3e4c170df4 [1,247 kB]                                                        
Fetched 12.5 MB in 11s (1,108 kB/s)                                                                                                                                             
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 103043 files and directories currently installed.)
Preparing to unpack .../libopencv_4.5.4-8-g3e4c170df4_arm64.deb ...
Unpacking libopencv (4.5.4-8-g3e4c170df4) ...
dpkg: error processing archive /var/cache/apt/archives/libopencv_4.5.4-8-g3e4c170df4_arm64.deb (--unpack):
 trying to overwrite '/usr/share/opencv4/haarcascades/haarcascade_eye.xml', which is also in package opencv-libs 4.5.0
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Preparing to unpack .../libopencv-dev_4.5.4-8-g3e4c170df4_arm64.deb ...
Unpacking libopencv-dev (4.5.4-8-g3e4c170df4) ...
dpkg: error processing archive /var/cache/apt/archives/libopencv-dev_4.5.4-8-g3e4c170df4_arm64.deb (--unpack):
 trying to overwrite '/usr/bin/opencv_annotation', which is also in package opencv-dev 4.5.0
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libopencv_4.5.4-8-g3e4c170df4_arm64.deb
 /var/cache/apt/archives/libopencv-dev_4.5.4-8-g3e4c170df4_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```
Hope to find some help!

dusty-nv, 2022-06-14:
Hi @enricosutera, can you try running it like this:

``` bash
DEBIAN_FRONTEND=noninteractive apt --fix-broken -y install
```

Or maybe you need to remove previous OpenCV first with `apt-get purge -y '*opencv*'`

enricosutera, 2022-06-15:
Hi @dusty-nv. I purged before running `rosdep` and everything worked fine! Thank you 
To be honest I'm not sure which the issue is

CourchesneA, 2022-06-21:
I have the same issue with ROS Humble. It seems opencv is being installed from a private binary:
https://github.com/dusty-nv/jetson-containers/blob/be0dca3d19b30e64d129c92e93425c9ede40d65f/Dockerfile.ros.galactic#L85
Is there any reason why we are not using the default debian package `opencv-dev` from APT ?

dusty-nv, 2022-06-21:
> Is there any reason why we are not using the default debian package `opencv-dev` from APT ?

The OpenCV packages from ubuntu apt are not built with CUDA enabled.  Also depending on the Ubuntu distro, they can be older versions of OpenCV.

Instead, I install the OpenCV 4.5 that I built with CUDA enabled, and then compile ROS so that it is built to use that version of OpenCV.   ROS is built from source and the apt packages are not used.   ROS Humble is based on 22.04 while the JetPack 5.0 containers use 20.04 so the apt packages aren't able to be used anyways.

tsadarsh, 2022-10-26:
I have a similar error showing up when I try to `apt upgrade`. I am using the `dustynv/ros:foxy-ros-base-l4t-r35.1.0` image.
Purging the `opencv` does not seem to be the best option. I want to use the CUDA enabled OpenCV.

This did not work out either:

> ```shell
> DEBIAN_FRONTEND=noninteractive apt --fix-broken -y install
> ```

Here is the error that I am facing:
```
Preparing to unpack .../libopencv_4.5.4-8-g3e4c170df4_arm64.deb ...
Unpacking libopencv (4.5.4-8-g3e4c170df4) ...
dpkg: error processing archive /var/cache/apt/archives/libopencv_4.5.4-8-g3e4c170df4_arm64.deb (--unpack):
 trying to overwrite '/usr/share/opencv4/haarcascades/haarcascade_eye.xml', which is also in package opencv-libs 4.5.0
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libopencv_4.5.4-8-g3e4c170df4_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

dusty-nv, 2022-10-26:
Hi @tsadarsh, can you try having apt-mark or dpkg hold/pin the package so it's not tried to be upgraded?   https://askubuntu.com/a/99780

tsadarsh, 2022-10-27:
That should solve the problem. Thank you @dusty-nv for the help.

fettahyildizz, 2022-11-03:
> That should solve the problem. Thank you @dusty-nv for the help.

@tsadarsh Were you able to update and install rosdeps by not letting libopencv package to be upgraded?

tsadarsh, 2022-11-04:
Hi @fettahyildizz. I did not try doing it that way.

You could try working by pulling this image (note the **tag**):
```docker pull tsadarsh/jetson:nano```

Let me know if it solves your problem.

fettahyildizz, 2022-11-04:
Hello @tsadarsh, unfortunately our working environment is ROS Noetic. but thanks anyway. Could you provide your dockerfile so I can find out how you install cv2

tsadarsh, 2022-11-04:
Our goal was to have a container that had PyTorch (Python >= 3.8 and CUDA cores enabled) along with ROS 2 Foxy. @waaeem02 did a lot of work in compiling many packages from source.
It was basically a hit and trial approach. Compile, run, fix, repeat. Once we got all the necessary packages installed we pushed the image to DockerHub. Hence no Dockerfile was made.

I am sure @waseem02 can be of some help.

You can still use our image and install ROS noetic in it.

tonynajjar, 2024-01-23:
2 years later I stumbled upon this issue as well

> Hi @tsadarsh, can you try having apt-mark or dpkg hold/pin the package so it's not tried to be upgraded? https://askubuntu.com/a/99780

@dusty-nv would it make sense to add this to the dockerfiles after installing the cuda-enabled opencv?

crytpy, 2024-02-02:
I've also stumbled upon this issue with humble when trying to build ros-humble-desktop

joeldushouyu, 2024-07-06:
> 2 years later I stumbled upon this issue as well using the humble images
> 
> > Hi @tsadarsh, can you try having apt-mark or dpkg hold/pin the package so it's not tried to be upgraded? https://askubuntu.com/a/99780
> 
> @dusty-nv would it make sense to add this to the dockerfiles after installing the cuda-enabled opencv?

I came across this issue today as trying to build the iron docker . I agree this should be set somewhere in the docker file

dusty-nv, 2024-07-06:
The primary opencv dockerfile changed such that it installs opencv-gpu from my pip server, so that the wheels are cached but will be automatically rebuilt as needed (i.e. for different versions of CUDA or Python)

IIRC what I settled on with rosdep was using `--skip-keys opencv ...` - is that not working for you @joeldushouyu ?  Did it work if you added the apt hold?

joeldushouyu, 2024-07-06:
> d on

Hello @dusty-nv. First of all, thanks for your work and for keeping up on this thread!

Maybe my situation may be a little bit different, not sure if I should open another thread or not.

Initially, I was trying to modify the existing [dustynv/ros:iron-desktop-l4t-r35.4.1](https://hub.docker.com/r/dustynv/ros/tags) to add more package that I need, so something like below

```Docker 
FROM  dustynv/ros:iron-desktop-l4t-r35.4.1
Install some ros-iron-packages
```
However, this gives me an error, saying it cannot find the package I need. After searching, I realize **14t-r35** is built in jetpack 5, which is built on top of **Ubuntu 20.04**, not **Ubuntu 22** as required for **iron** version of ros2.

Then, I  decide to use the existing docker **r36** image and build myself a fresh iron docker, since I dont' find any existing iron docker that is based on **r36**. The step for install ros-iron all came from the [official instruction](https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html)
```Dockerfile
FROM  dustynv/jetson-utils:r36.2.0
#https://stackoverflow.com/questions/20635472/using-the-run-instruction-in-a-dockerfile-with-source-does-not-work
# RUN rm /bin/sh && ln -s /bin/bash /bin/sh
# RUN apt-get update &&  apt-get -y install sudo
# #https://stackoverflow.com/questions/25845538/how-to-use-sudo-inside-a-docker-container
RUN useradd -m docker && echo "docker:docker" | chpasswd && adduser docker sudo  

RUN  apt update && apt install locales
RUN  locale-gen en_US en_US.UTF-8
RUN  update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
RUN  export LANG=en_US.UTF-8
RUN  apt install software-properties-common -y
RUN  add-apt-repository universe
RUN  apt update &&  apt install curl -y
RUN  curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
RUN  echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2/ubuntu $(. /etc/os-release && echo $UBUNTU_CODENAME) main" |  tee /etc/apt/sources.list.d/ros2.list > /dev/null
RUN  apt update -y 
# lock openc https://github.com/dusty-nv/jetson-containers/issues/158
RUN  apt upgrade -y
RUN  apt update -y 
# RUN  apt-mark hold libopencv  
RUN  apt install -y ros-iron-ros-desktop
```
And when the docker try to install the desktop version, I got the following error which complains about **libopencv**.

```bash

485.6 Selecting previously unselected package usb-modeswitch-data.
485.6 Preparing to unpack .../852-usb-modeswitch-data_20191128-4_all.deb ...
485.6 Unpacking usb-modeswitch-data (20191128-4) ...
485.7 Selecting previously unselected package usb-modeswitch.
485.7 Preparing to unpack .../853-usb-modeswitch_2.6.1-3ubuntu2_arm64.deb ...
485.7 Unpacking usb-modeswitch (2.6.1-3ubuntu2) ...
485.8 Errors were encountered while processing:
485.8  /tmp/apt-dpkg-install-xT6cyi/327-libopencv-core-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/328-libopencv-flann-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/331-libopencv-imgproc-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/332-libopencv-imgcodecs-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/334-libopencv-videoio-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/335-libopencv-highgui-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/337-libopencv-ml-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/338-libopencv-features2d-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/339-libopencv-calib3d-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/344-libopencv-dnn-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/345-libopencv-objdetect-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/347-libopencv-photo-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/349-libopencv-video-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/350-libopencv-shape-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/352-libopencv-stitching-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/354-libopencv-superres-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/356-libopencv-videostab-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/359-libopencv-contrib-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/360-libopencv-dev_4.5.4+dfsg-9ubuntu4_arm64.deb
485.8  /tmp/apt-dpkg-install-xT6cyi/488-opencv-data_4.5.4+dfsg-9ubuntu4_all.deb
486.1 E: Sub-process /usr/bin/dpkg returned an error code (1)
------

 3 warnings found (use --debug to expand):
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 191)
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 192)
 - LegacyKeyValueFormat: "ENV key=value" should be used instead of legacy "ENV key value" format (line 233)
Dockerfile:22
--------------------
  20 |     RUN  apt upgrade -y 
  21 |     # RUN  apt-mark hold libopencv  
  22 | >>> RUN  apt install ros-iron-desktop -y
  ```

I did tried to iuse the **hold** method on the libopencv before trying to install iron-desktop. However, I got the package not found error, as shown below
```bash
 => ERROR [ 15/104] RUN  apt-mark hold libopencv                                                                                                                  2.1s
------
 > [ 15/104] RUN  apt-mark hold libopencv:
2.071 E: Unable to locate package libopencv
2.071 E: No packages found
------
```

Thus, I wonder if you have any insight or solution on how I can get an iron version of ros that is based on **r36**? Thanks in advance!

dusty-nv, 2024-07-07:
@joeldushouyu try an `apt-get purge -y '*opencv*'` in the same RUN statement as the apt install, like:

```
RUN apt-get purge -y '*opencv*' && \
    apt install -y ros-iron-ros-desktop
```

Did you reflash your Jetson (presumably Orin) to JetPack 6?   Because the r36 images aren't compatible with r35, well at least not CUDA (although stranger things have happened)

To add source packages in your own ROS container, you can use rosinstall_generator and then rosdep followed by colcon build, like here:   https://github.com/dusty-nv/jetbot_ros/blob/d8e5ee1b17f5c66d038017b86f8920a496197ea9/Dockerfile#L76

or add them to the list here if rebuilding the base container:

https://github.com/dusty-nv/jetson-containers/blob/08cedc24adfc9a675462ab53c9148b54e7f33c29/packages/ros/ros2_build.sh#L79

joeldushouyu, 2024-07-07:
> @joeldushouyu try an `apt-get purge -y '*opencv*'` in the same RUN statement as the apt install, like:
>
> ```
> RUN apt-get purge -y '*opencv*' && \
>     apt install -y ros-iron-ros-desktop
> ```
>
> Did you [...]

@dusty-nv 

The trick of **removing** OpenCV before installing ros-iron-desktop worked!!! Thanks for the help! One quick question, is there a huge difference between the OpenCV package that came with JetPack 6 docker image and the opencv package install by Ubuntu?

I have flashed JetPack 6 onto Jetson **Nano**, and so far the docker seems working as I expected. I haven't had the chance to test and run the [dockerfie](https://github.com/dusty-nv/jetbot_ros/blob/d8e5ee1b17f5c66d038017b86f8920a496197ea9/Dockerfile#L76) in jetbot, the repo you pointed out. I would give it a try in the future. 

Thanks once again!!!

dusty-nv, 2024-07-08:
@joeldushouyu the OpenCV that comes with JetPack was built with GStreamer enabled.  The OpenCV that I build in these containers has CUDA and OpenCV enabled.

johnnynunez, 2024-07-08:
> @joeldushouyu the OpenCV that comes with JetPack was built with GStreamer enabled. The OpenCV that I build in these containers has CUDA and OpenCV enabled.

It should come with cuda also in jetpack, although it is easy to install. I have it with the latest version of everything. Thank you for your containers

closed 2025-02-21 by johnnynunez