dusty-nv/jetson-containers #159: libcublas.so.10: cannot open shared object file: No such file or directory

chris-volley, 2022-06-14:
I am building a simple container on the following image

FROM dustynv/l4t-ml:r32.6.1-py3
then when I got to run my script using OpenCV
File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 86, in bootstrap
    import cv2
ImportError: libcublas.so.10: cannot open shared object file: No such file or directory

From looking at the file it looks like OpenCV is included in the build dockerfile 
https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ml

What am I missing to get working properly.

dusty-nv, 2022-06-14:
Is the JetPack-L4T version of your device L4T R32.6.x?   You can check it with `cat /etc/nv_tegra_release`

Are you able to use other CUDA libraries in the container?   i.e. are they being properly mounted into `/usr/local/cuda` when `--runtime nvidia` is used?

chris-volley, 2022-06-15:
@dusty-nv  Thanks for the response

`cat /etc/nv_tegra_release`
# R32 (release), REVISION: 6.1, GCID: 27863751, BOARD: t186ref, EABI: aarch64, DATE: Mon Jul 26 19:36:31 UTC 2021

I am starting my container with this command: 
`docker run --gpus=all --volume ${PWD}:/vision --rm -it <tag> /bin/bash`

inside the container
`ls -l /usr/local/cuda`
lrwxrwxrwx 1 root root 20 Jul 23  2021 /usr/local/cuda -> /usr/local/cuda-10.2
The folder is populated with things as well.

dusty-nv, 2022-06-15:
Please try starting it like this:

``` bash
sudo docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.6.1-py3
```

and then running `python3 -c 'import tensorrt'` inside the container.  is that able to work?

chris-volley, 2022-06-15:
So I ran that command exactly for running inside the container.
running 
import tensorrt as above gives no printout, or errors. 
When I do python3 -c 'import cv2' I get the same libcublas error

dusty-nv, 2022-06-15:
There is something wrong with your jetpack install or nvidia-container-runtime install that is preventing the jetpack files from being mounted into the containers properly.

Can you try re-installing the `nvidia-container nvidia-container-csv-tensorrt` packages through apt?

If it continues not to work, I would just reflash the device or SD card before too long.

chris-volley, 2022-06-15:
Thanks
After removing packages and reinstalling those 2 I get the following try to run it. 

docker: Error response from daemon: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown.

dusty-nv, 2022-06-15:
Hmm okay, can you try this now:  https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-981761419

chris-volley, 2022-06-16:
So went with the reflash. After doing that installed the two packages above and the dependencies. I still get the libcublas.so error.

rgov, 2022-08-17:
@chris-volley What made you decide to close this issue? Did you find a way to resolve the issue?

I am seeing the same problem on a newly flashed Jetson Nano running L4T r32.7.1, the latest Docker, `nvidia-docker2`, etc. Can we have the issue re-opened?

Neither `nvcr.io/nvidia/l4t-base:r32.7.1` nor `nvcr.io/nvidia/l4t-ml:r32.7.1-py3` have `/usr/local/cuda-10.2/targets/aarch64-linux/lib`. There is a file `stubs/libcublas.so` but it does not get loaded even if you extend `LD_LIBRARY_PATH`.

chris-volley, 2022-08-17:
@rgov 

Yeah I was able to get it to work, suppose I should of included how,  so the nvidia runtime loads those into the image at runtime, or building if you have the default runtime set up in the config. 

What I had wrong was for space concerns I had moved cuda install on the host to another partition. It wasn't straightforward to find but, I had to update the 
/etc/nvidia-container-runtime/host-files-for-container.d/cuda.csv. 
this file indicates what is loaded by the nvidia runtime and where things are. After that I was golden.  

Hopefully that helps.

rgov, 2022-08-17:
Thanks! It gives me a pointer to something else being wrong with the container runtime. I filed a new issue here: https://github.com/NVIDIA/nvidia-container-runtime/issues/173