dusty-nv/jetson-containers #160: Error building ros2 containers

SimeonOA, 2022-06-15:
I am trying to build a custom Docker Image using ROS rolling on Jetson Nano running Ubuntu 20.04.

I have been working from the Dockerfile for Humble: https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.humble

I keep having the following error:

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rti_connext_dds_cmake_module: No definition of [rti-connext-dds-6.0.1] for OS version [bionic]
`

`The command '/bin/bash -c mkdir -p ${ROS_ROOT}/src &&     cd ${ROS_ROOT} &&     rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG}         launch_xml         launch_yaml         launch_testing         launch_testing_ament_cmake         demo_nodes_cpp         demo_nodes_py         example_interfaces         image_geometry         image_pipeline         image_transport         compressed_image_transport         compressed_depth_image_transport         > ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     apt-get update &&     cd ${ROS_ROOT} &&     rosdep init &&     rosdep update &&     rosdep install -y           --ignore-src        --from-paths src       --rosdistro ${ROS_DISTRO}       --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build &&     rm ${ROS_ROOT}/*.rosinstall' returned a non-zero code: 1
`

I have also run just the Dockerfile without any change (except the OpenCV lines commented out) and I keep having the same error so I do not think it is the customization I made.

I would appreciate any help I can get on this. 

Thank you

dusty-nv, 2022-06-15:
Hi @SimeonOA, when I tried to compile Humble for JetPack 4.x / 18.04, there were several errors which I was ultimately unable to resolve.

The `rti_connext_dds_cmake_module: No definition of [rti-connext-dds-6.0.1] for OS version [bionic]` error I was able to workaround by adding it to the `--skip-keys` flag.  But then it encountered some other compiler errors and I eventually gave up.  I would just stick with Foxy or Galactic on Nano.

CourchesneA, 2022-06-21:
Hi @SimeonOA , I am having a similar issue for ROS Humble, so I would be interested to see what you ended up doing. I see on your post you mention you are running Ubuntu 20 but your error shows `OS version [bionic]`. Is it not recognized properly ? While extra package are not available pre-compiled on Ubuntu 20, I was able to get the basic installation working on a Jetxon Xavier NX.

SimeonOA, 2022-07-20:
Hi @CourchesneA,

Dusty's suggestion fixed the issue but like he mentioned, there were other errors. I have been troubleshooting some of them and have progressed all the way to `colcon build` but the build is currently failing on one of the packages

I am running Ubuntu 20 on Jetson Nano. It is recognized properly. I think `OS version [bionic]` error was not referring to the device OS.

asimonov, 2022-07-29:
I am interested to build ROS2 Humble on jetpack 4.6.x as well. This is still production version for a while...

I am using this repo Dockerfile, but had to modify it a little, roughly as follows:

```
...

ARG ROS_PKG=desktop
ENV ROS_DISTRO=humble
ENV ROS_ROOT=/opt/ros/${ROS_DISTRO}
ENV ROS_PYTHON_VERSION=3

ENV RTI_NC_LICENSE_ACCEPTED yes

# 
# download/build ROS from source
#
RUN mkdir -p ${ROS_ROOT}/src && \
    cd ${ROS_ROOT} && \    
    rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG} \
		camera_calibration_parsers \
		camera_info_manager \
		cv_bridge \
		compressed_image_transport \
		compressed_depth_image_transport \
		demo_nodes_cpp \
		demo_nodes_py \
		example_interfaces \
		image_geometry \
		image_pipeline \
		image_transport \
		launch_xml \
		launch_yaml \
		launch_testing \
		launch_testing_ament_cmake \
                perception \
                rmw_fastrtps \
		v4l2_camera \
		vision_opencv \
		vision_msgs \
		> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
    cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
    vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
    # install dependencies using rosdep
    apt-get update && \
    cd ${ROS_ROOT} && \
    rm /etc/ros/rosdep/sources.list.d/20-default.list && \
    rosdep init && \
    rosdep update && \
    rosdep install -y \
    	  --ignore-src \
          --from-paths src \
	  --rosdistro ${ROS_DISTRO} \
	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv ignition-math6 rti-connext-dds-6.0.1 ignition-cmake2" && \
    rm -rf /var/lib/apt/lists/* && \
    apt-get clean

```

I added a few skipped keys in rosdep to avoid the errors, but now with above docker I get:

```
#0 7753.6 --- stderr: rclpy
#0 7753.6 CMake Error at CMakeLists.txt:154 (target_link_libraries):
#0 7753.6   Target "test_python_allocator" links to:
#0 7753.6 
#0 7753.6     pybind11::embed
#0 7753.6 
#0 7753.6   but the target was not found.  Possible reasons include:
#0 7753.6 
#0 7753.6     * There is a typo in the target name.
#0 7753.6     * A find_package call is missing for an IMPORTED target.
#0 7753.6     * An ALIAS target is missing.
#0 7753.6 
#0 7753.6 
#0 7753.6 
#0 7753.6 CMake Generate step failed.  Build files cannot be regenerated correctly.
#0 7753.6 ---
#0 7753.6 Failed   <<< rclpy [5min 10s, exited with code 1]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 [Processing: rclcpp, rclpy]
#0 7768.3 Aborted  <<< rclcpp [5min 7s]
#0 7768.7 
#0 7768.7 Summary: 187 packages finished [2h 1min 20s]
#0 7768.7   1 package failed: rclpy
#0 7768.7   1 package aborted: rclcpp
#0 7768.7   41 packages had stderr output: ament_copyright ament_cppcheck ament_cpplint ament_flake8 ament_index_python ament_lint ament_lint_cmake ament_mypy ament_package ament_pep257 ament_pycodestyle ament_uncrustify ament_xmllint domain_coordinator foonathan_memory_vendor google_benchmark_vendor iceoryx_posh ignition_cmake2_vendor ignition_math6_vendor launch launch_testing launch_xml launch_yaml libyaml_vendor mimick_vendor orocos_kdl_vendor osrf_pycommon rcl_lifecycle rclcpp rclpy rcutils rmw_connextdds_common rosidl_cli rosidl_runtime_py rpyutils rti_connext_dds_cmake_module rviz_rendering sensor_msgs_py tf2_eigen_kdl uncrustify_vendor zstd_vendor
#0 7768.7   139 packages not processed
------
```

if someone managed to built it on ubuntu18/jp4 - would be great to hear how to do this.

SimeonOA, 2022-07-29:
Hi @asimonov, 

I had a similar rclcpp issue on my Ubuntu 20 install. 

Alsora's suggestion [here](https://github.com/ros2/rclcpp/issues/1963#issuecomment-1175880242) helped me fix that error. 

I am currently stuck on a rosbag2 error for which I have opened an issue [here](https://github.com/ros2/rosbag2/issues/1047)

YoushaaMurhij, 2022-12-20:
Hi, I am facing the same problem here. Could you please suggest a solution? I am using `Dockerfile.ros.humble`
Thanks

manato, 2023-02-14:
Hi, all. First of all, thank you very much for providing an awesome project and sharing information.
For future reference, I managed to build a docker image using `Dockerfile.ros.humble` on my environment (Jetpack 4.6) by the following modification. 

My environment is:
```bash
$ cat /etc/nv_tegra_release 
# R32 (release), REVISION: 6.1, GCID: 27863751, BOARD: t186ref, EABI: aarch64, DATE: Mon Jul 26 19:36:31 UTC 2021
```

The modifications are as follows:
```diff
diff --git a/Dockerfile.ros.humble b/Dockerfile.ros.humble
index 25e294d..dbef9fc 100644
--- a/Dockerfile.ros.humble
+++ b/Dockerfile.ros.humble
@@ -35,6 +35,7 @@ RUN apt-get update && \
 		gnupg2 \
 		lsb-release \
 		ca-certificates \
+		software-properties-common \
     && rm -rf /var/lib/apt/lists/* \
     && apt-get clean
 
@@ -67,6 +68,21 @@ RUN apt-get update && \
     && rm -rf /var/lib/apt/lists/* \
     && apt-get clean
 
+#
+# upgrade gcc / g++ to version 11
+# this is needed to build rclcpp
+#
+RUN add-apt-repository -y ppa:ubuntu-toolchain-r/test && \
+    apt-get update && \
+    apt-get install -y --no-install-recommends \
+    gcc-11 \
+    g++-11 \
+    && rm -rf /var/lib/apt/lists/* \
+    && apt-get clean
+
+RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 1 \
+    --slave /usr/bin/g++ g++ /usr/bin/g++-11
+
 # install some pip packages needed for testing
 RUN python3 -m pip install -U \
 		argcomplete \
@@ -80,7 +96,8 @@ RUN python3 -m pip install -U \
 		flake8-quotes \
 		pytest-repeat \
 		pytest-rerunfailures \
-		pytest
+		pytest \
+		pybind11-global
 
 # 
 # install OpenCV (with CUDA)
@@ -98,7 +115,6 @@ RUN cd /tmp && ./opencv_install.sh ${OPENCV_URL} ${OPENCV_DEB}
 #
 RUN apt-get update && \
     apt-get install -y --no-install-recommends \
-		  software-properties-common \
 		  apt-transport-https \
 		  ca-certificates \
 		  gnupg \
@@ -123,6 +139,15 @@ RUN cmake --version
 RUN apt purge -y python3.9 libpython3.9* || echo "python3.9 not found, skipping removal" && \
     ls -ll /usr/bin/python*
 
+#
+# Install newer yaml-cpp manually
+#
+RUN git -C /tmp clone https://github.com/jbeder/yaml-cpp.git -b yaml-cpp-0.6.0 \
+    && cmake -S /tmp/yaml-cpp -B /tmp/yaml-cpp/BUILD -DBUILD_SHARED_LIBS=ON \
+    && cmake --build /tmp/yaml-cpp/BUILD --parallel $(nproc --ignore=1)  \
+    && cmake --install /tmp/yaml-cpp/BUILD \
+    && rm -r /tmp/yaml-cpp
+    
     
 # 
 # download/build ROS from source
@@ -150,6 +175,7 @@ RUN mkdir -p ${ROS_ROOT}/src && \
 		image_transport \
 		compressed_image_transport \
 		compressed_depth_image_transport \
+		--exclude rmw_connextdds \
 		> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall && \
@@ -158,6 +184,9 @@ RUN mkdir -p ${ROS_ROOT}/src && \
     rm -r ${ROS_ROOT}/src/ament_cmake && \
     git -C ${ROS_ROOT}/src/ clone https://github.com/ament/ament_cmake -b ${ROS_DISTRO} && \
 
+    # Disable rmw_connextdds
+    sed -i '/rmw_connextdds/d' src/rmw_implementation/package.xml && \
+
     # install dependencies using rosdep
     apt-get update && \
     cd ${ROS_ROOT} && \
@@ -171,6 +200,13 @@ RUN mkdir -p ${ROS_ROOT}/src && \
     rm -rf /var/lib/apt/lists/* && \
     apt-get clean && \
 
+    # Upgrade pybind11 because the one installed via apt is too old to build humble
+    apt-get purge -y pybind11-dev && \
+    python3 -m pip install -U pybind11-global && \
+
+    # Remove libyaml-cpp-dev because the one installed via apt is too old to build humble
+    apt-get purge -y libyaml-cpp-dev && \
+
     # build it!
     colcon build \
         --merge-install \
```

**NOTE**: To pass the build process on Jetpack 4.6 (w/ Ubuntu 18.04 bionic), I **disabled** `rmw_connextdds`

Hope this would be helpful for those who face a similar problem :)

dusty-nv, 2023-05-20:
As part of a larger effort to move the ROS2 compilation into a script for easier control flow (now [`ros2_build.sh`](scripts/ros2_build.sh)), I've added containers for Humble on JetPack 4.6 in commit https://github.com/dusty-nv/jetson-containers/commit/9eaecf0012755bbe4734a1f4dd54018cf298c36a.

closed 2023-05-20 by dusty-nv