dusty-nv/jetson-containers #162: It says package not found after source /opt/ros/foxy/install/setup.bash, how can I solve this problem?

leeyunhome, 2022-06-16:
Hello, @dusty-nv 

Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
---
Finished <<< topic_service_action_rclpy_example [5.48s]

Summary: 13 packages finished [18.5s]

I built the package with the colcon build --symlink-install command.
It says package not found after source /opt/ros/foxy/install/setup.bash, how can I solve this problem?

Thank you.

enricosutera, 2022-06-16:
What do you mean? You should source the workspace, not ros. 
Where did you use `colcon build --symlink-install `?

leeyunhome, 2022-06-16:
Hello, @enricosutera 

in ~/robot_ws

Thank you.

leeyunhome, 2022-06-16:
> What do you mean? You should source the workspace, not ros. Where did you use `colcon build --symlink-install `?

Hello, @enricosutera 

after source ~/robot_ws/install/local_setup.bash
[image]

Thank you.

enricosutera, 2022-06-21:
You're welcome! I think you can close the issue yourself!

closed 2022-06-21 by leeyunhome