dusty-nv/jetson-containers #163: Can these containers(from jetson-containers repo) run examples from jetson-inference repo?

medphisiker, 2022-06-16:
Hello, Dusty.

There is a folder "/.jetson-inference" inside the container with ROS 2 foxy and PyTorch from "jetson-containers" repo.
I wanted to ask if it is possible to run demos with neural networks from your repository "jetson-inference"(https://github.com/dusty-nv/jetson-inference) from docker image with ROS 2 foxy and PyTorch from "jetson-containers" repo? For example from image "foxy-pytorch-l4t-r32.7.1"?

The examples from your "ros_deep_learning"(https://github.com/dusty-nv/ros_deep_learning) repository are also very interesting.
In this repository everything is installing and running inside Jetson Nano OS.

Is it possible to run ROS 2 nodes for deep learning from the "ros_deep_learning" repository from docker image with ROS2 and PyTorch from "jetson-containers" repo?  For example from image "foxy-pytorch-l4t-r32.7.1"?

I'm quite a beginner, I apologize if the answers to these questions are obvious to everyone =)

dusty-nv, 2022-06-16:
Hi @medphisiker, yes you should be able to run the jetson-inference applications straight away in the `ros:*pytorch*` containers.  That's because I use jetson-inference container as a base of those containers (the jetson-inference container coincidentally includes pytorch)

You would need to install the ros_deep_learning repo into the container though.  For an example of doing that, see here:  https://github.com/dusty-nv/jetbot_ros/blob/d8e5ee1b17f5c66d038017b86f8920a496197ea9/Dockerfile#L123

medphisiker, 2022-06-17:
Thank you for the quick response and instructions)  I will experiment)

medphisiker, 2022-06-26:
"I use jetson-inference container as a base of those containers (the jetson-inference container coincidentally includes pytorch)"
Yes, I already have docker image with ROS2 Foxy and PyTorch from your repository [foxy-pytorch-l4t-r32.7.1](https://hub.docker.com/layers/ros/dustynv/ros/foxy-pytorch-l4t-r32.7.1/images/sha256-cd52c461a2ba468d29db365771c4f27b32d9702a90283bd1a5c634a55c98557e?context=explore)

When I try run docker with ROS2 only, docker pull all layers for [foxy-ros-base-l4t-r32.7.1](https://hub.docker.com/layers/ros/dustynv/ros/foxy-ros-base-l4t-r32.7.1/images/sha256-18b318aff2f44036fc4627beb59c2b5d7eeab2218c89904a01a810da6b87c1b8?context=explore) from foxy-pytorch-l4t-r32.7.1.
Both of them works the same.

These images also have a problem that was discussed in this issue [link tyo issue](https://github.com/dusty-nv/jetson-inference/issues/1437).
But this is another story =). Thank you for you help and fast reply. 
There is the answer, and I think that we can close this issue )

closed 2025-02-21 by johnnynunez