dusty-nv/jetson-containers #167: Is it possible to use ros2 gui using docker container?

leeyunhome, 2022-06-23:
Hello, @dusty-nv 

I am using foxy ros2 container with jetpack 4.6
I couldn't install turtlesim with apt

How can I try turtlesim using the provided docker container?
Can you guide me?
Is it possible to use ros2 gui using docker container?

Thank you.

CourchesneA, 2022-06-23:
hi @leeyunhome , there is a script available that does the setup to use the display for you: https://github.com/dusty-nv/jetson-containers/blob/master/scripts/docker_run.sh . The arguments are documented and easy to understand, you should be able to use the `-c` to specify the ros2 container you want to use

dusty-nv, 2022-06-23:
Yea you can run GUI apps from containers, like @CourchesneA pointed to the run script automatically sets that up - however if your question is about ROS desktop apps in particular (i.e. ros-desktop distro), then I've encountered a bunch of issues trying to build those, so I just stick to ros-core on the Jetson and if needed run ros-desktop (i.e. rviz) from a PC (which is probably how you would do it with an actual robot anyways, where the Jetson doesn't have a display attached)

leeyunhome, 2022-06-27:
Hello, @dusty-nv , @CourchesneA 

I made a publisher running on ros2 with docker container (foxy) on jetson nano (jetpack4.6).

I'm trying to get ros-desktop to run on my pc as you pointed out, but I don't know how.

Can you give me a guide?
[image]

Thank you.

leeyunhome, 2022-06-27:
> @leeyunhome , I would suggest you go through the tutorials for ros2, they are a great source of information for everything ros-related. Your question does not seems related to the containers provided by this repo so it might be better to post it on the official Q&A: https://answers.ros.org/questions/. Once you have ros2 installed on you machine on the same network as your jetson, you should be able to see the node running on the jetson out-of-the-box using `ros2 node list`. Make sure you have running the container in `--network=host` mode

Thank you.

leeyunhome, 2022-06-27:
> Yea you can run GUI apps from containers, like @CourchesneA pointed to the run script automatically sets that up - however if your question is about ROS desktop apps in particular (i.e. ros-desktop [...]

> 

Hello, @CourchesneA 

1. Could you tell me how to designate /dev/i2c when I run docker container using this docker_run.sh?
example : sudo docker run --runtime nvidia -it --rm --device=/dev/i2c-0 --device=/dev/i2c-1 --device=/dev/i2c-2 --device=/dev/i2c-3 --device=/dev/phm_adc --network host ros:foxy-ros-base-l4t-r32.6.1

2.  Instead of docker_run.sh, can I just follow the script to set the display using the command below?
sudo docker run --runtime nvidia -it --rm --device=/dev/i2c-0 --device=/dev/i2c-1 --device=/dev/i2c-2 --device=/dev/i2c-3 --device=/dev/phm_adc --network host ros:foxy-ros-base-l4t-r32.6.1

Thank you.

CourchesneA, 2022-06-28:
I am not sure I understand your questions properly, what you posted correctly mount the I2C devices in the container. If you are wondering how to identify which device got assigned to your hardware, you could use udev rules

For your second question, if you want to access a GUI in the container you need more configuration. You can hower use GUI apps such as rqt on the host or another pc and visualize your data