dusty-nv/jetson-containers #170: CAN support

lucasvandijck, 2022-07-01:
Is it possible to access GPIO pins, specifically CAN, within a container?

dusty-nv, 2022-07-01:
Hi @lucasvandijck, I'm not entirely sure - if you [search](https://www.google.com/search?q=jetson+gpio+container+site:forums.developer.nvidia.com&rlz=1C1GCEB_enUS859US859&sxsrf=ALiCzsaPYvzm5iwHJojJmuoiVxfXIZStRQ:1656692620495&sa=X&ved=2ahUKEwiM5JX5jNj4AhVJI0QIHSovAeEQrQIoBHoECBEQBQ&biw=1920&bih=937&dpr=1) the NVIDIA Jetson forums for 'GPIO container', you will find some results.   However I think with CAN bus, you should be able to just mount the CAN device(s)

lucasvandijck, 2022-07-02:
Yes i first tried to access the GPIO directly using various methods mentioned on the forum, but without any luck. Hadn't thought about passing through the interface though, thank you!

closed 2022-07-02 by lucasvandijck