dusty-nv/jetson-containers #176: l4t-base Dockerfile for jetpack5

asimonov, 2022-08-03:
@dusty-nv ,

In the base container here (https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.jetpack) you reference BASE_IMAGE=nvcr.io/nvidia/l4t-base:r34.1.

This image is on NGC here: https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-base

Which says:
"Starting with the r32.4.3 release, the Dockerfile for the l4t-base docker image is also being provided. This can be accessed at this [link](https://gitlab.com/nvidia/container-images/l4t-base). Users can use this to modify the contents to suit their needs."

Unfortunately the repo https://gitlab.com/nvidia/container-images/l4t-base was not updated with jetpack5 version of l4t-base: https://gitlab.com/nvidia/container-images/l4t-base/-/blob/master/Dockerfile.l4t

Is it possible to update it?

I found it quite useful to see how to build the container from docker.io/arm64v8/ubuntu:18.04

dusty-nv, 2022-08-03:
Hi @asimonov, let me check with the team that maintains the l4t-base container

dusty-nv, 2022-08-04:
Hi @asimonov, the l4t-base dockerfile will be updated in the gitlab on the next JetPack release

asimonov, 2022-08-04:
@dusty-nv thanks for checking!

asimonov, 2022-09-08:
@dusty-nv 

I am afraid the L4T base docker repos were not updated with 5.0.2 release.

Can you ask the team to update?

dusty-nv, 2022-09-08:
@asimonov this is the merge request for it:   https://gitlab.com/nvidia/container-images/l4t-base/-/merge_requests/4

I will check on the status, sorry about that.

closed 2025-02-21 by johnnynunez