dusty-nv/jetson-containers #177: Can't find docker image for L4T 32.7.2

RasoulZamani, 2022-08-04:
Hi, Im using Jetson TX2 with L4t:32.7.2
I want to run docker “l4t-ml”

As you suggest I run : ./scripts/docker_build_ml.sh all
but I get error :

Sending build context to Docker daemon 185.9kB
Step 1/26 : ARG BASE_IMAGE=[nvcr.io/nvidia/l4t-base:r32.4.4](http://nvcr.io/nvidia/l4t-base:r32.4.4)
Step 2/26 : FROM ${BASE_IMAGE}
manifest for [nvcr.io/nvidia/l4t-base:r32.7.2](http://nvcr.io/nvidia/l4t-base:r32.7.2) not found: manifest unknown: manifest unknown

it seems there is no image for my L4t version (7.2),

so what do I do?

also I try to pull image directly (without scripts in git):
sudo docker pull [nvcr.io/nvidia/l4t-ml:r32.7.2-py3](http://nvcr.io/nvidia/l4t-ml:r32.7.2-py3)

but getting same error!
though I knew another version aren’t compatible, but I try pulling them! another version of docker image like 32.6.1, 32. 7.1 was tested and getting another bad error :

error pulling image configuration: error parsing HTTP 403 response body: invalid character ‘<’ looking for beginning of value: "\n Access Denied < … \nYou don’t have permission to access "http&#…

jarvinenemma, 2022-10-26:
Hi,

I added the line you suggested @dusty-nv and managed to proceed to Step 14/30. However, I get the following error:

`Step 14/30 : RUN git clone https://github.com/pytorch/vision torchvision &&     cd torchvision &&     git checkout ${TORCHVISION_VERSION} &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision
 ---> Running in d1a19ebfadc7
Cloning into 'torchvision'...
Note: checking out 'v0.10.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 300a8a401 backport fixes to build/push docs on tags (#4060)
Traceback (most recent call last):
  File "setup.py", line 12, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 196, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 149, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
The command '/bin/sh -c git clone https://github.com/pytorch/vision torchvision &&     cd torchvision &&     git checkout ${TORCHVISION_VERSION} &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision' returned a non-zero code: 1
`

I don't know how to proceed from here. Could you help me with this?

dusty-nv, 2022-10-26:
> OSError: libcurand.so.10: cannot open shared object file: No such file or directory

Hi @jarvinenemma, have you set your default docker runtime to nvidia?   See here:   https://github.com/dusty-nv/jetson-containers#docker-default-runtime

jarvinenemma, 2022-10-27:
Hi @dusty-nv ,
Yes, I set the default runtime successfully. Managed to proceed to Step 16 as I've explained in https://github.com/dusty-nv/jetson-containers/issues/202#issuecomment-1291864970 

Thanks for your reply! :)

AlexKlimaj, 2023-02-28:
What about r32.7.3 ? Getting the same error. Going to try r32.7.1

dusty-nv, 2023-03-01:
@AlexKlimaj, yes I recall that you should be able to run r32.7.1 container on r32.7.3 BSP

closed 2025-02-21 by johnnynunez