dusty-nv/jetson-containers #181: ROS Build failing

Tbarkin121, 2022-08-14:
I've been trying to build foxy and I've run into a build error with rosidl_runtime_cpp

[image]

I've started with a fresh SD card flashed with : 
NVIDIA Jetson Nano (Developer Kit Version)
 L4T 32.7.1 [ JetPack UNKNOWN ]
   Ubuntu 18.04.6 LTS
   Kernel Version: 4.9.253-tegra

Anyone know of a work around? I've done a little googling but I haven't run into an answer that has worked yet.

tonynajjar, 2022-08-16:
I'm getting a similar issue with another package:

```
Starting >>> bondcpp
--- stderr: bondcpp                            
CMake Error at /code/ros2_ws/install/bond/share/bond/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/code/ros2_ws/install/bond/share/bond/cmake/bond__rosidl_generator_cExport.cmake'
Call Stack (most recent call first):
  /code/ros2_ws/install/bond/share/bond/cmake/bondConfig.cmake:41 (include)
  CMakeLists.txt:13 (find_package) 
```

I only started to have it recently so I'm suspecting this [PR](https://github.com/ament/ament_cmake/pull/398/files) to be the cause but I'm not sure

tonynajjar, 2022-08-16:
I think I found the issue, basically a version mismatch between ament_cmake and cmake. The images install the latest cmake version (3.24 currently) and ament_cmake just got a [fix](rosinstall_generator) to support cmake 3.24. It'll be a while until the next sync to get that fix.
@dusty-nv I would propose to downgrade cmake, what do you think? I can submit a PR

dusty-nv, 2022-08-16:
Thanks Tony, sure that sounds worth a try to downgrade cmake

tonynajjar, 2022-08-16:
I'm facing a blocker while trying to test. I can't even build Dockerfile.ros.galactic on master. 

```
The command '/bin/bash -c apt-get purge -y '*opencv*' || echo "previous OpenCV installation not found" &&     mkdir opencv &&     cd opencv &&     wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${OPENCV_URL} -O ${OPENCV_DEB} &&     tar -xzvf ${OPENCV_DEB} &&     dpkg -i --force-depends *.deb &&     apt-get update &&     apt-get install -y -f --no-install-recommends &&     dpkg -i *.deb &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     cd ../ &&     rm -rf opencv &&     PYTHON3_VERSION=`python3 -c 'import sys; version=sys.version_info[:3]; print("{0}.{1}".format(*version))'` &&     cp -r /usr/include/opencv4 /usr/local/include/opencv4 &&     cp -r /usr/lib/python${PYTHON3_VERSION}/dist-packages/cv2 /usr/local/lib/python${PYTHON3_VERSION}/dist-packages/cv2' returned a non-zero code: 1
```

dusty-nv, 2022-08-16:
You could comment that out to try and proceed - what is the actual error from it?

tonynajjar, 2022-08-16:
I tried to comment it out but opencv is needed afterwards for some packages, so I get more errors. I can show you the actual error tomorrow. You can also try to do a quick build so we can make sure it's not some local issue I got.

dusty-nv, 2022-08-16:
I am trying to build Foxy container right now for JetPack 5.0.2 / L4T R35.1.0, and didn't have any issue with OpenCV.

However, I did get a similar cmake error to you, so will look into downgrading cmake as well:

``` bash
[91m--- stderr: rosidl_runtime_cpp
CMake Error at /opt/ros/foxy/install/share/performance_test_fixture/cmake/ament_cmake_export_targets-extras.cmake:18 (message):
  Failed to find exported target names in
  '/opt/ros/foxy/install/share/performance_test_fixture/cmake/performance_test_fixtureExport.cmake'
Call Stack (most recent call first):
  /opt/ros/foxy/install/share/performance_test_fixture/cmake/performance_test_fixtureConfig.cmake:41 (include)
  CMakeLists.txt:19 (find_package)
```

tonynajjar, 2022-08-16:
Here is the opencv error I'm getting:

```
Step 21/32 : RUN apt-get purge -y '*opencv*' || echo "previous OpenCV installation not found" &&     mkdir opencv &&     cd opencv &&     wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${OPENCV_URL} -O ${OPENCV_DEB} &&     tar -xzvf ${OPENCV_DEB} &&     dpkg -i --force-depends *.deb &&     apt-get update &&     apt-get install -y -f --no-install-recommends &&     dpkg -i *.deb &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     cd ../ &&     rm -rf opencv &&     PYTHON3_VERSION=`python3 -c 'import sys; version=sys.version_info[:3]; print("{0}.{1}".format(*version))'` &&     cp -r /usr/include/opencv4 /usr/local/include/opencv4 &&     cp -r /usr/lib/python${PYTHON3_VERSION}/dist-packages/cv2 /usr/local/lib/python${PYTHON3_VERSION}/dist-packages/cv2
 ---> Running in 9681b63e450f
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package *opencv*
E: Couldn't find any package by glob '*opencv*'
E: Regex compilation error - Invalid preceding regular expression
E: Couldn't find any package by regex '*opencv*'
previous OpenCV installation not found
OpenCV-4.5.0-aarch6 100%[===================>]  56.94M  23.7MB/s    in 2.4s    
OpenCV-4.5.0-aarch64-dev.deb
OpenCV-4.5.0-aarch64-libs.deb
OpenCV-4.5.0-aarch64-licenses.deb
OpenCV-4.5.0-aarch64-main.deb
OpenCV-4.5.0-aarch64-python.deb
OpenCV-4.5.0-aarch64-scripts.deb
Selecting previously unselected package opencv-dev.
(Reading database ... 42048 files and directories currently installed.)
Preparing to unpack OpenCV-4.5.0-aarch64-dev.deb ...
Unpacking opencv-dev (4.5.0) ...
Selecting previously unselected package opencv-libs.
Preparing to unpack OpenCV-4.5.0-aarch64-libs.deb ...
Unpacking opencv-libs (4.5.0) ...
Selecting previously unselected package opencv-licenses.
Preparing to unpack OpenCV-4.5.0-aarch64-licenses.deb ...
Unpacking opencv-licenses (4.5.0) ...
Selecting previously unselected package opencv-main.
Preparing to unpack OpenCV-4.5.0-aarch64-main.deb ...
Unpacking opencv-main (4.5.0) ...
Selecting previously unselected package opencv-python.
Preparing to unpack OpenCV-4.5.0-aarch64-python.deb ...
Unpacking opencv-python (4.5.0) ...
Selecting previously unselected package opencv-scripts.
Preparing to unpack OpenCV-4.5.0-aarch64-scripts.deb ...
Unpacking opencv-scripts (4.5.0) ...
Setting up opencv-licenses (4.5.0) ...
Setting up opencv-scripts (4.5.0) ...
dpkg: opencv-libs: dependency problems, but configuring anyway as you requested:
 opencv-libs depends on libatlas3-base; however:
  Package libatlas3-base is not installed.
 opencv-libs depends on libavcodec58 (>= 7:4.0); however:
  Package libavcodec58 is not installed.
 opencv-libs depends on libavformat58 (>= 7:4.2); however:
  Package libavformat58 is not installed.
 opencv-libs depends on libavutil56 (>= 7:4.0); however:
  Package libavutil56 is not installed.
 opencv-libs depends on libc6 (>= 2.29); however:
  Version of libc6:arm64 on system is 2.27-3ubuntu1.6.
 opencv-libs depends on libgcc-s1 (>= 4.2); however:
  Package libgcc-s1 is not installed.
 opencv-libs depends on libstdc++6 (>= 9); however:
  Version of libstdc++6:arm64 on system is 8.4.0-1ubuntu1~18.04.
 opencv-libs depends on libswscale5 (>= 7:4.0); however:
  Package libswscale5 is not installed.
 opencv-libs depends on libtbb2 (>= 2020.1); however:
  Package libtbb2 is not installed.
 opencv-libs depends on libtesseract4; however:
  Package libtesseract4 is not installed.

Setting up opencv-libs (4.5.0) ...
dpkg: opencv-main: dependency problems, but configuring anyway as you requested:
 opencv-main depends on libgcc-s1 (>= 3.0); however:
  Package libgcc-s1 is not installed.

Setting up opencv-main (4.5.0) ...
dpkg: opencv-python: dependency problems, but configuring anyway as you requested:
[... 106 lines cut ...]
  Package libtesseract4 is not installed.

dpkg: error processing package opencv-libs (--install):
 dependency problems - leaving unconfigured
Setting up opencv-licenses (4.5.0) ...
dpkg: dependency problems prevent configuration of opencv-main:
 opencv-main depends on libgcc-s1 (>= 3.0); however:
  Package libgcc-s1 is not installed.

dpkg: error processing package opencv-main (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of opencv-python:
 opencv-python depends on libgcc-s1 (>= 3.0); however:
  Package libgcc-s1 is not installed.

dpkg: error processing package opencv-python (--install):
 dependency problems - leaving unconfigured
Setting up opencv-scripts (4.5.0) ...
Errors were encountered while processing:
 opencv-dev
 opencv-libs
 opencv-main
 opencv-python
The command '/bin/bash -c apt-get purge -y '*opencv*' || echo "previous OpenCV installation not found" &&     mkdir opencv &&     cd opencv &&     wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${OPENCV_URL} -O ${OPENCV_DEB} &&     tar -xzvf ${OPENCV_DEB} &&     dpkg -i --force-depends *.deb &&     apt-get update &&     apt-get install -y -f --no-install-recommends &&     dpkg -i *.deb &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     cd ../ &&     rm -rf opencv &&     PYTHON3_VERSION=`python3 -c 'import sys; version=sys.version_info[:3]; print("{0}.{1}".format(*version))'` &&     cp -r /usr/include/opencv4 /usr/local/include/opencv4 &&     cp -r /usr/lib/python${PYTHON3_VERSION}/dist-packages/cv2 /usr/local/lib/python${PYTHON3_VERSION}/dist-packages/cv2' returned a non-zero code: 1
```

dusty-nv, 2022-08-16:
Hmm I have not seen that...which JetPack/L4T are you building on?

If you remove this `--skip-keys` line, you should also be able to comment out this OpenCV RUN command and workaround it that way for now.

tonynajjar, 2022-08-16:
Sorry I just realized that I completely skipped the Readme and assumed I can just build with docker build. I'll follow the instructions and report back

dusty-nv, 2022-08-16:
Ah okay - yea, the build scripts dynamically set a bunch of the docker args based on what JetPack-L4T version you are on (including the base image, OpenCV binary package, ect)

dusty-nv, 2022-08-16:
OK, got Foxy building again by downgrading cmake to version 3.23.3 and committed the fix in https://github.com/dusty-nv/jetson-containers/commit/d5e9d3aab9341ba4b66d01663256956a5b50d9bc

Will try the same thing with Galactic next

dusty-nv, 2022-08-17:
OK, Galactic built successfully with a similar patch:  https://github.com/dusty-nv/jetson-containers/commit/a7a878e6f3995dbc25acd710127156b4fae5a2aa

tonynajjar, 2022-08-17:
Thanks, I was also preparing a fix. My fix consisted mainly of:

`apt-get install cmake-data=3.23.2-0kitware1ubuntu18.04.1 cmake=3.23.2-0kitware1ubuntu18.04.1 && apt-mark hold cmake cmake-data`

With your fix are we sure that cmake can't be accidentally upgrade with other packages?

tonynajjar, 2022-08-17:
https://github.com/dusty-nv/jetson-containers/pull/183

dusty-nv, 2022-08-17:
> `apt-get install cmake-data=3.23.2-0kitware1ubuntu18.04.1 cmake=3.23.2-0kitware1ubuntu18.04.1`

Unfortunately this doesn't work when building the JetPack 5.x containers because those use Ubuntu 20.04 and the package names are different...

dusty-nv, 2022-08-17:
Closing this issue out with the updated PR https://github.com/dusty-nv/jetson-containers/pull/183

Thanks to @tonynajjar for all the help!

closed 2022-08-17 by dusty-nv

ChronoCookie, 2023-02-05:
Encountered this build error with Ubuntu 20.04 in a different context. Just wanted to leave that here in case anyone using Galactic and Ubuntu 20.04 stumbles across this Issue during troubleshooting as I did:

`sudo apt-get install cmake-data=3.23.2-0kitware1ubuntu20.04.1 cmake=3.23.2-0kitware1ubuntu20.04.1 && sudo apt-mark hold cmake cmake-data`

Admittedly just the same thing as already mentioned above.

> `apt-get install cmake-data=3.23.2-0kitware1ubuntu18.04.1 cmake=3.23.2-0kitware1ubuntu18.04.1 && apt-mark hold cmake cmake-data`