dusty-nv/jetson-containers #184: Jetson Nano container build failed with ROS2 Galactic + Pytorch

ParadoxRobotics, 2022-08-17:
Hi, 

I'm currently using a ROS2 Galactic container for my robotics project on my Jetson Nano 4Gb and I want it to add the machine learning framework Pytorch to it.
Unfortunately when build the container with ROS2 Galactic and Pytorch, I ended up in an OCI Runtime fail : 
```
failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown
```

Do you have any Idea why this error occur ?

Best regards !

Here is the full output : 
```
neurotronic@MARVIN01:~/jetson-containers$ ./scripts/docker_build_ros.sh --distro galactic --with-pytorch
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.6.1
l4t-base image:  nvcr.io/nvidia/l4t-base:r32.6.1
selecting OpenCV for L4T R32.6.1...
OPENCV_URL=https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
ROS_DISTRO:   galactic
ROS_PACKAGE:  ros_base
WITH_PYTORCH: on
WITH_SLAM:    off

Building container ros:galactic-ros-base-l4t-r32.6.1
BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.6.1

Building ros:galactic-ros-base-l4t-r32.6.1 container...
Sending build context to Docker daemon  188.4kB
[WARNING]: Empty continuation line found in:
    RUN mkdir -p ${ROS_ROOT}/src &&     cd ${ROS_ROOT} &&     rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG} 		launch_xml 		launch_yaml 		launch_testing 		launch_testing_ament_cmake 	demo_nodes_cpp 		demo_nodes_py 		example_interfaces 		camera_calibration_parsers 		camera_info_manager 		cv_bridge 		v4l2_camera 		vision_opencv 		vision_msgs 	image_geometry 		image_pipeline 		image_transport 		compressed_image_transport 		compressed_depth_image_transport 	> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     rm -r ${ROS_ROOT}/src/ament_cmake &&     git -C ${ROS_ROOT}/src/ clone https://github.com/ament/ament_cmake -b ${ROS_DISTRO} &&     apt-get update &&     cd ${ROS_ROOT} &&     rosdep init &&     rosdep update &&     rosdep install -y     	  --ignore-src        --from-paths src 	  --rosdistro ${ROS_DISTRO} 	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build &&     rm ${ROS_ROOT}/*.rosinstall
[WARNING]: Empty continuation lines will become errors in a future release.
Step 1/32 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.5.0
Step 2/32 : FROM ${BASE_IMAGE}
 ---> 1735192d3d51
Step 3/32 : ARG ROS_PKG=ros_base
 ---> Using cache
 ---> cdc45e7b376d
Step 4/32 : ENV ROS_DISTRO=galactic
 ---> Using cache
 ---> 8ef79422e79d
Step 5/32 : ENV ROS_ROOT=/opt/ros/${ROS_DISTRO}
 ---> Using cache
 ---> 8c9611056127
Step 6/32 : ENV ROS_PYTHON_VERSION=3
 ---> Using cache
 ---> e06a42dbca63
Step 7/32 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> 6bf69b0ce618
Step 8/32 : ENV SHELL /bin/bash
 ---> Using cache
 ---> 4c24f222ec96
Step 9/32 : SHELL ["/bin/bash", "-c"]
 ---> Using cache
 ---> 0b0d2c2fb40d
Step 10/32 : WORKDIR /tmp
 ---> Using cache
 ---> dd97348ff8ba
Step 11/32 : RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
 ---> Running in 2bd6302518e0
failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: error adding seccomp filter rule for syscall clone3: permission denied: unknown

```

dusty-nv, 2022-08-17:
Hi @ParadoxRobotics, you can find the fixes for this here:

* https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-962597353
* https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-995090398

If you still can't get it to work, then I would just re-flash the device and not do the `apt-get upgrade`

BTW, pre-built Galactic+PyTorch container images are on DockerHub here:  https://hub.docker.com/repository/docker/dustynv/ros/tags?page=1&ordering=last_updated&name=galactic-pytorch

ParadoxRobotics, 2022-08-17:
Thanks for your answer !
I will look at the pre-build on Dockerhub

closed 2025-02-21 by johnnynunez