dusty-nv/jetson-containers #186: Getting 401 unauthorized when trying to pull images

nakai-omer, 2022-08-18:
Tried a couple different images, all are returning `unauthorized: authentication required`

```
sudo docker pull nvcr.io/ea-linux4tegra/l4t-jetpack:r35.1.0
Error response from daemon: unauthorized: authentication required
```

Even trying to access the url manually in the browser causes a 401 error:

dusty-nv, 2022-08-18:
Sorry about that @nakai-omer, just switched these to the public URLs in https://github.com/dusty-nv/jetson-containers/commit/0b6c72e0718f34f8f5ee53bf6e2a548809ac3c80

Please try pulling the latest from jetson-containers repo and try again

nakai-omer, 2022-08-18:
wow you are the most amazing repo owner/maintainer ever!

closed 2022-08-18 by nakai-omer

dusty-nv, 2022-08-18:
Haha thanks, happy to help!  👍