dusty-nv/jetson-containers #187: TensorRT?

frankvp11, 2022-08-24:
Hi sorry if this question is kinda useless, but do all the containers ie (l4t-ml the pytorch and tensorflow ones) have tensorrt inside them? Also im kinda struggling to get the containers working the way I would like them to. How can I copy a couple of files from the host (the device) into the container, and do you happen to have a script that can install the TensorRT-OSS into it aswell? 

Also, when I use the ./scripts/docker_build_ml.sh script, it returns the error that I don't have skbuild installed when it is trying to install cmake. I am currently trying to do the Jetpack container so ill let you know how that goes.

dusty-nv, 2022-08-24:
Hi @frankvp11, on JetPack 4.x TensorRT automatically gets mounted into the containers from the host device when `--runtime nvidia` is used

On JetPack 5.x, TensorRT is installed inside the ML containers themselves

The easiest way to get files into the container is just to mount the directories with the `--volume` option

Unless you need to customize the ML based containers, these containers are already built and up on NGC so you can just pull them as opposed to needing to build them yourself

frankvp11, 2022-08-24:
Ok that makes sense. However, @dusty-nv when I run sudo docker run -it --runtime nvidia nvcr.io/nvidia/l4t-ml:r32.7.1-py3 and I try to run /trtexec command I get  the error that:
error while loading shared libraries: libnvinfer_plugin.so.8: cannot open shared object file: No such file or directory

dusty-nv, 2022-08-24:
Can you confirm that you have the libnvinfer libraries present both outside and inside the container?

``` bash
ls -ll /usr/lib/aarch64-linux-gnu/libnvinfer*
-rw-r--r-- 1 root root  30228664 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_builder_resource.so.8.2.1
lrwxrwxrwx 1 root root        26 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so -> libnvinfer_plugin.so.8.2.1
lrwxrwxrwx 1 root root        26 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.8 -> libnvinfer_plugin.so.8.2.1
-rw-r--r-- 1 root root  15459744 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.8.2.1
-rw-r--r-- 1 root root  18173206 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin_static.a
lrwxrwxrwx 1 root root        19 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer.so -> libnvinfer.so.8.2.1
lrwxrwxrwx 1 root root        19 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer.so.8 -> libnvinfer.so.8.2.1
-rw-r--r-- 1 root root 164944232 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer.so.8.2.1
-rw-r--r-- 1 root root 362217032 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_static.a
```

frankvp11, 2022-08-24:
With that being said- (moreso a docker question now) but do you know how I can basically take all my host device installations -ie TensorRT, Cuda10.2, cudnn, python and anything else that comes with the Jetpack 4.6.2 and use that as kinda of a base image or has those all installed exactly as it would be in the host? and then create kinda a simple dockerfile to copy a couple files and install a couple simple easy packages?

frankvp11, 2022-08-24:
This is on the host-
ecoation@ecoation-desktop:/media/ecoation/primary3/jetson-containers$ ls -ll /usr/lib/aarch64-linux-gnu/libnvinfer*
-rw-r--r-- 1 root root  30228664 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_builder_resource.so.8.2.1
lrwxrwxrwx 1 root root        26 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so -> libnvinfer_plugin.so.8.2.1
lrwxrwxrwx 1 root root        26 Aug 22 12:03 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.8 -> libnvinfer_plugin.so.8.4.3
-rwxr-xr-x 1 root root   7302560 Aug 22 12:03 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.8.4.3
-rw-r--r-- 1 root root  18173206 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin_static.a
lrwxrwxrwx 1 root root        19 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer.so -> libnvinfer.so.8.2.1
lrwxrwxrwx 1 root root        19 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer.so.8 -> libnvinfer.so.8.2.1
-rw-r--r-- 1 root root 164944232 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer.so.8.2.1
-rw-r--r-- 1 root root 362217032 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_static.a

This is in the container:
ecoation@ecoation-desktop:/media/ecoation/primary3/jetson-containers$ sudo docker run -it --runtime nvidia nvcr.io/nvidia/l4t-ml:r32.7.1-py3
allow 10 sec for JupyterLab to start @ http://172.17.0.2:8888 (password nvidia)
JupterLab logging location:  /var/log/jupyter.log  (inside the container)
root@74e896fbc46c:/# ls -ll /usr/lib/aarch64-linux-gnu/libnvinfer*
lrwxrwxrwx 1 root root        19 Aug 24 18:23 /usr/lib/aarch64-linux-gnu/libnvinfer.so -> libnvinfer.so.8.2.1
lrwxrwxrwx 1 root root        19 Aug 24 18:23 /usr/lib/aarch64-linux-gnu/libnvinfer.so.8 -> libnvinfer.so.8.2.1
-rw-r--r-- 1 root root 164944232 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer.so.8.2.1
-rw-r--r-- 1 root root  30228664 Nov 17  2021 /usr/lib/aarch64-linux-gnu/libnvinfer_builder_resource.so.8.2.1
lrwxrwxrwx 1 root root        26 Aug 24 18:23 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so -> libnvinfer_plugin.so.8.2.1
-rw-r--r-- 1 root root         0 Feb  4  2022 /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.8.2.1

dusty-nv, 2022-08-24:
> With that being said- (moreso a docker question now) but do you know how I can basically take all my host device installations -ie TensorRT, Cuda10.2, cudnn, python and anything else that comes with [...]

On JetPack 4.x, l4t-base should already include the JetPack packages like CUDA/cuDNN/TensorRT/ect when `--runtime nvidia` is used

frankvp11, 2022-08-24:
Hmm thats strange because TensorRT is still being tricky with me - referring to the issue mentioned above

dusty-nv, 2022-08-24:
> ecoation@ecoation-desktop:/media/ecoation/primary3/jetson-containers$ ls -ll /usr/lib/aarch64-linux-gnu/libnvinfer*
> -rw-r--r-- 1 root root 30228664 Nov 17 2021 [...]

OK, I think I see what is happening...at some point it appears that you installed a newer TRT plugins library `libnvinfer_plugin.so.8.4.3`

However this newer version isn't in `/etc/nvidia-container-runtime/host-files-for-container.d/tensorrt.csv`, because it doesn't come with JetPack

Try changing this line from `/etc/nvidia-container-runtime/host-files-for-container.d/tensorrt.csv`

`lib, /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.8.2.1`

to this:

`lib, /usr/lib/aarch64-linux-gnu/libnvinfer_plugin.so.8.4.3`

Then try running the container again

frankvp11, 2022-08-24:
Yes, I followed the instructions from https://github.com/NVIDIA-AI-IOT/deepstream_tao_apps/tree/master/TRT-OSS/Jetson 
so that I could have the updated tensorrt OSS. I will try running the container again in hopes of it succeeding. Give me a second

frankvp11, 2022-08-24:
Ok - It worked, however let me play around with it some more to see if it *truly* works. Im going to try and mount the volume that I would like and test if it creates an engine

frankvp11, 2022-08-24:
^ in reference to the above statement- trtexec worked. I have yet to convert, but thanks in advance

frankvp11, 2022-08-24:
@dusty-nv Thank you so much- It was able to convert the onnx into engine.

closed 2022-08-24 by frankvp11