dusty-nv/jetson-containers #190: `./scripts/docker_build_jetpack.sh` fail

SkalskiP, 2022-08-30:
Hi, when I'm running `./scripts/docker_build_jetpack.sh` on my Jetson Xavier it fails during the execution of this part of `Dockerfile.jetpack`:

```
#
# install CUDA Toolkit
#    
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
		  cuda-toolkit-* \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean
```

The error message that I'm getting is:

```
Unpacking cuda-nvrtc-10-2 (10.2.89-1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-HjrbFB/62-cuda-nvrtc-10-2_10.2.89-1_arm64.deb (--unpack):
 unable to create '/usr/local/cuda-10.2/targets/aarch64-linux/lib/libnvrtc.so.10.2.89.dpkg-new' (while processing './usr/local/cuda-10.2/targets/aarch64-linux/lib/libnvrtc.so.10.2.89'): Read-only file system
dpkg: error while cleaning up:
 unable to remove newly-extracted version of '/usr/local/cuda-10.2/targets/aarch64-linux/lib/libnvrtc.so.10.2.89': Read-only file system
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Selecting previously unselected package cuda-nvrtc-dev-10-2.
Preparing to unpack .../63-cuda-nvrtc-dev-10-2_10.2.89-1_arm64.deb ...
Unpacking cuda-nvrtc-dev-10-2 (10.2.89-1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-HjrbFB/63-cuda-nvrtc-dev-10-2_10.2.89-1_arm64.deb (--unpack):
 unable to create '/usr/local/cuda-10.2/targets/aarch64-linux/include/nvrtc.h.dpkg-new' (while processing './usr/local/cuda-10.2/targets/aarch64-linux/include/nvrtc.h'): Read-only file system
dpkg: error while cleaning up:
 unable to remove newly-extracted version of '/usr/local/cuda-10.2/targets/aarch64-linux/include/nvrtc.h': Read-only file system
...
```

@dusty-nv or anybody else, do you maybe happen to know why is that?

dusty-nv, 2022-08-30:
Hi, you shouldn't need this anymore, instead please use l4t-jetpack container from NGC.  I should remove it from this repo.

SkalskiP, 2022-08-31:
@dusty-nv do you mean this container: https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-jetpack? It only supports JetPack `5.x`, at least judging by available tags. I'm looking for a base image containing TensorRT that I could use with JetPack `4.5.0`.

dusty-nv, 2022-08-31:
> I'm looking for a base image containing TensorRT that I could use with JetPack `4.5.0`.

You should use `l4t-base` container with `--runtime nvidia` and TensorRT will automatically be mounted in from the host device.
The "fat containers" which have CUDA/cuDNN/TensorRT installed inside the containers themselves only started with JetPack 5.x

BTW I've removed the jetpack.dockerfile from jetson-containers repo in commit https://github.com/dusty-nv/jetson-containers/commit/39496f8eba51ababb0cfb625fc70410163e4fe43

SkalskiP, 2022-08-31:
Okay, I'll definitely try. I have a question, however. My goal is to make TensorRT accessible through python 3.8 installed in virtual env inside docker image. IS that possible?

dusty-nv, 2022-08-31:
I think you would have to rebuilt the TensorRT Python bindings for Python 3.8 - i haven't tried this.  I think they are on the TensorRT github though

SkalskiP, 2022-08-31:
Is that what you mean https://github.com/NVIDIA/TensorRT/tree/main/python?

dusty-nv, 2022-08-31:
Yes, I believe that's it

SkalskiP, 2022-08-31:
Thanks I'll give it a try!

closed 2025-02-21 by johnnynunez