dusty-nv/jetson-containers #191: Unable to reproduce opencv build

arpaterson, 2022-09-05:
I am unable to build OpenCV4.5.4 on a fresh JP5.0.2 image (AGX Orin Dev kit)
Not in a container.

I have the packages from `scripts` installed, and I'm using the same flags as in `dockerfile.opencv`.

It will not compile with opengl enabled due to 
`libopencv_core.so.4.5.4: undefined reference to `glXGetProcAddressARB'`
I know that particular error is not strictly related to jetson-containers.

Is opencv for jetpack / jetson-containers built strictly using the method in these files? Or have workarounds or patches been applied?

I can see the binaries are built and stored elsewhere, then downloaded, installed and tweaked.

arpaterson, 2022-09-05:
opencv 4.5.0 builds with those flags (`-D WITH_OPENGL=ON`), but the GUI configuration shows
```
--   GUI:  
--     GTK+:                        YES (ver 3.24.20) 
--       GThread :                  YES (ver 2.64.6) 
--       GtkGlExt:                  NO 
--     OpenGL support:              NO 
--     VTK support:                 NO 
```

dusty-nv, 2022-09-05:
Hi @arpaterson, here are scripts that build OpenCV outside of container:  https://github.com/mdegans/nano_build_opencv/

nakai-omer, 2022-09-07:
@dusty-nv Running the script from https://github.com/mdegans/nano_build_opencv/, fails with the following error:
```
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 opencv-libs : Depends: libavcodec57 (>= 7:3.4.8) but it is not installable or
                        libavcodec-extra57 (>= 7:3.4.8) but it is not installable
               Depends: libavformat57 (>= 7:3.4.8) but it is not installable
               Depends: libavutil55 (>= 7:3.4.8) but it is not installable
               Depends: libswscale4 (>= 7:3.4.8) but it is not installable
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

```

closed 2025-02-21 by johnnynunez