dusty-nv/jetson-containers #192: No RGB frames with realsense2 D435 inside the container

harel314, 2022-09-15:
Hi,
I am trying to run 2 realsense cameras D435 and D435i with ros2 foxy (in order to implement shared memory with zero copy), and 
I am using the ```ros:foxy-ros-base-l4t-r35.1.0``` container, with:

ubuntu version 20.04.5 LTS
L4T version Linux vision-server 4.9.253-tegra
Build cuda_11.4.r11.4

I builded from source the realsense2 2.48.0 since it's the same version i use outside the container.

I modified the docker_run.sh so it may detect the cameras (and it works)
and i modified  the ```/etc/docker/daemon.json``` as mentioned in the guide.

The only problem is that when I run the camera's realsense-viewer or ros2 launch file, I don't get any RGB frames, only depth frames, with the following error:

[image]

[image]

I haven't modified the camera settings and it works fine outside the container.

any suggestions?

thanks in advance

dusty-nv, 2022-09-15:
Hi @harel314, sorry I don't have experience with RealSense cameras so I'm not sure what the issue is.

javadan, 2022-09-16:
@harel314 
Two realsense cameras might be more complicated than one... I'd unplug one of them for now.  

I've managed to get realsense D455 with RGB and depth working inside L4T containers.  You can try with --privileged in the docker run command, to rule out some permission issues.

It looks like you're running both a ROS realsense camera node, and realsense-viewer at the same time?

I suspect both of those would be trying to get a lock on the camera device.    

Just running the realsense-viewer, with one camera, should pick up the RGB and depth streams.  If that's not working stand-alone, then something's probably wrong with your container settings.  Chances are, realsense-viewer works by itself, and it's the ROS node grabbing the camera's device first.

harel314, 2022-09-18:
Thank you both for your comments ! It's working now :D

I managed to pass over it by building the camera from source WITHOUT using ```-DBUILD_WITH_CUDA=true```

For now I have two cameras working both with ros2 foxy wrapper, so it's good.

closed 2022-09-18 by harel314