dusty-nv/jetson-containers #194: OSError: libcurand.so.10: cannot open shared object file: No such file or directory

Basavaraj-PN, 2022-09-19:
Image: foxy-pytorch-l4t-r32.6.1

dusty-nv, 2022-09-19:
If you run `ls -ll /usr/local/cuda/lib64` outside of the container, are the libraries there?

It seems like your NVIDIA Runtime isn't mounting CUDA/ect into the container correctly.  They need to be installed on your device with JetPack.

Basavaraj-PN, 2022-09-19:
Yes they are there outside the container.

dusty-nv, 2022-09-19:
Do you have these files as well?

``` bash
ls -ll /etc/nvidia-container-runtime/host-files-for-container.d/
total 32
-rw-r--r-- 1 root root    26 May 23  2021 cuda.csv
-rw-r--r-- 1 root root  4250 Jul 13  2021 cudnn.csv
-rw-r--r-- 1 root root 12240 Feb  2  2022 l4t.csv
-rw-r--r-- 1 root root  1590 Jan 14  2022 tensorrt.csv
-rw-r--r-- 1 root root   325 Aug 11  2020 visionworks.csv
```

If it continues not working, you may want to re-install the NVIDIA container packages from apt, or re-flash your device.

Basavaraj-PN, 2022-09-19:
thanks you, what is the apt command to reinstall nvidia container package from apt?

dusty-nv, 2022-09-19:
Do an apt-cache search to identify the packages:

``` bash
$ apt-cache search nvidia-container
libnvidia-container-tools - NVIDIA container runtime library (command-line tools)
libnvidia-container0 - NVIDIA container runtime library
libnvidia-container1 - NVIDIA container runtime library
nvidia-container-csv-cuda - Jetpack CUDA CSV file
nvidia-container-csv-cudnn - Jetpack CUDNN CSV file
nvidia-container-csv-tensorrt - Jetpack TensorRT CSV file
nvidia-container-csv-visionworks - Jetpack VisionWorks CSV file
nvidia-container-runtime - NVIDIA container runtime
nvidia-container-toolkit - NVIDIA container runtime hook
nvidia-container - NVIDIA Container Meta Package
```

Then uninstall and re-install these

RobotBramhana, 2022-09-21:
I am also facing the same problem. I suspect that this occurs when we issue ```apt upgrade``` locally in jetson nano. Are there any specific versions for the above packages that should be installed (after/for downgrading)?

dusty-nv, 2022-09-21:
Hi @RobotBramhana, you can try this:  https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-995090398

Basavaraj-PN, 2022-09-21:
I will try to

peredwardsson, 2022-10-05:
> If you run ls -ll /usr/local/cuda/lib64 outside of the container, are the libraries there?
If this is not true, and I run this script on a Jetson AGX Xavier, how do I install these libs?

dusty-nv, 2022-10-05:
> If this is not true, and I run this script on a Jetson AGX Xavier, how do I install these libs?

You should be able to do `sudo apt-get install cuda-toolkit-*`

achbogga, 2023-05-04:
@dusty-nv Can you run these containers from an amd64 host with Desktop GPU like RTX series for example?

dusty-nv, 2023-05-04:
@achbogga you can, but I haven't tested it in a bit, and you would probably need to build it against a newer NGC pytorch base container to support RTX series.     

`docker pull dustynv/jetson-inference:22.06`
https://hub.docker.com/repository/docker/dustynv/jetson-inference/tags?page=1&ordering=last_updated&name=22

You can run the `docker/run.sh` and `docker/build.sh` scripts like you can on Jetson.  You can update the x86 base container version here:  https://github.com/dusty-nv/jetson-inference/blob/1bc786dffb052044289e6e5cee1b60e0e5ce15e6/docker/build.sh#L48

achbogga, 2023-05-05:
@dusty-nv I have a strict requirement to run ROS-melodic (Ubuntu 18.04 only supported) for my application. The Jetson HW I am using is AGX Orin 64GB dev kit which runs on Jetpack 5.0+ which runs on Ubuntu 20.04+. On top of this I need my system to use pytorch on python 3.8 along with getting ros-melodic to support python3.8! Realistically, what would be my options given all this information. Can you run Jetpack 4.6.1 l4t container on a host Jetpack 5.0+? If so, how should I proceed? Any help would be appreciated!

dusty-nv, 2023-05-05:
Hi @achbogga, you can't run JetPack 4 containers on JetPack 5 (not with GPU acceleration, at least), however you could try building Melodic from source (similar to how I build [Noetic from source](https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.noetic)).  I'm not sure if you would encounter errors or not building Melodic since 20.04 is newer than what it was released against.  I feel the bigger issue though is that ROS Melodic uses Python 2.7 not Python 3, no idea if it would just work rebuilding it, here is what I could find about it:

* https://answers.ros.org/question/295012/how-can-i-install-ros-melodic-with-python3/?answer=295088#post-id-295088
* https://gist.github.com/azidanit/9950aa5408acdbe25f0ec431654da8d6

achbogga, 2023-05-05:
> Hi @achbogga, you can't run JetPack 4 containers on JetPack 5 (not with GPU acceleration, at least), however you could try building Melodic from source (similar to how I build [Noetic from source](https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.noetic)). I'm not sure if you would encounter errors or not building Melodic since 20.04 is newer than what it was released against. I feel the bigger issue though is that ROS Melodic uses Python 2.7 not Python 3, no idea if it would just work or not rebuilding it - here is what I could find about it:
> 
> * https://answers.ros.org/question/295012/how-can-i-install-ros-melodic-with-python3/?answer=295088#post-id-295088
> * https://gist.github.com/azidanit/9950aa5408acdbe25f0ec431654da8d6

Thanks @dusty-nv I might end up using Jetpack 5 container with all the latest stuff and creating a web app with FastAPI to serve a ROS melodic client sitting in an 18.04 machine!

closed 2025-02-21 by johnnynunez