dusty-nv/jetson-containers #195: Trying to install ros2 humble desktop version in Jetson Xavier via Docker

TZECHIN6, 2022-09-21:
Hi Dusty, I just pull the pre-built docker and find out that it is a ros-base, not desktop version.

I found the Dockerfile you provided, is it simply changing the ARG ROS_PKG=ros_base >>> =desktop will do the job?

Thanks.

dusty-nv, 2022-09-21:
Running `scripts/docker_build_ros.sh --distro humble --package desktop` will build it with desktop.  However I'm not sure if all the desktop packages will build or not.

TZECHIN6, 2022-09-22:
Thanks dustynv, just tried to build with this command, and it took me several hours to build.

One think I find out is that an extra dock image was built `nvcr.io/nvidia/l4t-jetpack    r35.1.0    02d7eb2c1adb    6 weeks ago    10.4GB`. Is it normal?

Might I know why there is no an official desktop version of ROS has been pushed to dockerhub?

TZECHIN6, 2022-09-22:
Update:

1. running `scripts/docker_build_ros.sh --distro humble --package desktop` can successfully build the image (haven't tested the overall completeness)
2. running `scripts/docker_build_ros.sh --distro foxy --package desktop` fail to build, below is the error code:
```
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/c/cppcheck/cppcheck_1.90-4build1_arm64.deb  Connection failed [IP: 185.125.190.36 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y cppcheck] failed
```
```
The command '/bin/bash -c mkdir -p ${ROS_ROOT}/src &&     cd ${ROS_ROOT} &&     rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG} 		launch_xml 		launch_yaml 		launch_testing 		launch_testing_ament_cmake 		demo_nodes_cpp 		demo_nodes_py example_interfaces 		camera_calibration_parsers 		camera_info_manager 		cv_bridge 		v4l2_camera 		vision_opencv 		vision_msgs 		image_geometry 		image_pipeline 		image_transport 		compressed_image_transport 		compressed_depth_image_transport 		> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     rm -r ${ROS_ROOT}/src/ament_cmake &&     git -C ${ROS_ROOT}/src/ clone https://github.com/ament/ament_cmake -b ${ROS_DISTRO} &&     apt-get update &&     cd ${ROS_ROOT} &&     rosdep init &&     rosdep update &&     rosdep install -y     	  --ignore-src        --from-paths src 	  --rosdistro ${ROS_DISTRO} 	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build &&     rm ${ROS_ROOT}/*.rosinstall' returned a non-zero code: 1
```
As I am trying to install a driver for LiDAR, not sure does it support ROS2 humble or not, at least the official document say it support foxy.

dusty-nv, 2022-09-22:
> One think I find out is that an extra dock image was built `nvcr.io/nvidia/l4t-jetpack r35.1.0 02d7eb2c1adb 6 weeks ago 10.4GB`. Is it normal?

That wasn't built, it was pulled from NVIDIA NGC container repository.  It's the base container that's used.

> Might I know why there is no an official desktop version of ROS has been pushed to dockerhub?

In the past the desktop version didn't build...I will try again on jetpack 5.x

dusty-nv, 2022-09-22:
> E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/universe/c/cppcheck/cppcheck_1.90-4build1_arm64.deb  Connection failed [IP: 185.125.190.36 80]

It appears to be a network connection issue - you may want to try again

TZECHIN6, 2022-09-23:
Thanks for your reply dusty-nv. Please try building desktop version again on JetPack 5.x, I will appreciate it! As I am now facing a issue that the LiDAR driver seem only works on desktop version.

One more question to ask, when I execute the script/docker, is it pointing to that Dockerfile.ros.{distro} for the build? So, I could modify dockerfile and re-use the build script?

TZECHIN6, 2022-09-23:
After reading several documents, ROS2 galactic seem a better choice than foxy, therefore I choose to working with this distro for my proejct. However,... 

Error occurs when running `./scripts/docker_build_ros.sh --distro galactic --package desktop`
```
Starting >>> qt_gui_cpp
--- stderr: qt_gui_cpp
CMake Error at /usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: Python3_INCLUDE_DIRS Python3_LIBRARIES
  Development Development.Module Development.Embed) (found version "3.9.5")
Call Stack (most recent call first):
  /usr/share/cmake-3.24/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.24/Modules/FindPython/Support.cmake:3203 (find_package_handle_standard_args)
  /usr/share/cmake-3.24/Modules/FindPython3.cmake:490 (include)
  /opt/ros/galactic/install/share/python_qt_binding/cmake/sip_helper.cmake:8 (find_package)
  src/qt_gui_cpp_sip/CMakeLists.txt:56 (include)

---
Failed   <<< qt_gui_cpp [5.09s, exited with code 1]
Aborted  <<< urdf [39.0s]
Aborted  <<< fastrtps [33min 21s]
Aborted  <<< rviz_ogre_vendor [37min 25s]

Summary: 119 packages finished [48min 21s]
  1 package failed: qt_gui_cpp
  3 packages aborted: fastrtps rviz_ogre_vendor urdf
  15 packages had stderr output: console_bridge_vendor foonathan_memory_vendor google_benchmark_vendor iceoryx_posh libyaml_vendor mimick_vendor osrf_testing_tools_cpp pybind11_vendor qt_gui_cpp rti_connext_dds_cmake_module rviz_ogre_vendor shared_queues_vendor urdfdom_headers yaml_cpp_vendor zstd_vendor
  192 packages not processed
The command '/bin/bash -c mkdir -p ${ROS_ROOT}/src &&     cd ${ROS_ROOT} &&     rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG} 		launch_xml 		launch_yaml 		launch_testing 		launch_testing_ament_cmake 		demo_nodes_cpp 		demo_nodes_py 		example_interfaces 		camera_calibration_parsers 		camera_info_manager 		cv_bridge 		v4l2_camera 		vision_opencv 		vision_msgs 		image_geometry 		image_pipeline 		image_transport 		compressed_image_transport 		compressed_depth_image_transport 		> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     rm -r ${ROS_ROOT}/src/ament_cmake &&     git -C ${ROS_ROOT}/src/ clone https://github.com/ament/ament_cmake -b ${ROS_DISTRO} &&     apt-get update &&     cd ${ROS_ROOT} &&     rosdep init &&     rosdep update &&     rosdep install -y     	  --ignore-src        --from-paths src 	  --rosdistro ${ROS_DISTRO} 	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build &&     rm ${ROS_ROOT}/*.rosinstall' returned a non-zero code: 1
```
A package called `qt_gui_cpp` is unable to build. Any suggestion for this? Thanks~

dusty-nv, 2022-09-23:
> A package called `qt_gui_cpp` is unable to build. Any suggestion for this? Thanks~

I get the same error on galactic and foxy, i haven't been able to figure it out.
I was able to get humble-desktop building for jetpack 5.x though.

dusty-nv, 2022-09-23:
I'm trying a patch similar to this for Foxy to see if that helps:

https://github.com/dusty-nv/jetson-containers/commit/1dad0ee1762359ec5b98a7d0a289a1ab3f99897d#diff-dead3dd0c4b0b9ec576fd4a6b423d0a545bf00351f51c626e089c7d2e3d06c53R133

dusty-nv, 2022-09-23:
OK, it appears to be building now with commit https://github.com/dusty-nv/jetson-containers/commit/37d18d587c784165e9810f1aa754ec603dbf193e

TZECHIN6, 2022-09-24:
> 

Yes, I was also having the situation as you mentioned, Humble can be built for JetPack R35.1.0, however, it is not compatible with my LiDAR hardware drivers... Sadly I cant use Humble in my current project.

TZECHIN6, 2022-09-24:
> 

I will definitely try it out now! Thanks, dustynv~

I will try to build it for ROS2 Galactic and will give feedback here.

dusty-nv, 2022-09-24:
@TZECHIN6 FYI I've uploaded Foxy/Galactic/Humble desktop container images for JetPack 5:

https://hub.docker.com/repository/registry-1.docker.io/dustynv/ros/tags?name=desktop
https://github.com/dusty-nv/jetson-containers

TZECHIN6, 2022-09-29:
Thx @dusty-nv I have tested out the Dockerfile, only the opencv part is not succesfull, other works great. Do you have idea why?

Btw, I am new to docker, if I want to use your image from a start point, and add something new on top of it. Is it possible? I saw other saying that I could create a Dockerfile like below:
```
FROM <existing image name>

RUN apt-get update && install sth...
```

If I plan to do so, do I still need to add back the ros_entrypoint.sh? 

```
COPY ./packages/ros_entrypoint.sh /ros_entrypoint.sh

RUN sed -i \
    's/ros_env_setup="\/opt\/ros\/$ROS_DISTRO\/setup.bash"/ros_env_setup="${ROS_ROOT}\/install\/setup.bash"/g' \
    /ros_entrypoint.sh && \
    cat /ros_entrypoint.sh

RUN echo 'source ${ROS_ROOT}/install/setup.bash' >> /root/.bashrc

ENTRYPOINT ["/ros_entrypoint.sh"]
CMD ["bash"]
WORKDIR /
```

TZECHIN6, 2022-09-30:
> What's the error that you get regarding OpenCV?

For OpenCV issue, the first time when I executed the build script there was an error (sorry for not logging down at that moment), however, I commented it out and built the docker image successfully, then rerun again the build script. This time I didn't comment the OpenCV part, it successfully run.... Not sure why..

If I remember correctly, the error was talking about fail to add KEY.

> Yes, you can build your own docker container on top of mine

Thx dusty-nv, I will use the pull method to rebuild my container.

TZECHIN6, 2022-09-30:
One more "stupid" question:

For custom dockerfile with yours pre-built base image, instead of running the build script provided, a simple docker build -f dockerfile -t name would do the job?

dusty-nv, 2022-09-30:
> For custom dockerfile with yours pre-built base image, instead of running the build script provided, a simple docker build -f dockerfile -t name would do the job?

Yes, you can use the simple command - my build script won't build your dockerfile.  That build script is setup build my dockerfile and sets a bunch of build variables based on the system environment that it detects.

closed 2022-10-06 by TZECHIN6