dusty-nv/jetson-containers #198: Activate Jupterlab On Child Image Of L4T-ML

Willis-Aero, 2022-10-18:
I changed the BASE_IMAGE of the Noetic ROS container to l4t-ml as referenced here: 
https://github.com/dusty-nv/jetson-containers/issues/53

Parent image: l4t-ml:r32.7.1-py3
Child image: noetic-ros-base-l4t-r32.7.1-py3

Is there any way to activate Jupyterlab on the child image? 
I'm having trouble adapting the docker run command to activate Jupyterlab on the child image.

It still works correctly on the base image with the command:
docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.7.1-py3

Thank you for your time.

dusty-nv, 2022-10-19:
Hi @Willis-Aero, the reason this happens, is that the ROS containers specify their own docker entrypoint, for setting up the ROS environment:

https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/Dockerfile.ros.noetic#L71

So it's that [`ros_entrypoint.sh`](https://github.com/dusty-nv/jetson-containers/blob/master/packages/ros_entrypoint.sh) command that gets run instead of the command that launches jupyterlab.

To manually start JupyterLab, you can just run `jupyter lab --ip 0.0.0.0 --port 8888 --allow-root` in the container

Or if you prefer, you can add these commands to `jetson-containers/packages/ros_entrypoint.sh` and rebuild the container:

https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/Dockerfile.ml#L154

Willis-Aero, 2022-10-19:
Thank you @dusty-nv for the assistance. The terminal command worked for activating jupyterlab in the container.

There is one last issue if you could help as the command results in an error:
No web browser found: could not locate runnable browser.

Willis-Aero, 2022-10-19:
> Hi @Willis-Aero, the reason this happens, is that the ROS containers specify their own docker entrypoint, for setting up the ROS environment: [...]

@dusty-nv 

Can l get assistance on the correct bash structure? 

Line 11:

`
echo "ROS_DISTRO $ROS_DISTRO"

do
        jupyter lab --ip 0.0.0.0 --port 8888 --allow-root &> /var/log/jupyter.log
	echo "allow 10 sec for JupyterLab to start @ http://$(hostname -I | cut -d' ' -f1):8888 (password nvidia)" 
	echo "JupterLab logging location:  /var/log/jupyter.log  (inside the container)" 
done

exec "$@"
`