dusty-nv/jetson-containers #202: OSError: libcurand.so.10: cannot open shared object file: No such file or directory

jarvinenemma, 2022-10-26:
Hi,

I added the line you suggested @dusty-nv and managed to proceed to Step 14/30. However, I get the following error:

`
Step 14/30 : RUN git clone https://github.com/pytorch/vision torchvision &&     cd torchvision &&     git checkout ${TORCHVISION_VERSION} &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision
 ---> Running in d1a19ebfadc7
Cloning into 'torchvision'...
Note: checking out 'v0.10.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 300a8a401 backport fixes to build/push docs on tags (#4060)
Traceback (most recent call last):
  File "setup.py", line 12, in <module>
    import torch
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 196, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 149, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
The command '/bin/sh -c git clone https://github.com/pytorch/vision torchvision &&     cd torchvision &&     git checkout ${TORCHVISION_VERSION} &&     python3 setup.py install &&     cd ../ &&     rm -rf torchvision' returned a non-zero code: 1
`

I don't know how to proceed from here. Could you help me with this?

Here's nv_tegra_release:
`cat /etc/nv_tegra_release
R32 (release), REVISION: 7.2, GCID: 30192233, BOARD: t210ref, EABI: aarch64, DATE: Wed Apr 20 21:34:48 UTC 2022`

_Originally posted by @jarvinenemma in https://github.com/dusty-nv/jetson-containers/issues/177#issuecomment-1291703942_

jarvinenemma, 2022-10-26:
Resolved the error using https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-995090398

Remember to restart docker after performing the aforementioned steps!
`sudo systemctl daemon-reload`
`sudo systemctl restart docker`

I'm trying to build the ML-containers again, and the error in Step 14 disappeared. I'll update this thread after the build completes.

dusty-nv, 2022-10-26:
Hi @jarvinenemma, can you try adding a `RUN pip3 install --verbose scikit-build` before in the dockerfile?

jarvinenemma, 2022-10-27:
Thanks for the tip. I'll try that later today.

CodiTrixta, 2023-01-31:
Hi, im getting the same issue now. 

ive tried this which still didnt resolve it. 

> Hi @jarvinenemma, can you try adding a `RUN pip3 install --verbose scikit-build` before in the dockerfile?

Will try to reinstall docker as mentioned here and will let you know if that resolves it

> Resolved the error using [#108 (comment)](https://github.com/dusty-nv/jetson-containers/issues/108#issuecomment-995090398)
> 
> Remember to restart docker after performing the aforementioned steps! `sudo systemctl daemon-reload` `sudo systemctl restart docker`
> 
> I'm trying to build the ML-containers again, and the error in Step 14 disappeared. I'll update this thread after the build completes.

CodiTrixta, 2023-01-31:
Ive tried the solution above with still no success. 
I already had the latest docker installed and nvidia is my default runtime and ive restarted

Please help

dusty-nv, 2023-01-31:
If you already tried adding `RUN pip3 install --verbose scikit-build` earlier in the dockerfile (before cmake gets upgraded), then I would recommend uncommenting this block of code:

https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/Dockerfile.pytorch#L108

and then commenting out this line:

https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/Dockerfile.pytorch#L118

CodiTrixta, 2023-02-01:
Im failing here, before even getting to cmake

> https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/Dockerfile.pytorch#L79

CodiTrixta, 2023-02-01:
My Issue is slightly different then the above issue ive realised, I get a segmentation fault.
```
Cloning into 'torchvision'...
Note: checking out 'v0.10.0'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 300a8a4013 backport fixes to build/push docs on tags (#4060)
Segmentation fault (core dumped)
```
When I look at my usage at the time im not at 60% memory usage

dusty-nv, 2023-02-01:
@CodiTrixta if you check the dmesg kernel log after this occurs, does it mention anything about being out of memory (OOM) or otherwise?  If so, you may want to mount additional swap memory.

Feki-Tech, 2023-03-13:
i'm using this BASE_IMAGE="nvcr.io/nvidia/l4t-base:r32.7.1. By running the following command ./scripts/docker_build_ml.sh pytorch i received this error in step 9: ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.7.2
L4T Base Image:   nvcr.io/nvidia/l4t-base:r32.7.1
selecting OpenCV for L4T R32.7.2...
OPENCV_URL=https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
Python3 version:  3.6
building PyTorch torch-1.9.0-cp36-cp36m-linux_aarch64.whl, torchvision v0.10.0, torchaudio v0.9.0, cuda arch 5.3;6.2;7.2
Building l4t-pytorch:r32.7.2-pth1.9-py3 container...
[sudo] password for velox: 
Sending build context to Docker daemon  179.7kB
Step 1/30 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.4.4
Step 2/30 : FROM ${BASE_IMAGE}
 ---> 7a58a57ab551
Step 3/30 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> 61db3c30c3d1
Step 4/30 : RUN apt-get update &&     apt-get install -y --no-install-recommends             python3-pip 		  python3-dev 		  libopenblas-dev 		  libopenmpi-dev             openmpi-bin             openmpi-common 		  gfortran 		  libomp-dev     && rm -rf /var/lib/apt/lists/*     && apt-get clean
 ---> Using cache
 ---> fc4afd7976b9
Step 5/30 : RUN pip3 install --no-cache-dir setuptools Cython wheel
 ---> Using cache
 ---> decb5dbf1045
Step 6/30 : RUN pip3 install --no-cache-dir --verbose numpy
 ---> Using cache
 ---> abfe5fb114e4
Step 7/30 : ARG PYTORCH_URL=https://nvidia.box.com/shared/static/lufbgr3xu2uha40cs9ryq1zn4kxsnogl.whl
 ---> Using cache
 ---> 2648c66700b5
Step 8/30 : ARG PYTORCH_WHL=torch-1.2.0-cp36-cp36m-linux_aarch64.whl
 ---> Using cache
 ---> 0b1cc0d07682
Step 9/30 : RUN wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${PYTORCH_URL} -O ${PYTORCH_WHL} &&     pip3 install --no-cache-dir --verbose ${PYTORCH_WHL} &&     rm ${PYTORCH_WHL}
 ---> Running in 93f82c7f42a4
failed to create shim: OCI runtime create failed: runc create failed: unable to start container process: exec: "/bin/sh": stat /bin/sh: no such file or directory: unknown

Feki-Tech, 2023-03-13:
I also had a similar problem by executing the following command. 
./scripts/docker_build_ml.sh tensorflow : 
Testsuite summary for Protocol Buffers 3.19.4
============================================================================
# TOTAL: 7
# PASS:  6
# SKIP:  0
# XFAIL: 0
# FAIL:  1
# XPASS: 0
# ERROR: 0

The command '/bin/sh -c cd /tmp &&     wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${PROTOBUF_URL}/$PROTOBUF_DIR.zip &&     wget --quiet --show-progress --progress=bar:force:noscroll --no-check-certificate ${PROTOBUF_URL}/$PROTOC_DIR.zip &&     unzip ${PROTOBUF_DIR}.zip -d ${PROTOBUF_DIR} &&     unzip ${PROTOC_DIR}.zip -d ${PROTOC_DIR} &&     cp ${PROTOC_DIR}/bin/protoc /usr/local/bin/protoc &&     cd ${PROTOBUF_DIR}/protobuf-${PROTOBUF_VERSION} &&     ./autogen.sh &&     ./configure --prefix=/usr/local &&     make -j$(nproc) &&     make check -j4 &&     make install &&     ldconfig &&     cd python &&     python3 setup.py build --cpp_implementation &&     python3 setup.py test --cpp_implementation &&     python3 setup.py bdist_wheel --cpp_implementation &&     cp dist/*.whl /opt &&     pip3 install dist/*.whl &&     cd ../../../ &&     rm ${PROTOBUF_DIR}.zip &&     rm ${PROTOC_DIR}.zip &&     rm -rf ${PROTOBUF_DIR} &&     rm -rf ${PROTOC_DIR}' returned a non-zero code: 2

dusty-nv, 2023-03-14:
> failed to create shim: OCI runtime create failed: runc create failed: unable to start container process: exec: "/bin/sh": stat /bin/sh: no such file or directory: unknown

@Feki-Tech as a test, are you able to run the `nvcr.io/nvidia/l4t-base:r32.7.1` container with GPU enabled?  Like this:

``` bash
scripts/docker_run.sh -c nvcr.io/nvidia/l4t-base:r32.7.1
```

CodiTrixta, 2023-03-14:
> @CodiTrixta if you check the dmesg kernel log after this occurs, does it mention anything about being out of memory (OOM) or otherwise? If so, you may want to mount additional swap memory.

Hi Dusty, sorry for not replying, yea adding additional swap memory would of helped but I ended up just building everything that I needed locally and not using docker. Funny enough I did add swap memory anyways as I needed to use more

Feki-Tech, 2023-04-03:
> > failed to create shim: OCI runtime create failed: runc create failed: unable to start container process: exec: "/bin/sh": stat /bin/sh: no such file or directory: unknown
> 
> @Feki-Tech as a test, are you able to run the `nvcr.io/nvidia/l4t-base:r32.7.1` container with GPU enabled? Like this:
> 
> ```shell
> scripts/docker_run.sh -c nvcr.io/nvidia/l4t-base:r32.7.1
> ```

this working without any problems

closed 2025-02-21 by johnnynunez