dusty-nv/jetson-containers #204: Error installing cv_bridge package

dtoopran, 2022-10-28:
Hi I am using the pre-built image `dustynv/ros:noetic-pytorch-l4t-r34.1.1`. When I try to install cv_bridge with `apt-get install ros-noetic-cv-bridge`, it fails with this error:
`Selecting previously unselected package libopencv-dev.
Preparing to unpack .../70-libopencv-dev_4.5.4-8-g3e4c170df4_arm64.deb ...
Unpacking libopencv-dev (4.5.4-8-g3e4c170df4) ...
dpkg: error processing archive /tmp/apt-dpkg-install-TOxRmc/70-libopencv-dev_4.5.4-8-g3e4c170df4_arm64.deb (--unpack):
 trying to overwrite '/usr/bin/opencv_annotation', which is also in package opencv-dev 4.5.0
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Selecting previously unselected package python3-opencv.
Preparing to unpack .../71-python3-opencv_4.2.0+dfsg-5_arm64.deb ...
Unpacking python3-opencv (4.2.0+dfsg-5) ...
Selecting previously unselected package ros-noetic-cv-bridge.
Preparing to unpack .../72-ros-noetic-cv-bridge_1.16.2-1focal.20221003.133755_arm64.deb ...
Unpacking ros-noetic-cv-bridge (1.16.2-1focal.20221003.133755) ...
Selecting previously unselected package proj-bin.
Preparing to unpack .../73-proj-bin_6.3.1-1_arm64.deb ...
Unpacking proj-bin (6.3.1-1) ...
Errors were encountered while processing:
 /tmp/apt-dpkg-install-TOxRmc/48-libopencv_4.5.4-8-g3e4c170df4_arm64.deb
 /tmp/apt-dpkg-install-TOxRmc/70-libopencv-dev_4.5.4-8-g3e4c170df4_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)`

dusty-nv, 2022-10-28:
Hi @dtoopran, can you try uninstalling the existing opencv packages before installing cv-bridge?   `apt-get purge -y '*opencv*'`

See this thread also:  https://github.com/dusty-nv/jetson-containers/issues/158

dtoopran, 2022-10-28:
Yes, that works. But later if I try to `import cv2`, then we get this error: `ImportError: libopencv_intensity_transform.so.4.5: cannot open shared object file: No such file or directory` Do I have to reinstall the OpenCV packages?

dusty-nv, 2022-10-28:
You may want to uninstall/reinstall cv2, or install my original opencv packages again

dusty-nv, 2022-11-02:
@fettahyildizz here is how I install OpenCV+CUDA build in other containers:   https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/Dockerfile.pytorch#L157

There are scripts and pre-built binaries that do it so that you don't need to build it from source.   BTW the binaries URL for JetPack 5 is here:   https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/scripts/opencv_version.sh#L14   (this normally gets set dynamically in the Dockerfile in my build scripts based on the JetPack version that's detected on the system)

phideltaee, 2023-02-08:
I solved it: I was using the wrong OPENCV_URL for the l4t-r35 version. Solved by running the following in the terminal. 

```
OPENCV_URL="https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz" 
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
```

Then running 
`cd /tmp && ./opencv_install.sh ${OPENCV_URL} ${OPENCV_DEB}`

nslewis-cat, 2023-06-13:
I did the same as you, but I think ./opencv_install.sh ${OPENCV_URL} ${OPENCV_DEB} uninstall ros-noetic-cv-bridge for me.

tgjantos, 2024-01-12:
I stumbled upon this issue and found a potentially more sophisticated solution thanks to @Chris-Bee.

The main issue is that the cv_bridge package is installed through `apt-get install ros-noetic-cv-bridge` is not compatible with the OpenCV version provided by JetPack or the Docker image. Therefore, the easiest solution, as described [here](https://github.com/aau-cns/insane_dataset_tools/tree/main#additional-steps-for-non-python-3-ros-installs-noetic), is to simply pull cv_bridge source-code and build it yourself. It will then find/use the installed OpenCV version

closed 2025-02-21 by johnnynunez