dusty-nv/jetson-containers #205: ROS Humble build fail

jenamax, 2022-11-10:
Hello! I am trying to build a docker container with ROS Humble on my PC by command ./scripts/docker_build_ros.sh --distro humble and get an error during the dependencies installation. 

The following packages were automatically installed and are no longer required:
  libjsoncpp1 librhash0 libuv1
Use 'apt autoremove' to remove them.
The following NEW packages will be installed:
  rti-connext-dds-6.0.1
0 upgraded, 1 newly installed, 0 to remove and 61 not upgraded.
Need to get 154 MB of archives.
After this operation, 551 MB of additional disk space will be used.
Get:1 http://packages.ros.org/ros2/ubuntu focal/main amd64 rti-connext-dds-6.0.1 amd64 6.0.1.25-1 [154 MB]
Preconfiguring packages ...
rti-connext-dds-6.0.1: RTI license agreement for non-commercial and pre-commercial use NOT accepted.
Fetched 154 MB in 19min 41s (130 kB/s)
Selecting previously unselected package rti-connext-dds-6.0.1.
(Reading database ... 68258 files and directories currently installed.)
Preparing to unpack .../rti-connext-dds-6.0.1_6.0.1.25-1_amd64.deb ...
config: RTI license agreement for non-commercial and pre-commercial use NOT accepted.
rti-connext-dds-6.0.1: RTI license agreement for non-commercial and pre-commercial use NOT accepted.
dpkg: error processing archive /var/cache/apt/archives/rti-connext-dds-6.0.1_6.0.1.25-1_amd64.deb (--unpack):
 new rti-connext-dds-6.0.1 package pre-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 /var/cache/apt/archives/rti-connext-dds-6.0.1_6.0.1.25-1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
ERROR: the following rosdeps failed to install                                                                        
  apt: command [apt-get install -y rti-connext-dds-6.0.1] failed                                                      
executing command [apt-get install -y python3-dev]
executing command [apt-get install -y python3-lark]
executing command [apt-get install -y python3-pytest-timeout]
executing command [apt-get install -y libtinyxml-dev]
executing command [apt-get install -y libconsole-bridge-dev]
executing command [apt-get install -y python3-mypy]
executing command [apt-get install -y python3-pytest-mock]
executing command [apt-get install -y pybind11-dev]
executing command [apt-get install -y pydocstyle]
executing command [apt-get install -y google-mock]
executing command [apt-get install -y libboost-python1.71.0]
executing command [apt-get install -y libboost-dev]
executing command [apt-get install -y libboost-python-dev]
executing command [apt-get install -y libssl-dev]
executing command [apt-get install -y graphviz]
executing command [apt-get install -y libbenchmark-dev]
executing command [apt-get install -y libzstd-dev]
executing command [apt-get install -y libeigen3-dev]
executing command [apt-get install -y liborocos-kdl-dev]
executing command [apt-get install -y uncrustify]
executing command [apt-get install -y python3-netifaces]
executing command [apt-get install -y python3-lxml]
executing command [apt-get install -y python3-cryptography]
executing command [apt-get install -y rti-connext-dds-6.0.1]
The command '/bin/bash -c mkdir -p ${ROS_ROOT}/src &&     cd ${ROS_ROOT} &&     rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG}              launch_xml              launch_yaml             launch_testing          launch_testing_ament_cmake            demo_nodes_cpp          demo_nodes_py           example_interfaces              camera_calibration_parsers            camera_info_manager             cv_bridge               v4l2_camera             vision_opencv                 vision_msgs             image_geometry          image_pipeline          image_transport      compressed_image_transport               compressed_depth_image_transport                > ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     rm -r ${ROS_ROOT}/src/ament_cmake &&     git -C ${ROS_ROOT}/src/ clone https://github.com/ament/ament_cmake -b ${ROS_DISTRO} &&     apt-get update &&     cd ${ROS_ROOT} &&     rosdep init &&     rosdep update &&     rosdep install -y          --ignore-src        --from-paths src    --rosdistro ${ROS_DISTRO}       --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build &&     rm ${ROS_ROOT}/*.rosinstall' returned a non-zero code: 1

dusty-nv, 2022-11-10:
Hi @jenamax, this error/package doesn't typically happen - I am guessing it's related to building on x86.  Regardless, it won't work building it on x86 for Jetson like that - if you want it to be for aarch64, you would need to build it on Jetson.   For x86, I would just use one of the upstream ROS containers.

jenamax, 2022-11-20:
Hi! Now I am trying to build ROS Humble image on Jetson TX2. I have installed Jetpack 4.6.2 with l4t 32.7.2. And I got hte similar error:

```
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Add distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rti_connext_dds_cmake_module: No definition of [rti-connext-dds-6.0.1] for OS version [bionic]
The command '/bin/bash -c mkdir -p ${ROS_ROOT}/src &&     cd ${ROS_ROOT} &&     rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG} 		launch_xml 		launch_yaml 		launch_testing 		launch_testing_ament_cmake 		demo_nodes_cpp 		demo_nodes_py 		example_interfaces 		camera_calibration_parsers 		camera_info_manager cv_bridge 		v4l2_camera 		vision_opencv 		vision_msgs 		image_geometry 		image_pipeline 		image_transport 		compressed_image_transport compressed_depth_image_transport 		> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     apt-get update &&     cd ${ROS_ROOT} &&     rosdep init &&     rosdep update &&     rosdep install -y     	  --ignore-src        --from-paths src 	  --rosdistro ${ROS_DISTRO} 	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build &&     rm ${ROS_ROOT}/*.rosinstall' returned a non-zero code: 1
```

dusty-nv, 2022-11-21:
You might want to try adding `rti-connext-dds` or similar to `--skip-keys`     
I'm not exactly sure what the ros key for rti-connext-dds-6.0.1 is
The default RWM is cyclone but it seems to want to install rti-connext-dds too