dusty-nv/jetson-containers #206: Humble container based on Jammy

ChaimLukasMaier-FDTech, 2022-11-25:
Hi,
I am trying to use ROS2 Humble inside of a docker container on a Jetson Xavier NX.

When downloading and running [dustynv/ros:humble-ros-base-l4t-r35.1.0](https://hub.docker.com/layers/dustynv/ros/humble-ros-base-l4t-r35.1.0/images/sha256-927ba32e701da41ee51e6b63989addc3d67a5938404a1dae9c7d1df996d835d8?context=explore) from docker hub, lsb_release -a tells me the image is 20.04 focal.
But the envvar ROS_DISTRO is properly seet to humble.

**The problem is that rosdep (and other parts of the ecosystem) do not propperly support humble on focal.**

Would it be possible to get a container that is based on Ubuntu 22.04?

dusty-nv, 2022-11-25:
> Would it be possible to get a container that is based on Ubuntu 22.04?

You can get those from OSRF but they won't support GPU.   GPU support is provided via l4t-base / l4t-jetpack based containers which are using Ubuntu 20.04.   Hence I build Humble from source in these containers.  Other packages you may need to build from source as well.

ckurtz22, 2023-04-24:
Is there any plans or ways to improve this? It makes ROS 2 Humble basically unusable, and Humble will be the only non end of life distro for ROS2 in a month.

nakai-omer, 2023-05-18:
@dusty-nv do you know if it will be in the start of Q3? This is a show stopper for us, we tried building from source, but every little update to a dependency causes a few hours of build time, because of having to build every single dep recursively from source.

ckurtz22, 2023-05-18:
> @dusty-nv do you know if it will be in the start of Q3? This is a show stopper for us, we tried building from source, but every little update to a dependency causes a few hours of build time, because of having to build every single dep recursively from source. 

The new jetpack won't fix this issue lol... It will only support the Orin platform because Nvidia loves making things unnecessarily complicated with little support...

closed 2025-02-21 by johnnynunez