dusty-nv/jetson-containers #207: Ros2 Humble was built but tests fails

sezer-muhammed, 2022-12-03:
Hi, I built ros2 humble.
But I cant source it with /opt/..../humble/.. command.
Also test failing.

I did not put SS because I think that I did something wrong.
After buiding ros2 humble, should I acces directly command line, or should I open docker etc?

dusty-nv, 2022-12-04:
Hi @sezer-muhammed, you need to run the docker container, it is built inside the container.

sezer-muhammed, 2022-12-04:
> Hi @sezer-muhammed, you need to run the docker container, it is built inside the container.

Hi Sir, May I ask what commands should I write in order to use ros2 humble? I did not get it.

sezer-muhammed, 2022-12-04:
> Hi @sezer-muhammed, you need to run the docker container, it is built inside the container.

Hi sir, I run the docker. There is no jetson_inference etc. I built jetson inference inside the docker from the source. Is this the way?

dusty-nv, 2022-12-05:
Hi @sezer-muhammed, you could use the `scripts/docker_run.sh` from jetson-containers:

``` bash
cd jetson-containers
scripts/docker_run.sh -c ros:humble-ros-base-l4t-r34.1.1
```

You may need to change the l4t version in the ros container tag depending on your jetpack-l4t version.  You can also use one of the pre-built images from here:  https://github.com/dusty-nv/jetson-containers#pre-built-container-images

closed 2025-02-21 by johnnynunez