dusty-nv/jetson-containers #208: Is it possible to build a contrainer with ubuntu 14.04 for chainer 1.17 on Jetson Xavier NX?

lx2u16, 2022-12-06:
Hi,

I am working on reproducing the BranchyNet paper [https://github.com/kunglab/branchynet](url) on my Jetson Xavier NX.

It requires old version software such as python2.7, chainer 1.17.0, etc. The chainer 1.17.0 installation guide recommends using Ubuntu OS 14.04.

I wonder if that's possible to build a container with ubuntu 14.04 for chainer 1.17 on Jetson Xavier NX?

Thanks.

dusty-nv, 2022-12-07:
@lx2u16 if you don't need GPU acceleration, then yes you can use any base container.  If you need GPU, then on Jetson it should be using l4t-base or l4t-jetpack, and for Xavier these use Ubuntu 18.04 or 20.04

dusty-nv, 2022-12-07:
For Xavier NX, they would support Ubuntu 18.04 for JetPack 4.x and Ubuntu 20.04 for JetPack 5.x

With such an old OS you may also run into incompatibilities with the CUDA/cuDNN versions and such if it is using those too

lx2u16, 2022-12-07:
@dusty-nv Thanks. If I only need CPUs, is the base container support Ubuntu 14.04? Do old boards support 14.04? e.g. Jetson Nano or others?

closed 2025-02-21 by johnnynunez