dusty-nv/jetson-containers #209: Is it possible to build on platform linux/amd64?

bluecamel, 2022-12-16:
I've tried running `./scripts/docker_build_ros.sh --distro galactic --package ros_base` and have a lot of issues.

First, `local-gen` is not found, so I installed `locales` through apt, but now it can't find `/usr/bin/python`.

The pytorch base image (`nvcr.io/nvidia/pytorch:22.04-py3`) seems to have python installed by conda (`/opt/conda/bin/python`).  The other base image (`nvcr.io/nvidia/l4t-base:r32.5.0`) has `/usr/bin/python3`, but still fails on the `update-alternatives` step:

```
Step 15/36 : RUN update-alternatives --install /usr/bin/python python /usr/bin/python3 1
 ---> Running in 5f98927dab86
update-alternatives: error: alternative path /usr/bin/python3 doesn't exist
```

Thanks so much for any help!

dusty-nv, 2022-12-16:
Hi @bluecamel, I haven't tested building the containers from this repo on x86 and I wouldn't expect them to work.   What I would do is just use the upstream ROS containers that already exist for x86:  https://hub.docker.com/_/ros/

bluecamel, 2022-12-19:
Got it, thanks!

closed 2022-12-19 by bluecamel

Brac24, 2023-02-26:
Just wanted to leave a comment for future reference that it is possible. Not the easiest thing to do but also not horrible but there is a learning curve in understanding Docker. I used the following references to get it setup: https://medium.com/@Smartcow_ai/building-arm64-based-docker-containers-for-nvidia-jetson-devices-on-an-x86-based-host-d72cfa535786 and https://www.stereolabs.com/docs/docker/building-arm-container-on-x86/

I mainly used the medium article for the most part. I've been able to build docker images for jetson nano using my x86 machine. I was a lot more motivated to get this working since I only had access to Jetson Nano which I once built rtabmap and rtabmap_ros from source and took me about 14 hours. Now I can not only build rtabmap but some of its larger dependencies from source in about 2 to 3 hours which is still a lot but a lot more tolerable if it is ever needed.

There's also this interesting git repo https://github.com/defunctzombie/jetson-nano-image-maker that builds jetson nano images using Docker on GitHub with GitHub Actions. So you might be able to reference this repo as a way to build your own docker images on GitHub instead of your local machine.