dusty-nv/jetson-containers #210: jetpack 5 + container + VPI = EGL_BAD_DISPLAY or must loigin to Ubuntu GUI

asimonov-TII, 2022-12-22:
hi @dusty-nv ,

Not sure if this is best place to ask, but may be you have an idea...

I am seeing the following issue: https://forums.developer.nvidia.com/t/jetpack-5-0-vpi2-0-run/213798

And it only happens in docker. If VPI is installed on host system then it runs fine in headless mode.
If I run a sample in docker it only runs if I login to Ubuntu GUI beforehand. It cannot be run in headless mode over ssh.

I am on jetpack 5.0.2 with docker container loosely based on you work here.
Inside container I install VPI 2.1.6 from nvidia repos.
Then i build and run provided VPI example 3 for Harris corner detection.

```
apt install -y nvidia-vpi vpi2-samples
...

root@roc0:~# cd /opt/nvidia/vpi2/samples/03-harris_corners/

root@roc0:/opt/nvidia/vpi2/samples/03-harris_corners# mkdir build
root@roc0:/opt/nvidia/vpi2/samples/03-harris_corners# cd build/
root@roc0:/opt/nvidia/vpi2/samples/03-harris_corners/build# cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable exact version "11.4") 
-- Found OpenCV: /usr (found version "4.5.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/nvidia/vpi2/samples/03-harris_corners/build
root@roc0:/opt/nvidia/vpi2/samples/03-harris_corners/build# make
[ 50%] Building CXX object CMakeFiles/vpi_sample_03_harris_corners.dir/main.cpp.o
[100%] Linking CXX executable vpi_sample_03_harris_corners
[100%] Built target vpi_sample_03_harris_corners

# running with Ubuntu GUI logged in:
root@roc0:/opt/nvidia/vpi2/samples/03-harris_corners/build# ./vpi_sample_03_harris_corners cuda ~/IMG_4759.jpg 

1394 keypoints found

#running after logging out from Ubuntu GUI:
root@roc0:/opt/nvidia/vpi2/samples/03-harris_corners/build# ./vpi_sample_03_harris_corners cuda ~/IMG_4759.jpg 
VPI_ERROR_INTERNAL: (EGL_BAD_DISPLAY)

```

mlovelacetcore, 2023-02-17:
I am seeing the same issue on Jetson AGX Xavier, running JetPack 5.0.2 as well as the ORIN platform running the JetPack 5.0.2.  I wish I had a fix to include. This reply is mostly to help me monitor the thread.

dusty-nv, 2023-02-19:
not entirely sure how to fix other than not running VPI samples headlessly, found this - https://forums.developer.nvidia.com/t/vpi-does-not-work-inside-of-docker-container/230192/10?u=dusty_nv

asimonov, 2023-02-20:
@dusty-nv that is a helpful thread, but not entirely. in my case i run my code written for VPI without need for display and it does crash without physical display attached. I think this is the case.

mlovelacetcore, 2023-02-20:
So, I found that running "unset DISPLAY" before executing my application seems to work when headless. You might try that and see if it gives results more like what we are expecting. I'm still experimenting with it so it might not be a complete fix.

asimonov-TII, 2023-02-21:
@mlovelacetcore yes, that works! 

DISPLAY is always set when i run the docker but is not set on host system. that was why VPI examples worked on host but not in docker.

closed 2023-02-21 by asimonov-TII