dusty-nv/jetson-containers #212: /tmp/.docker.xauth got destroyed when computer shutdown

TZECHIN6, 2022-12-30:
### Issue

As stated in the title, when using the `docker_run.sh` to start a container  **without** `--rm` and **not** terminated (container only stop not remove), if the computer shutdown and restarts, the user no longer be able to get access into container `docker exec -it <name> bash`. An error taking about the `/tmp/.docker.xauth` not being found.

If I `cd` into /tmp/ directory and list, this file is confirmed not exist...

By far the only way I solve it is to `docker rm <container>` and docker_run.sh again. This is not ideal as I need to compile the project every single time when using ROS2 `rosdep install` `colcon build`.

Any idea how should I fix it?

I could share the error message once I get access to my workstation.

dusty-nv, 2022-12-30:
Hi @TZECHIN6, I'm not entirely sure, but I believe you could either re-run these XAUTH-related commands to recreate the `/tmp/.docker.xauth` file:  

https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/scripts/docker_run.sh#L107

or you could disable the XAUTH stuff entirely by commenting out those commands, and removing these options from the `docker run` command in the script:  

https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/scripts/docker_run.sh#L115

TZECHIN6, 2023-01-01:
Thx @dusty-nv, I would like to know what will I expect if I comment out those lines? As I am using ROS2, most of the application requires GUI. Will this break it?

For method 1, I have tried but no luck, recreated it doesn't fix the `file cannot find` issue. Am I the only one having this issue, as I always use your run script for starting a docker with GUI application. And this problem hits every single time, so there might be a pattern or sth else.

Might be I will stick with method 2 by now... if removing those lines will not affect the use of GUI in docker.

dusty-nv, 2023-01-02:
Those lines are only for if you want to view GUI applications remotely (via SSH X11-tunnelling) that are running in the container.  If you have a display attached to your Jetson you shouldn't need it.

dusty-nv, 2023-01-02:
@TZECHIN6 you may also be able to change this line to create the file in somewhere other than `/tmp` - then it will survive a reboot:

https://github.com/dusty-nv/jetson-containers/blob/eb2307d40f0884d66310e9ac34633a4c5ef2e083/scripts/docker_run.sh#L108

TZECHIN6, 2023-01-03:
I just tested it out, after shutdown it is gone, however when I run `docker start <name>` if it can successfully start up. The `.docker.xauth` will be generated.

Do you think setting `ENV QT_X11_NO_MITSHM 1` in Dockerfile can solve this? 

I am now working on my office workstation, which it had this issue before, but now it is fixed somehow.... I did tweak the dockerfile a bit and current version seem is stable.

Thanks @dusty-nv , at least I know how to temporally solve this issue, although the root cause is still cannot be identified.

TZECHIN6, 2023-01-04:
Below is the run script I have been used for most of my docker container, thanks for your template!

```bash
#!/bin/bash

CONTAINER_IMAGE="osrf/ros:galactic-desktop-focal"

USER_VOLUME="/home/thomasluk/temp_ws/roscpp_tutorial_ws:/roscpp_tutorial"
USER_COMMAND=""

echo "CONTAINER_IMAGE: $CONTAINER_IMAGE"
echo "USER_VOLUME:     $USER_VOLUME"
echo "USER_COMMAND:    '$USER_COMMAND'"

# Give docker root user X11 permissions
xhost +local:docker

# Enable SSH X11 forwarding inside container (https://stackoverflow.com/q/48235040)
XAUTH=/tmp/.docker.xauth
xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -
chmod 777 $XAUTH

# Run the container
docker run --name roscpp-tutorial-devel -it --rm \
    -e XAUTHORITY=$XAUTH \
    -e DISPLAY=$DISPLAY \
    -v $XAUTH:$XAUTH \
    -v /tmp/.X11-unix/:/tmp/.X11-unix \
    -v $USER_VOLUME $CONTAINER_IMAGE $USER_COMMAND
```
However, the `xauth` always throw a error when running it from clean:
```bash
[sudo] password for thomasluk: 
CONTAINER_IMAGE: osrf/ros:galactic-desktop-focal
USER_VOLUME:     /home/thomasluk/temp_ws/roscpp_tutorial_ws:/roscpp_tutorial
USER_COMMAND:    ''
non-network local connections being added to access control list
xauth:  unable to rename authority file /tmp/.docker.xauth, use /tmp/.docker.xauth-n
```

closed 2023-03-07 by TZECHIN6