dusty-nv/jetson-containers #213: opencv-python does not have dist-info and also has a circular symlink

DuaneNielsen, 2022-12-30:
Hey, thanks for the images they are very useful.

One thing I noticed is that the cv2 module installed does not contain a dist-info directory.  This confuses pip and setuptools and creates a bit of mayhem when the user includes a package that requires opencv-python.  Since I think setuptools will go ahead and install another copy of cv2 on top of the cv2 that's already installed.

Specifically..

```
python3 -c "import importlib; print(importlib.util.find_spec('cv2'))"
```
ModuleSpec(name='cv2', loader=<_frozen_importlib_external.SourceFileLoader object at 0xffff98f0b280>, origin='/usr/local/lib/python3.8/dist-packages/cv2/__init__.py', submodule_search_locations=['/usr/local/lib/python3.8/dist-packages/cv2'])

```
pip3 list | grep opencv
```
produces no output, due to a missing dist-info directory in /usr/lib/python3.8/dist-packages/

more problematic is the setuptools if we run

```
cd /usr/lib/python3.8/dist-packages/
python3 -c "import setuptools; print(setuptools.find_packages('.'))"
```
we get...

['jetson', 'jetson_utils', 'jetson_inference', 'Jetson', 'cv2', 'tensorrt', 'uff', 'graphsurgeon', 'jetson.utils', 'jetson.inference', 'Jetson.Utils', 'Jetson.Inference', 'cv2.cv2', 'cv2.cv2.cv2', 'cv2.cv2.cv2.cv2', 'cv2.cv2.cv2.cv2.cv2', 'cv2.cv2.cv2.cv2.cv2.cv2', 'cv2.cv2.cv2.cv2.cv2.cv2.cv2', 'cv2.cv2.cv2.cv2.cv2.cv2.cv2.cv2', 'cv2.cv2.cv2.cv2.cv2.cv2.cv2.cv2.cv2'....

if we delete the self referential symlink we get...

```
rm cv2/cv2
python3 -c "import setuptools; print(setuptools.find_packages('.'))"
```
['cv2', 'jetson', 'jetson_utils', 'jetson_inference', 'Jetson', 'tensorrt', 'uff', 'graphsurgeon', 'jetson.utils', 'jetson.inference', 'Jetson.Utils', 'Jetson.Inference', 'uff.converters', 'uff.bin', 'uff.model', 'uff.converters.tensorflow']

now lets see if we can get the version of cv2

```
cd /usr/lib/python3.8/dist-packages/
python3 -c "import pkg_resources; print(pkg_resources.get_distribution('cv2').version)"
```
pkg_resources.DistributionNotFound: The 'cv2' distribution was not found and is required by the application

but for tensorrt in the same directory...

```
cd /usr/lib/python3.8/dist-packages/
python3 -c "import pkg_resources; print(pkg_resources.get_distribution('tensorrt').version)"
```
8.4.1.5

I'm guessing this is because tensorrt has a dist-info and cv2 does not... so even though we can import the module and use it.. we can't really build any setuptools based python modules on top of this...  I think.. I'm not an expert in python3 packages.. but that's what I'm observing when I try to install my own packages on top of this image.

For a workaround, I did figure out that you can upgrade opencv-python to 4.5.0.6 using pip, as long as you delete the existing opencv directories and symlinks...

```
rm -rf /usr/lib/python3.8/dist-packages/cv2 && rm /usr/local/lib/python3.8/dist-packages/cv2 && pip3 install opencv-python==4.5.5.64
```
So I just added this to the top of my Dockerfile...

```
FROM dustynv/ros:noetic-pytorch-l4t-r35.1.0

RUN rm -rf /usr/lib/python3.8/dist-packages/cv2 && rm /usr/local/lib/python3.8/dist-packages/cv2 && pip3 install opencv-python==4.5.5.64
```

Anyway.. hope this helps someone!

Cheers...

CourchesneA, 2023-01-17:
I think the issue here is that the version that pip provides is not compiled to use CUDA

dusty-nv, 2023-09-06:
After doing some digging, I found this is how opencv-python pip wheel is built:  https://github.com/opencv/opencv-python#manual-builds

Currently I just build/install opencv with CUDA in the way their docs describe it.  However, maybe that opencv-python wheel needs built too.  That alone would probably be insufficient though, for people using OpenCV from C++ are used to the normal install locations for C++.   Also it doesn't address a similar package conflict that occurs with apt (apt not recognizing that OpenCV is already installed).  It's kind of a mess the way OpenCV manages it TBH

closed 2025-02-21 by johnnynunez