dusty-nv/jetson-containers #215: Get L4T version in docker image

CourchesneA, 2023-01-11:
I would like to know how to get the L4T version in a docker image.

We used images built from jetpack 5.0, which is L4T r34. We are now using jetpack 5.1 and L4T r35, but it is sometime hard to verify which image is running in our machines since we now longer have the L4T version in the name of our image. Is there a way I can verify what is the L4T version of the image ?

I noticed there are two different ways to check listed here:
https://github.com/dusty-nv/jetson-containers/blob/master/scripts/l4t_version.sh
However, in the container the file `nv_tegra_release` is empty and the package `nvidia-l4t-core` is not installed

CourchesneA, 2023-01-11:
Thanks. In both of those case, that would give the L4T version of the host, is that correct ? We realized recently that we had a mismatch (running docker images from L4T R34 base while on a R35 host) and even though everything was working fine we want to make sure wwe can detect / prevent that in the future.

dusty-nv, 2023-01-11:
Yes, doing it dynamically like that would give you the L4T version of the host.

If you want the L4T version that the container was built for, then I would make another `ENV` in the dockerfile that gets set to the L4T version that the container was built with.

CourchesneA, 2023-01-11:
awesome, thanks !

closed 2023-01-11 by CourchesneA