dusty-nv/jetson-containers #216: GPG key failure on apt update

peredwardsson, 2023-01-20:
Unsure where this goes, feel free to close if inappropriate.

Here's me downloading `dustynv/ros:humble-ros-base-l4t-r35.1.0` and trying to run `apt update`. 
```
$docker run -it --network=host dustynv/ros:humble-ros-base-l4t-r35.1.0 apt update
sourcing   /opt/ros/humble/install/setup.bash
ROS_ROOT   /opt/ros/humble
ROS_DISTRO humble
Get:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease [265 kB]
Get:2 https://repo.download.nvidia.com/jetson/common r35.1 InRelease [2,555 B]
Get:3 http://packages.ros.org/ros2/ubuntu focal InRelease [4,685 B]
Get:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Get:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [108 kB]
Get:6 https://repo.download.nvidia.com/jetson/common r35.1/main arm64 Packages [19.1 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Get:8 http://ports.ubuntu.com/ubuntu-ports focal/multiverse arm64 Packages [139 kB]
Get:9 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 Packages [1,234 kB]
Get:10 https://apt.kitware.com/ubuntu focal InRelease [15.5 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports focal/restricted arm64 Packages [1,317 B]
Get:12 http://ports.ubuntu.com/ubuntu-ports focal/universe arm64 Packages [11.1 MB]
Get:13 http://packages.ros.org/ros2/ubuntu focal/main arm64 Packages [1,143 kB]
Err:10 https://apt.kitware.com/ubuntu focal InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 42D5A192B819C5DA
Get:14 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe arm64 Packages [1,193 kB]
Get:15 http://ports.ubuntu.com/ubuntu-ports focal-updates/multiverse arm64 Packages [9,068 B]
Get:16 http://ports.ubuntu.com/ubuntu-ports focal-updates/restricted arm64 Packages [5,265 B]
Get:17 http://ports.ubuntu.com/ubuntu-ports focal-updates/main arm64 Packages [2,102 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports focal-backports/universe arm64 Packages [27.8 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports focal-backports/main arm64 Packages [54.8 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports focal-security/multiverse arm64 Packages [3,252 B]
Get:21 http://ports.ubuntu.com/ubuntu-ports focal-security/main arm64 Packages [1,714 kB]
Get:22 http://ports.ubuntu.com/ubuntu-ports focal-security/universe arm64 Packages [892 kB]
Get:23 http://ports.ubuntu.com/ubuntu-ports focal-security/restricted arm64 Packages [5,003 B]
Reading package lists... Done
W: GPG error: https://apt.kitware.com/ubuntu focal InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 42D5A192B819C5DA
E: The repository 'https://apt.kitware.com/ubuntu focal InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```
It fails due to some GPG error. Any image I base on this image will have errors running `apt update`, which means that `apt update && ...` always fails. A quick google suggests that this is a manual fix of adding a public key (listed in the error message), so is this something that should be added to your image or is it a matter of waiting for someone else to update their public keys?

ikhann, 2023-01-22:
Hi!
Add this layer before `apt update`:

```
RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | tee /usr/share/keyrings/kitware-archive-keyring.gpg >/dev/null \
    && echo 'deb [signed-by=/usr/share/keyrings/kitware-archive-keyring.gpg] https://apt.kitware.com/ubuntu/ focal main' | tee /etc/apt/sources.list.d/kitware.list >/dev/null
```

Lets-a-go, 2023-01-22:
After updating your keyfile, if you wish to avoid this error in the future (kitware rotates keys periodically), you can install the kitware-archive-keyring package to automatically follow kitware's key rotations. It will put this keyring at /usr/share/keyrings/kitware-archive-keyring.gpg, hence the first command removing your manually-created file is necessary. To do it, just issue the following two commands from a command prompt with appropriate privileges:
  sudo rm /usr/share/keyrings/kitware-archive-keyring.gpg
  sudo apt-get install kitware-archive-keyring
 
BTW, the info for this and the comment above originates from https://apt.kitware.com This is step 4. Step 3 (the prior one) should have already been executed first to ensure everything is setup correctly (mainly from the 'echo' onward).

ikhann, 2023-01-23:
> > Hi! Add this layer before `apt update`:
> > ```
> > RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | tee /usr/share/keyrings/kitware-archive-keyring.gpg >/dev/null \
> >     && echo 'deb [signed-by=/usr/share/keyrings/kitware-archive-keyring.gpg] https://apt.kitware.com/ubuntu/ focal main' | tee /etc/apt/sources.list.d/kitware.list >/dev/null
> > ```
> 
> Could I suggest that this line be added to the base image, or is that not appropriate?

Hi,
Yes, you can also add it to the base image instead of the user-defined image

dusty-nv, 2023-01-23:
@ikhann @peredwardsson I'll add this to the next version of the ROS images (but for the current version, use your workaround).   Just do be clear, do I need to add this:

```
sudo rm /usr/share/keyrings/kitware-archive-keyring.gpg
sudo apt-get install kitware-archive-keyring
```

or this:

```
RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | tee /usr/share/keyrings/kitware-archive-keyring.gpg >/dev/null \
    && echo 'deb [signed-by=/usr/share/keyrings/kitware-archive-keyring.gpg] https://apt.kitware.com/ubuntu/ focal main' | tee /etc/apt/sources.list.d/kitware.list >/dev/null
```

or both?  (and in what order?)

ikhann, 2023-01-23:
> @ikhann @peredwardsson I'll add this to the next version of the ROS images (but for the current version, use your workaround). Just do be clear, do I need to add this:
>
> ```
> sudo rm [...]

Hi @dusty-nv!

I just added these two lines before `apt update`:

```
RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | tee /usr/share/keyrings/kitware-archive-keyring.gpg >/dev/null \
    && echo 'deb [signed-by=/usr/share/keyrings/kitware-archive-keyring.gpg] https://apt.kitware.com/ubuntu/ focal main' | tee /etc/apt/sources.list.d/kitware.list >/dev/null
```

this is enough to update the key from Kitware and keep working. 

You can also find some clarifications from NVIDIA [here](https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/issues/50#issuecomment-1399556235).

dusty-nv, 2023-01-23:
OK thanks @ikhann, I will follow the directions from https://apt.kitware.com/    sorry about that!

dusty-nv, 2023-02-17:
Changed the dockerfiles in https://github.com/dusty-nv/jetson-containers/commit/e8a04010c5369c5cfd13ebe112b957f2bdfefb28 for this issue

imstevenpmwork, 2023-02-21:
Hello!
I'm trying to build an image on top of: humble-desktop-l4t-r35.1.0
However, I get the exact same behavior described in this ticket. I tried adding the layer suggested by @ikhann before the apt update in my Dockefile, but I still got the same error GPG msg.
Am I missing something else?

Brac24, 2023-02-26:
Not sure if my issue is related but I have an issue with installing ros2 dependencies with rosdep when running docker build on my Dockerfile. I am using  dustynv/ros:foxy-ros-base-l4t-r35.2.1 as my base image. It starts off ok but then it seems to fail to install any of the dependencies. 
[image]
The image is a snippet but that same error shows up for every dependency it tries to install.

Looking online it seemed that it had to do with updating the repository keys but it did not work. But the weird thing is that it does work if I connect to the dustynv/ros:foxy-ros-base-l4t-r35.2.1 container and manually update the keys then I run rosdep it will properly install the ros2 dependencies for the ros2 packages I'm trying to build.

I am running this line: curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg to update the key. Not sure if it does properly update the key if I run it within the container versus while running the docker build of the image.

ggoretkin-bdai, 2023-03-20:
> Hi! Add this layer before `apt update`:
>
> ```
> RUN wget -O - https://apt.kitware.com/keys/kitware-archive-latest.asc 2>/dev/null | gpg --dearmor - | tee /usr/share/keyrings/kitware-archive-keyring.gpg [...]

I believe this is triggering a warning for us

```
W: Target Packages (main/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:50 and /etc/apt/sources.list.d/kitware.list:1
```

impala454, 2023-07-06:
Adding this worked for me to remove the extra offending line:

    sed -i '/kitware.com/d' /etc/apt/sources.list

impala454, 2023-07-06:
FWIW you can also solve this with a one liner, if you don't care about installing kitware's keyring repo.  This will simply update the offending key:

    curl -sSL https://apt.kitware.com/keys/kitware-archive-latest.asc | apt-key add -

closed 2025-02-21 by johnnynunez

dkyol, 2025-04-03:
I am not sure these solutions still apply?

`apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).`

Is there are work around for installing docker?