dusty-nv/jetson-containers #217: ROS humble for r32.7.1 possible?

bjajoh, 2023-01-24:
I love these containers but due to missing CSI support on jetpack 5 for some carriers and cameras I need to rely on r32.7.1. (As probably a lot of people)
However there is no image here. Is it possible to add this?
What are the stepping stones?
Anyway I can assist?

dusty-nv, 2023-01-24:
@bjajoh as I recall, I tried to build Humble for JetPack 4, but the delta between Ubuntu 18.04 and Ubuntu 22.04 was too large and I was unable to overcome the build errors in order to get it compiling.  You are welcome to try and build it of course!

bjajoh, 2023-01-25:
@dusty-nv Thanks for your quick reply! I will give it a go and create a PR if I find a way to do it!

closed 2023-01-25 by bjajoh